lvgl/lv_esp_idf #134: with ST7789 display controller error in ili9xxx.c data undeclared

lefty01, 2020-05-22:
with current master branch I  configure for ST7789 display controller and get errors while ili9xxx displayer driver files are being compiled.
(This is supposed to run on board TTGO-T see here: https://github.com/Xinyuan-LilyGO/TTGO-T-Display)
following is part of the sdkconfig file after menuconfig
--- sdkconfig ---
```
# LittlevGL (LVGL) TFT Display controller
#
CONFIG_LVGL_PREDEFINED_DISPLAY_NONE=y
# CONFIG_LVGL_PREDEFINED_DISPLAY_WROVER4 is not set
# CONFIG_LVGL_PREDEFINED_DISPLAY_M5STACK is not set
# CONFIG_LVGL_PREDEFINED_DISPLAY_M5STICK is not set
# CONFIG_LVGL_PREDEFINED_DISPLAY_ERTFT0356 is not set
# CONFIG_LVGL_PREDEFINED_DISPLAY_ADA_FEATHERWING is not set
# CONFIG_LVGL_PREDEFINED_DISPLAY_RPI_MPI3501 is not set
# CONFIG_LVGL_PREDEFINED_DISPLAY_WEMOS_LOLIN is not set
CONFIG_LVGL_TFT_DISPLAY_CONTROLLER_ST7789=y
CONFIG_LVGL_TFT_DISPLAY_PROTOCOL_SPI=y
CONFIG_LVGL_PREDEFINED_PINS_NONE=y
# CONFIG_LVGL_PREDEFINED_PINS_38V4 is not set
# CONFIG_LVGL_PREDEFINED_PINS_30 is not set
# CONFIG_LVGL_PREDEFINED_PINS_38V1 is not set
# CONFIG_LVGL_PREDEFINED_PINS_TKOALA is not set
# CONFIG_LVGL_TFT_DISPLAY_USER_CONTROLLER_ILI9341 is not set
# CONFIG_LVGL_TFT_DISPLAY_USER_CONTROLLER_ILI9486 is not set
# CONFIG_LVGL_TFT_DISPLAY_USER_CONTROLLER_ILI9488 is not set
CONFIG_LVGL_TFT_DISPLAY_USER_CONTROLLER_ST7789=y
# CONFIG_LVGL_TFT_DISPLAY_USER_CONTROLLER_HX8357 is not set
# CONFIG_LVGL_TFT_DISPLAY_USER_CONTROLLER_SH1107 is not set
# CONFIG_LVGL_TFT_DISPLAY_USER_CONTROLLER_SSD1306 is not set
# CONFIG_LVGL_TFT_DISPLAY_USER_CONTROLLER_FT81X is not set
CONFIG_LVGL_TFT_DISPLAY_SPI_HSPI=y
# CONFIG_LVGL_TFT_DISPLAY_SPI_VSPI is not set
CONFIG_LVGL_DISPLAY_WIDTH=135
CONFIG_LVGL_DISPLAY_HEIGHT=240
# CONFIG_LVGL_INVERT_DISPLAY is not set
CONFIG_LVGL_ENABLE_BACKLIGHT_CONTROL=y
CONFIG_LVGL_BACKLIGHT_ACTIVE_LVL=y

```
** compile log error:
```
/home/al/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -DLV_CONF_INCLUDE_SIMPLE=1 -Iconfig -I../components/lvgl_esp32_drivers/lvgl_tft -I/data/al/esp/esp-idf/components/newlib/platform_include -I/data/al/esp/esp-idf/components/freertos/include -I/data/al/esp/esp-idf/components/freertos/xtensa/include -I/data/al/esp/esp-idf/components/heap/include -I/data/al/esp/esp-idf/components/log/include -I/data/al/esp/esp-idf/components/lwip/include/apps -I/data/al/esp/esp-idf/components/lwip/include/apps/sntp -I/data/al/esp/esp-idf/components/lwip/lwip/src/include -I/data/al/esp/esp-idf/components/lwip/port/esp32/include -I/data/al/esp/esp-idf/components/lwip/port/esp32/include/arch -I/data/al/esp/esp-idf/components/soc/src/esp32/. -I/data/al/esp/esp-idf/components/soc/src/esp32/include -I/data/al/esp/esp-idf/components/soc/include -I/data/al/esp/esp-idf/components/esp_rom/include -I/data/al/esp/esp-idf/components/esp_common/include -I/data/al/esp/esp-idf/components/esp_system/include -I/data/al/esp/esp-idf/components/xtensa/include -I/data/al/esp/esp-idf/components/xtensa/esp32/include -I/data/al/esp/esp-idf/components/esp32/include -I/data/al/esp/esp-idf/components/driver/include -I/data/al/esp/esp-idf/components/driver/esp32/include -I/data/al/esp/esp-idf/components/esp_ringbuf/include -I/data/al/esp/esp-idf/components/efuse/include -I/data/al/esp/esp-idf/components/efuse/esp32/include -I/data/al/esp/esp-idf/components/espcoredump/include -I/data/al/esp/esp-idf/components/esp_timer/include -I/data/al/esp/esp-idf/components/soc/soc/esp32/. -I/data/al/esp/esp-idf/components/soc/soc/esp32/include -I/data/al/esp/esp-idf/components/soc/soc/esp32/../include -I/data/al/esp/esp-idf/components/vfs/include -I/data/al/esp/esp-idf/components/esp_wifi/include -I/data/al/esp/esp-idf/components/esp_wifi/esp32/include -I/data/al/esp/esp-idf/components/esp_event/include -I/data/al/esp/esp-idf/components/esp_netif/include -I/data/al/esp/esp-idf/components/esp_eth/include -I/data/al/esp/esp-idf/components/tcpip_adapter/include -I/data/al/esp/esp-idf/components/app_trace/include -I../components/lvgl -I../components/lvgl/lvgl -mlongcalls -Wno-frame-address   -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -Og -std=gnu99 -Wno-old-style-declaration -D_GNU_SOURCE -DIDF_VER=\"v4.2-dev-1575-ga3520970f\" -DESP_PLATFORM -MD -MT esp-idf/lvgl_tft/CMakeFiles/__idf_lvgl_tft.dir/ili9486.c.obj -MF esp-idf/lvgl_tft/CMakeFiles/__idf_lvgl_tft.dir/ili9486.c.obj.d -o esp-idf/lvgl_tft/CMakeFiles/__idf_lvgl_tft.dir/ili9486.c.obj   -c ../components/lvgl_esp32_drivers/lvgl_tft/ili9486.c
../components/lvgl_esp32_drivers/lvgl_tft/ili9486.c: In function 'ili9486_init':
../components/lvgl_esp32_drivers/lvgl_tft/ili9486.c:121:21: error: 'data' undeclared (first use in this function)
  ili9486_send_data(&data, 1);
                     ^~~~
../components/lvgl_esp32_drivers/lvgl_tft/ili9486.c:121:21: note: each undeclared identifier is reported only once for each function it appears in
[1148/1227] Building C object esp-idf/lvgl_tft/CMakeFiles/__idf_lvgl_tft.dir/ili9488.c.obj
FAILED: esp-idf/lvgl_tft/CMakeFiles/__idf_lvgl_tft.dir/ili9488.c.obj 
/home/al/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -DLV_CONF_INCLUDE_SIMPLE=1 -Iconfig -I../components/lvgl_esp32_drivers/lvgl_tft -I/data/al/esp/esp-idf/components/newlib/platform_include -I/data/al/esp/esp-idf/components/freertos/include -I/data/al/esp/esp-idf/components/freertos/xtensa/include -I/data/al/esp/esp-idf/components/heap/include -I/data/al/esp/esp-idf/components/log/include -I/data/al/esp/esp-idf/components/lwip/include/apps -I/data/al/esp/esp-idf/components/lwip/include/apps/sntp -I/data/al/esp/esp-idf/components/lwip/lwip/src/include -I/data/al/esp/esp-idf/components/lwip/port/esp32/include -I/data/al/esp/esp-idf/components/lwip/port/esp32/include/arch -I/data/al/esp/esp-idf/components/soc/src/esp32/. -I/data/al/esp/esp-idf/components/soc/src/esp32/include -I/data/al/esp/esp-idf/components/soc/include -I/data/al/esp/esp-idf/components/esp_rom/include -I/data/al/esp/esp-idf/components/esp_common/include -I/data/al/esp/esp-idf/components/esp_system/include -I/data/al/esp/esp-idf/components/xtensa/include -I/data/al/esp/esp-idf/components/xtensa/esp32/include -I/data/al/esp/esp-idf/components/esp32/include -I/data/al/esp/esp-idf/components/driver/include -I/data/al/esp/esp-idf/components/driver/esp32/include -I/data/al/esp/esp-idf/components/esp_ringbuf/include -I/data/al/esp/esp-idf/components/efuse/include -I/data/al/esp/esp-idf/components/efuse/esp32/include -I/data/al/esp/esp-idf/components/espcoredump/include -I/data/al/esp/esp-idf/components/esp_timer/include -I/data/al/esp/esp-idf/components/soc/soc/esp32/. -I/data/al/esp/esp-idf/components/soc/soc/esp32/include -I/data/al/esp/esp-idf/components/soc/soc/esp32/../include -I/data/al/esp/esp-idf/components/vfs/include -I/data/al/esp/esp-idf/components/esp_wifi/include -I/data/al/esp/esp-idf/components/esp_wifi/esp32/include -I/data/al/esp/esp-idf/components/esp_event/include -I/data/al/esp/esp-idf/components/esp_netif/include -I/data/al/esp/esp-idf/components/esp_eth/include -I/data/al/esp/esp-idf/components/tcpip_adapter/include -I/data/al/esp/esp-idf/components/app_trace/include -I../components/lvgl -I../components/lvgl/lvgl -mlongcalls -Wno-frame-address   -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -Og -std=gnu99 -Wno-old-style-declaration -D_GNU_SOURCE -DIDF_VER=\"v4.2-dev-1575-ga3520970f\" -DESP_PLATFORM -MD -MT esp-idf/lvgl_tft/CMakeFiles/__idf_lvgl_tft.dir/ili9488.c.obj -MF esp-idf/lvgl_tft/CMakeFiles/__idf_lvgl_tft.dir/ili9488.c.obj.d -o esp-idf/lvgl_tft/CMakeFiles/__idf_lvgl_tft.dir/ili9488.c.obj   -c ../components/lvgl_esp32_drivers/lvgl_tft/ili9488.c
../components/lvgl_esp32_drivers/lvgl_tft/ili9488.c: In function 'ili9488_init':
../components/lvgl_esp32_drivers/lvgl_tft/ili9488.c:125:21: error: 'data' undeclared (first use in this function)
  ili9488_send_data(&data, 1);
                     ^~~~
../components/lvgl_esp32_drivers/lvgl_tft/ili9488.c:125:21: note: each undeclared identifier is reported only once for each function it appears in
[... cut ...]
... note, this does at least bring up the display but I think the resolution might be wrong or something but this is a different issue ;)

For this error I'm wondering if ili9xxx.c shall be compiled at all if st7789 is configured?
and if this is okay then looks to me that some of the config options might not be correct ... 

**ESP-IDF version**
current git master 

**Development kit used**
TTGO-T (https://bit.ly/36nUNK2)

**Development machine OS**
same result on either Linux Mint 19.1 or a Fedora 31

C47D, 2020-05-22:
Hi,

Thanks for the report, I will look at it. The ilixxx source files are compiled even when not using them, I will try to find a way to change that.

C47D, 2020-05-22:
Can you test this branch? https://github.com/lvgl/lv_port_esp32/tree/fix_134

lefty01, 2020-05-23:
H, yeah after checkout fix_134 branch idf.py build was compiling the code without error! thx!
so I can confirm this is working for me.

also tried one build with
CONFIG_LVGL_TFT_DISPLAY_CONTROLLER_ILI9341=y
this compiles as well (showing your pragma msg)

```
/home/aloeff/code/misc/github.com/lv_port_esp32/components/lvgl_esp32_drivers/lvgl_tft/ili9488.c: In function 'ili9488_init':
/home/aloeff/code/misc/github.com/lv_port_esp32/components/lvgl_esp32_drivers/lvgl_tft/ili9488.c:111:9: note: #pragma message: ILI9488 - LANDSCAPE
 #pragma message "ILI9488 - LANDSCAPE"
         ^~~~~~~
[ 74%] Building C object esp-idf/lv_examples/CMakeFiles/__idf_lv_examples.dir/lv_examples/lv_tests/lv_test_objx/lv_test_btnm/lv_test_btnm.c.obj
/home/aloeff/code/misc/github.com/lv_port_esp32/components/lvgl_esp32_drivers/lvgl_tft/ili9486.c: In function 'ili9486_init':
/home/aloeff/code/misc/github.com/lv_port_esp32/components/lvgl_esp32_drivers/lvgl_tft/ili9486.c:107:9: note: #pragma message: ILI9486 - LANDSCAPE
 #pragma message "ILI9486 - LANDSCAPE"
         ^~~~~~~
[ 74%] Building C object esp-idf/lv_examples/CMakeFiles/__idf_lv_examples.dir/lv_examples/lv_tests/lv_test_objx/lv_test_canvas/lv_test_canvas.c.obj
/home/aloeff/code/misc/github.com/lv_port_esp32/components/lvgl_esp32_drivers/lvgl_tft/ili9341.c: In function 'ili9341_init':
/home/aloeff/code/misc/github.com/lv_port_esp32/components/lvgl_esp32_drivers/lvgl_tft/ili9341.c:147:13: note: #pragma message: ILI9341 - LANDSCAPE
     #pragma message "ILI9341 - LANDSCAPE"
             ^~~~~~~

```

C47D, 2020-05-23:
The problem is how we were handling the rotation on the ILI9xxx controllers, as you saw, all the *.c files on lvgl_tft directory are compiled, even those not used by your display controller.

We've working on getting this repo ready for LVGL v7, hope this issue solves with that improvements.

C47D, 2020-06-04:
Fixed with 8ba9d9d1662400b0626fa7f4766f34f234c30a08

closed 2020-06-04 by C47D