lvgl/lv_esp_idf #146: Strange Adafruit TFT FeatherWing - 3.5" 480x320 issue

petestewardson-schneider, 2020-06-04:
Hi

I'm using the Adafruit 3.5"  TFT Featherwing with an Adafruit Huzzah32 board and having an issue.

The display appears to be working fine using this example code in that it it shows the write/list/chart example screens. However the touch operation is not working, and every 3s roughly the screen switches to the next of the three tabs in the example cycling around them. I am using the configuration for the TFT featherwing in menuconfig but this appears to also select settings for the touch controller without a way to change the CS so I'm not sure if it is this. 

I'm building on Windows with ESP-idf 4.0.1 and everything compiles and runs without error but not what I'd expect on screen.

C47D, 2020-06-04:
> However the touch operation is not working, and every 3s roughly the screen switches to the next of the three tabs in the example cycling around them.

The example is configured to automatically switch tabs.

We're about to merge #110 into master, could you test it again later today? I't updates to LVGL v7 an several bug fixes.

embeddedt, 2020-06-04:
@petestewardson-schneider https://github.com/lvgl/lv_port_esp32/blob/c1f43bfb090df293059ab52baabbdbdd8df00712/components/lv_examples/lv_ex_conf.h#L45

petestewardson-schneider, 2020-06-10:
OK so I've updated to the latest code after the merge  and disabled the slideshow as above. I'm still getting no touch input in the example app. 

I'm new to using esp-idf but is there anyway I can help to diagnose with the HW I have?

petestewardson-schneider, 2020-06-11:
Hi, 

I've attached the config, but there is clearly something not correct when using the monitor command ( I didn't know about that one so thanks )

`--- idf_monitor on COM8 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:7064
load:0x40078000,len:13072
ho 0 tail 12 room 4
load:0x40080400,len:3896
0x40080400: _init at ??:?

entry 0x40080688
I (30) boot: ESP-IDF v4.1-beta2 2nd stage bootloader
I (30) boot: compile time 10:52:01
I (30) boot: chip revision: 1
I (33) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (40) boot.esp32: SPI Speed      : 40MHz
I (45) boot.esp32: SPI Mode       : DIO
I (50) boot.esp32: SPI Flash Size : 2MB
I (54) boot: Enabling RNG early entropy source...
I (60) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (85) boot:  2 factory          factory app      00 00 00010000 00100000
I (93) boot: End of partition table
I (97) boot_comm: chip revision: 1, min. application chip revision: 0
I (104) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x0c54c ( 50508) map
I (132) esp_image: segment 1: paddr=0x0001c574 vaddr=0x3ffb0000 size=0x0225c (  8796) load
I (136) esp_image: segment 2: paddr=0x0001e7d8 vaddr=0x40080000 size=0x00404 (  1028) load
0x40080000: _WindowOverflow4 at C:/Users/SESA406760/Desktop/esp-idf_4_1/esp-idf/components/freertos/xtensa_vectors.S:1778

I (140) esp_image: segment 3: paddr=0x0001ebe4 vaddr=0x40080404 size=0x01434 (  5172) load
I (150) esp_image: segment 4: paddr=0x00020020 vaddr=0x400d0020 size=0x43e90 (278160) map
0x400d0020: _stext at ??:?

I (263) esp_image: segment 5: paddr=0x00063eb8 vaddr=0x40081838 size=0x09950 ( 39248) load
0x40081838: esp_intr_enable at C:/Users/SESA406760/Desktop/esp-idf_4_1/esp-idf/components/esp32/intr_alloc.c:826

I (286) boot: Loaded app from partition at offset 0x10000
I (286) boot: Disabling RNG early entropy source...
I (287) cpu_start: Pro cpu up.
I (290) cpu_start: Application information:
I (295) cpu_start: Project name:     lvgl-demo
I (300) cpu_start: App version:      d01c8d3
I (305) cpu_start: Compile time:     Jun 11 2020 11:09:07
I (311) cpu_start: ELF file SHA256:  18f1ef3fde52e208...
I (317) cpu_start: ESP-IDF:          v4.1-beta2
I (323) cpu_start: Starting app cpu, entry point is 0x400811fc
0x400811fc: call_start_cpu1 at C:/Users/SESA406760/Desktop/esp-idf_4_1/esp-idf/components/esp32/cpu_start.c:275

I (0) cpu_start: App cpu up.
I (333) heap_init: Initializing. RAM available for dynamic allocation:
I (340) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (346) heap_init: At 3FFCE010 len 00011FF0 (71 KiB): DRAM
I (352) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (359) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (365) heap_init: At 4008B188 len 00014E78 (83 KiB): IRAM
I (371) cpu_start: Pro cpu start user code
I (389) spi_flash: detected chip: gd
I (390) spi_flash: flash io: dio
W (390) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (400) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (20) lvgl_helpers: Display hor size: 480, ver size: 320
I (20) lvgl_helpers: Display buffer size: 19200
I (20) lvgl_helpers: Initializing shared SPI master
I (30) lvgl_helpers: Configuring SPI host VSPI_HOST (2)
I (30) lvgl_helpers: MISO pin: 19, MOSI pin: 18, SCLK pin: 5
I (40) lvgl_helpers: Max transfer size: 38400 (bytes)
I (50) lvgl_helpers: Initializing SPI bus...
I (50) disp_spi: Adding SPI device
I (50) disp_spi: Clock speed: 26000000Hz, mode: 0, CS pin: 15
I (190) HX8357: Initialization.
I (990) STMPE610: Initialization.
I (1060) STMPE610: SPI_CFG = 0x8
E (1060) STMPE610: Incorrect version: 0x0`

[sdkconfig.txt](https://github.com/lvgl/lv_port_esp32/files/4764096/sdkconfig.txt)

richard-mckenna, 2020-07-02:
I'm just trying to get this working on the FeatherWing as well. Display is working but touch not working. I've tested the hardware with the Arduino [touchpaint_featherwing example application](https://github.com/adafruit/Adafruit_HX8357_Library/tree/master/examples/touchpaint_featherwing) and both display and touch works.

Is there anything I can try?

[sdkconfig.txt](https://github.com/lvgl/lv_port_esp32/files/4864032/sdkconfig.txt)

Same error in idf.py monitor

```
--- idf_monitor on COM6 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6924
load:0x40078000,len:14076
ho 0 tail 12 room 4
load:0x40080400,len:4304
entry 0x400806e8
I (71) boot: Chip Revision: 1
I (71) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (41) boot: ESP-IDF v4.0.1-dirty 2nd stage bootloader
I (41) boot: compile time 11:58:53
I (41) boot: Enabling RNG early entropy source...
I (46) boot: SPI Speed      : 40MHz
I (50) boot: SPI Mode       : DIO
I (54) boot: SPI Flash Size : 2MB
I (58) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (77) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (84) boot:  2 factory          factory app      00 00 00010000 00100000
I (92) boot: End of partition table
I (96) boot_comm: chip revision: 1, min. application chip revision: 0
I (103) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x0c618 ( 50712) map
I (130) esp_image: segment 1: paddr=0x0001c640 vaddr=0x3ffb0000 size=0x021b4 (  8628) load
I (134) esp_image: segment 2: paddr=0x0001e7fc vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at C:/Users/richa/Desktop/esp-idf-2/components/freertos/xtensa_vectors.S:1778

I (137) esp_image: segment 3: paddr=0x0001ec04 vaddr=0x40080400 size=0x0140c (  5132) load
I (148) esp_image: segment 4: paddr=0x00020018 vaddr=0x400d0018 size=0x43b90 (277392) map
0x400d0018: _stext at ??:?

I (254) esp_image: segment 5: paddr=0x00063bb0 vaddr=0x4008180c size=0x0990c ( 39180) load
0x4008180c: esp_intr_enable at C:/Users/richa/Desktop/esp-idf-2/components/esp32/intr_alloc.c:810

I (278) boot: Loaded app from partition at offset 0x10000
I (278) boot: Disabling RNG early entropy source...
I (279) cpu_start: Pro cpu up.
I (282) cpu_start: Application information:
I (287) cpu_start: Project name:     lvgl-demo
I (292) cpu_start: App version:      d144046
I (297) cpu_start: Compile time:     Jul  2 2020 12:42:05
I (303) cpu_start: ELF file SHA256:  9f229b833fdb4bfd...
I (309) cpu_start: ESP-IDF:          v4.0.1-dirty
I (314) cpu_start: Starting app cpu, entry point is 0x40081210
0x40081210: call_start_cpu1 at C:/Users/richa/Desktop/esp-idf-2/components/esp32/cpu_start.c:271

I (0) cpu_start: App cpu up.
I (325) heap_init: Initializing. RAM available for dynamic allocation:
I (331) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (337) heap_init: At 3FFCE760 len 000118A0 (70 KiB): DRAM
I (344) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (350) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (356) heap_init: At 4008B118 len 00014EE8 (83 KiB): IRAM
I (363) cpu_start: Pro cpu start user code
I (381) spi_flash: detected chip: generic
I (381) spi_flash: flash io: dio
W (381) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (392) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (20) lvgl_helpers: Display hor size: 480, ver size: 320
I (20) lvgl_helpers: Display buffer size: 19200
I (20) lvgl_helpers: Initializing shared SPI master
I (30) lvgl_helpers: Configuring SPI host VSPI_HOST (2)
I (30) lvgl_helpers: MISO pin: 19, MOSI pin: 18, SCLK pin: 5
I (40) lvgl_helpers: Max transfer size: 38400 (bytes)
I (50) lvgl_helpers: Initializing SPI bus...
I (50) disp_spi: Adding SPI device
I (50) disp_spi: Clock speed: 26000000Hz, mode: 0, CS pin: 15
I (190) HX8357: Initialization.
I (990) STMPE610: Initialization.
I (1070) STMPE610: SPI_CFG = 0x8
E (1070) STMPE610: Incorrect version: 0x0
```

petestewardson-schneider, 2020-07-24:
I've not been able to address this issue so far.

petestewardson-schneider, 2021-01-05:
Is there any help anyone can give on this? It seems it's confirmed not to be working but I don't know where to start to debug.