lvgl/lv_esp_idf #147: lv_theme_material_init not found if configured with NO_THEME

Cikkcakk, 2020-06-05:
We use GitHub issues for development related discussions.
Please use the [forum](https://forum.littlevgl.com/) to ask questions.

**Describe the issue**

**Code to reproduce the issue**

**Expected Results**

**Actual Results**

**ESP32 Chip version**

**ESP-IDF version**

**Development kit used**

**Development machine OS**

**Compilation warnings/errors (if available)**

**If possible, copy the compilation log into a file and attach it here**

C47D, 2020-06-05:
Can you test this branch https://github.com/lvgl/lv_port_esp32/tree/kconfig_themes

I've added theme configuration options in the menuconfig interface, you'll need to select an theme init function.

C47D, 2020-06-07:
Merged into master, update your clone to test it.

Cikkcakk, 2020-06-10:
there are no problems anymore with the latest master

closed 2020-06-10 by Cikkcakk