lvgl/lv_esp_idf #149: IL3820 (GDEH029A1) crashes when setting to portrait modes

huming2207, 2020-06-08:
**Describe the issue**

Looks like the hardware rotation related code are not implemented for IL3820, and it may cause crashes when setting to portrait (or inverted portrait) modes.

I think I know why it crashes with garbage result rendered. Because this register is not set correctly: 

[image]

I will try fixing it later tonight and submit a PR.

**Code to reproduce the issue**

1. Use the latest code at this commit: https://github.com/lvgl/lv_port_esp32/commit/9b987457161681a184d9421e960f20f56b22612b

2. Enable IL3820, set to 296x128 resolution, and portrait modes in menuconfig

**Expected Results**

It should just works

**Actual Results**

It crashes like this, with some garbage results rendered on the screen:

```
ELF file SHA256: 3f5709268e81b560

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:7068
load:0x40078000,len:13696
load:0x40080400,len:4000
0x40080400: _init at ??:?

entry 0x40080688
I (29) boot: ESP-IDF v4.2-dev-1660-g7d7521367 2nd stage bootloader
I (29) boot: compile time 12:21:49
I (29) boot: chip revision: 1
I (33) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (40) boot.esp32: SPI Speed      : 40MHz
I (45) boot.esp32: SPI Mode       : DIO
I (50) boot.esp32: SPI Flash Size : 2MB
I (54) boot: Enabling RNG early entropy source...
I (60) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (85) boot:  2 factory          factory app      00 00 00010000 00100000
I (93) boot: End of partition table
I (97) boot_comm: chip revision: 1, min. application chip revision: 0
I (104) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x0bde0 ( 48608) map
I (132) esp_image: segment 1: paddr=0x0001be08 vaddr=0x3ffb0000 size=0x02358 (  9048) load
I (136) esp_image: segment 2: paddr=0x0001e168 vaddr=0x40080000 size=0x00404 (  1028) load
0x40080000: _WindowOverflow4 at /home/hu/esp/esp-idf/components/freertos/xtensa/xtensa_vectors.S:1730

I (139) esp_image: segment 3: paddr=0x0001e574 vaddr=0x40080404 size=0x01aa4 (  6820) load
I (150) esp_image: segment 4: paddr=0x00020020 vaddr=0x400d0020 size=0x27330 (160560) map
0x400d0020: _stext at ??:?

I (217) esp_image: segment 5: paddr=0x00047358 vaddr=0x40081ea8 size=0x09934 ( 39220) load
0x40081ea8: esp_restart_noos at /home/hu/esp/esp-idf/components/esp32/system_api_esp32.c:98

I (241) boot: Loaded app from partition at offset 0x10000
I (241) boot: Disabling RNG early entropy source...
I (242) cpu_start: Pro cpu up.
I (245) cpu_start: Application information:
I (250) cpu_start: Project name:     lvgl-demo
I (255) cpu_start: App version:      9b98745-dirty
I (261) cpu_start: Compile time:     Jun  8 2020 14:34:16
I (267) cpu_start: ELF file SHA256:  3f5709268e81b560...
I (273) cpu_start: ESP-IDF:          v4.2-dev-1660-g7d7521367
I (279) cpu_start: Starting app cpu, entry point is 0x400816f0
0x400816f0: call_start_cpu1 at /home/hu/esp/esp-idf/components/esp32/cpu_start.c:286

I (271) cpu_start: App cpu up.
I (290) heap_init: Initializing. RAM available for dynamic allocation:
I (297) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (303) heap_init: At 3FFBC308 len 00023CF8 (143 KiB): DRAM
I (309) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (315) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (322) heap_init: At 4008B7DC len 00014824 (82 KiB): IRAM
I (328) cpu_start: Pro cpu start user code
I (346) spi_flash: detected chip: gd
I (347) spi_flash: flash io: dio
W (347) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (357) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (10) lvgl_helpers: Display hor size: 296, ver size: 128
I (10) lvgl_helpers: Display buffer size: 4736
I (10) lvgl_helpers: Initializing SPI master for display
I (20) lvgl_helpers: Configuring SPI host HSPI_HOST (1)
I (30) lvgl_helpers: MISO pin: -1, MOSI pin: 21, SCLK pin: 22
I (30) lvgl_helpers: Max transfer size: 9472 (bytes)
I (40) lvgl_helpers: Initializing SPI bus...
I (50) disp_spi: Adding SPI device
I (50) disp_spi: Clock speed: 40000000Hz, mode: 0, CS pin: 17
I (50) IL3820: init
I (110) IL3820: clear
I (1810) IL3820: clear
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400df29f  PS      : 0x00060930  A0      : 0x800d6a89  A1      : 0x3ffc1f50  
0x400df29f: lv_disp_is_true_double_buf at /home/hu/Projects/lv_port_esp32/build/../components/lvgl/lvgl/src/lv_hal/lv_hal_disp.c:400

A2      : 0x00000000  A3      : 0x3ffc1f70  A4      : 0x00000000  A5      : 0x00000000  
A6      : 0x3ffb43c4  A7      : 0x00000001  A8      : 0x800d6836  A9      : 0x3ffc1f30  
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x400d600c  A13     : 0x3ffb43c4  
0x400d600c: lv_obj_design at /home/hu/Projects/lv_port_esp32/build/../components/lvgl/lvgl/src/lv_core/lv_obj.c:3507

A14     : 0x00000127  A15     : 0x00000000  SAR     : 0x00000020  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x400013f9  LEND    : 0x4000140d  LCOUNT  : 0xfffffffc  

Backtrace:0x400df29c:0x3ffc1f50 0x400d6a86:0x3ffc1f70 0x400d6bdf:0x3ffc1fa0 0x400d6cb9:0x3ffc1fd0 0x400d6d1c:0x3ffc1ff0 0x400e069d:0x3ffc2020 0x400e0788:0x3ffc2040 0x400d33d2:0x3ffc2060 0x40085505:0x3ffc20c0
0x400df29c: lv_disp_is_true_double_buf at /home/hu/Projects/lv_port_esp32/build/../components/lvgl/lvgl/src/lv_hal/lv_hal_disp.c:399

0x400d6a86: lv_refr_area_part at /home/hu/Projects/lv_port_esp32/build/../components/lvgl/lvgl/src/lv_core/lv_refr.c:507

0x400d6bdf: lv_refr_area at /home/hu/Projects/lv_port_esp32/build/../components/lvgl/lvgl/src/lv_core/lv_refr.c:455 (discriminator 2)

0x400d6cb9: lv_refr_areas at /home/hu/Projects/lv_port_esp32/build/../components/lvgl/lvgl/src/lv_core/lv_refr.c:377

0x400d6d1c: _lv_disp_refr_task at /home/hu/Projects/lv_port_esp32/build/../components/lvgl/lvgl/src/lv_core/lv_refr.c:199

0x400e069d: lv_task_exec at /home/hu/Projects/lv_port_esp32/build/../components/lvgl/lvgl/src/lv_misc/lv_task.c:398 (discriminator 1)

0x400e0788: lv_task_handler at /home/hu/Projects/lv_port_esp32/build/../components/lvgl/lvgl/src/lv_misc/lv_task.c:142

0x400d33d2: guiTask at /home/hu/Projects/lv_port_esp32/build/../main/main.c:117

0x40085505: vPortTaskWrapper at /home/hu/esp/esp-idf/components/freertos/xtensa/port.c:143

ELF file SHA256: 3f5709268e81b560

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:7068
load:0x40078000,len:13696
load:0x40080400,len:4000
0x40080400: _init at ??:?

```

**ESP32 Chip version**

v1

**ESP-IDF version**

v4.2-dev-1660-g7d7521367

**Development kit used**

ESP-32S v1.1

**Development machine OS**

Ubuntu 20.04

huming2207, 2020-06-08:
Hello @C47D and @aircable 

Firstly, according to the code at the latest master commit, it does NOT have any rotation support for IL3820. It definitely won't work.  

So far I've tried adding the support to get it working, but no luck with these attempts:

1. Swapping `EPD_PANEL_WIDTH` and `EPD_PANEL_HEIGHT`: won't work - crash
2. Swapping the width and height + set the entry mode (register 0x11) to 0x01 or 0x03: won't work - crash
3. Swap the stuff mentioned above, also try swapping the `area->x1`, `area->x2` to `area->y1`, `area->y2` and some other width/height related macros, it won't crash any more but the screen will display some random garbage result.

Anyway I can't fix it for now as I have something else need to be done soon. But I'm happy to help if necessary.

By the way, the code should be cleaned up with some sorts of formatter/linter. Some spelling or abbreviations are wrong or weird. The overall code format makes me feels a bit hard to read.

Regards,
Jackson

C47D, 2020-06-08:
I agree the code needs to be clean up.

aircable, 2020-06-08:
I'm sorry I have not tested landscape mode, only portrait. And I'm pretty
sure it needs some
extra code for the transfer of partial blocks to the display.

I'll be back on working on the driver soon. In the meantime please feel
free to clean up the code.
I know, I have my own coding style ;-)

C47D, 2020-06-08:
Don't worry @aircable, I'm trying to get a display with this controller to test it.

huming2207, 2020-06-08:
Hello @C47D 

> Does the code works in landscape mode? If so, can you copy the monitor output in here?

It works in landscape mode. However, now the current landscape mode in `menuconfig` works but the screen is portrait (i.e. in 128x296, like this: https://github.com/aircable/lv_port_esp32/blob/master/images/Waveshare_EPD2in9.png), while the portrait mode in `menuconfig` will cause crashes or garbage result.

> I'm also trying to get a display with this controller to test it.

Just grab a Waveshare 2.9 inch module: https://www.waveshare.com/wiki/2.9inch_e-Paper_Module, or this one with a connector board if necessary: https://www.aliexpress.com/item/32218091929.html. They are the same.

huming2207, 2020-06-08:
Also looks like for this panel, according to GoodDisplay's demo code, this default orientation is  **128x296** instead of **296x128**. So now the landscape mode orientation is actually correct.

C47D, 2020-06-08:
> Just grab a Waveshare 2.9 inch module: https://www.waveshare.com/wiki/2.9inch_e-Paper_Module, or this one with a connector board if necessary: https://www.aliexpress.com/item/32218091929.html. They are the same.

Thanks for the link, I tried to buy some displays last weekend on ebay and they don't have the usual postal service available, and I can't afford the shipment cost. I will try to source the display locally, I'm based in Mexico, so it's a bit hard.

I checked the IL3028 code and indeed it doesn't handle portrait mode, I will try to send you some patches to enable this.

huming2207, 2020-06-08:
> Thanks for the link, I tried to buy some displays last weekend on ebay and they don't have the usual postal service available, and I can't afford the shipment cost. I will try to source the display [...]

Hello @C47D 

You should try the GoodDisplay official store on AliExpress.

If you really want to get one, I can contact GoodDisplay sales directly as I know a guy there. Please send me an email if necessary. 

Regards,
Jackson

huming2207, 2020-06-09:
By the way @C47D @aircable ,

I think this kind of polling is not acceptable for production, as it wastes the CPU times anyway: https://github.com/lvgl/lv_port_esp32/blob/8bc5e18432f710a12981f69721b9e34327ffa596/components/lvgl_esp32_drivers/lvgl_tft/il3820.c#L94

It really should be replaced to interrupt instead.

C47D, 2020-06-09:
Sure, contributions are very welcomed, I'm waiting for the confirmation of the display I tried to buy, it should take a long time to arrive tho.

huming2207, 2020-06-09:
Hello @C47D 

> We can make this project better together, this is the first revision of this particular driver and it can be improved, other drivers have been improved and surely can improve even more.

Yea I totally agree. This is just a PoC level thing. But it should be approved later.

If you have any difficulties on buying those panels, please send me an email at huming2207@gmail.com . I can help you out.

Jackson

huming2207, 2020-06-09:
Hello @C47D @aircable 

Here's the latest status. The code can be found here: https://github.com/huming2207/lv_port_esp32/tree/fix/il3820-rotate

I've adjusted something in `il3820_flush()` and `il3820_set_px_cb()` to get it properly flushed to framebuffer in 296x128 direction, instead of 128x296. I've also set the R11h data entry register to 0x03. But still no luck.

Here's what it looks like for now.

[image]

The actual dem UI code is drawing a 10px * 5px square, like this: 

```c
    static lv_style_t obj_style;
    lv_style_set_bg_color(&obj_style, LV_STATE_DEFAULT, LV_COLOR_BLACK);

    lv_obj_t* obj = lv_obj_create(lv_scr_act(), NULL);
    lv_obj_set_size(obj, 10, 5);
    lv_obj_align(obj1, NULL, LV_ALIGN_IN_TOP_LEFT, 0, 0);
    lv_obj_add_style(obj1, LV_OBJ_PART_MAIN, &obj_style);
```

From the latest pattern rendered, it looks like the panel hasn't been rotated to 296x128 at all. Maybe I've done something stupid. Please point it out if possible. Thanks.

Meanwhile I'm not too sure for the algorithm used in `il3820_set_px_cb()`. I think my implementation for the portrait mode is totally wrong. Please take a look.

And also I don't understand why we need to round the x axis in the `round_cb()` in the original landscape mode. @aircable Could you please explain why you've done that? Thanks!

Regards,
Jackson

huming2207, 2020-06-10:
Hi @C47D 

> I'm not 100% sure but when using monochrome displays the LVGL theme configuration should be updated, it's not the same as using a RGB TFT display, how are you configuring the LVGL themes?

Yea I did set to monochrome theme with ascii8 font.

Jackson

C47D, 2020-06-23:
Hi @huming2207, sorry for the late reply, just got the display.

huming2207, 2020-06-24:
@C47D Thanks for your update. What's your current config?

C47D, 2020-06-24:
I'm using it in landscape mode, the weird thing is that I think it's crashing somewhere on the lvgl code and not in the SPI code.
I'm cleaning up the code and adding comments to it in a separated branch, I haven't pushed it to remote.

huming2207, 2020-06-25:
> I'm using it in landscape mode, the weird thing is that I think it's crashing somewhere on the lvgl code and not in the SPI code.
> I'm cleaning up the code and adding comments to it in a separated branch, I haven't pushed it to remote.

Because the buffer gets overflow, as the landscape mode coordinates are wrong.

huming2207, 2020-06-25:
What I understood is, we need to do is to rotate the framebuffer first, before pushing to the panel. Unlike U8g2 or some other UI libraries, LVGL does not support rotating directly. We need to either 1. implement the framebuffer rotation for these panels or 2. add native rotation support for LVGL, like U8g2 did.

C47D, 2020-06-25:
Hi,

Here's what I've tried, this is the first time I work with e-paper displays:
- Try to replicate the issue on the CodeBlock simulator, I can't because the win_driver doesn't work with `LV_COLOR_DEPTH=1`.
- Set `DISP_BUF_SIZE` to `LV_HOR_RES_MAX*LV_VER_RES_MAX`, this make the esp32 not to hardfault, but I get garbage as output, this is where we need to do what you say. But LVGL does support rotation, on main, the `lv_disp_drv_t disp_drv` have a field named `rotated`, we need to set it to 1. I need to make this work for all other drivers, so we don't need to change the `LV_HOR_RES_MAX` and `LV_VER_RES_MAX` in `lv_conf.h`.

I'm just guessing that the overflow is caused when calculating the `byte_index` in the `il3820_set_px_cb function`:

```c
void il3820_set_px_cb(struct _disp_drv_t * disp_drv, uint8_t* buf, lv_coord_t buf_w, lv_coord_t x, lv_coord_t y,
        lv_color_t color, lv_opa_t opa) {

	/* buf_w will be ignored, the configured CONFIG_LVGL_DISPLAY_HEIGHT and _WIDTH,
	   and CONFIG_LVGL_DISPLAY_ORIENTATION_LANDSCAPE and _PORTRAIT will be used. */ 		
    uint16_t byte_index = 0;
    uint8_t  bit_index = 0;

	byte_index = y + (( x>>3 ) * CONFIG_LVGL_DISPLAY_HEIGHT);
	bit_index  = x & 0x7;

    if ( color.full != 0 ) {
        BIT_SET(buf[byte_index], 7 - bit_index);
    } else {
        BIT_CLEAR(buf[byte_index], 7 - bit_index);
    }
}
```

I don't have any jtag to debug this properly, so I need to learn how this displays graphic RAM gets updated, how to properly configure the Data Entry Mode and make it work for all four configurations.

I will push my local work branch, it's mostly cleanup of the il3820 driver code.

C47D, 2020-06-27:
Hi,

Just wanted to share my progress, I've added some logging to `set_px` and `flush` functions, here's the output for printing an 'a' into the top left side of the display.

```
I (10) lvgl_helpers: Display hor size: 128, ver size: 296
I (10) lvgl_helpers: Display buffer size: 4736
I (10) lvgl_helpers: Initializing SPI master for display
I (20) lvgl_helpers: Configuring SPI host HSPI_HOST (1)
I (30) lvgl_helpers: MISO pin: -1, MOSI pin: 23, SCLK pin: 18
I (30) lvgl_helpers: Max transfer size: 9472 (bytes)
I (40) lvgl_helpers: Initializing SPI bus...
I (50) disp_spi: Adding SPI device
I (50) disp_spi: Clock speed: 8000000Hz, mode: 0, CS pin: 21
I (50) IL3820: init
I (110) IL3820: clear
I (1800) IL3820: clear
I (2310) IL3820: px: {2:2}, b: 2
I (2310) IL3820: CLEAR(buf[2], 5)
I (2310) IL3820: px: {3:2}, b: 3
I (2310) IL3820: CLEAR(buf[2], 4)
I (2320) IL3820: px: {4:2}, b: 4
I (2320) IL3820: CLEAR(buf[2], 3)
I (2320) IL3820: px: {5:3}, b: 5
I (2330) IL3820: CLEAR(buf[3], 2)
I (2330) IL3820: px: {2:4}, b: 2
I (2340) IL3820: CLEAR(buf[4], 5)
I (2340) IL3820: px: {3:4}, b: 3
I (2340) IL3820: CLEAR(buf[4], 4)
I (2340) IL3820: px: {4:4}, b: 4
I (2350) IL3820: CLEAR(buf[4], 3)
I (2360) IL3820: px: {5:4}, b: 5
I (2360) IL3820: CLEAR(buf[4], 2)
I (2360) IL3820: px: {1:5}, b: 1
I (2360) IL3820: CLEAR(buf[5], 6)
I (2370) IL3820: px: {5:5}, b: 5
I (2380) IL3820: CLEAR(buf[5], 2)
I (2380) IL3820: px: {2:6}, b: 2
I (2380) IL3820: CLEAR(buf[6], 5)
I (2380) IL3820: px: {3:6}, b: 3
I (2390) IL3820: CLEAR(buf[6], 4)
I (2400) IL3820: px: {4:6}, b: 4
I (2400) IL3820: CLEAR(buf[6], 3)
I (2400) IL3820: px: {5:6}, b: 5
I (2400) IL3820: CLEAR(buf[6], 2)
I (2410) IL3820: linelen: 16
I (2420) IL3820: IL3820_COLUMNS: 16
I (2420) IL3820: address: 0
I (2420) IL3820: flush: 0,127 at 0,295
```

And the 'a' being displayed with some notations
[image]

From this I can assume that if I want to "rotate" the display 180° I should set the X and Y counters to the max values on this display (296 and 128), and set the counters to decrement `_XDYDY`. I guess this is what you've been saying since the beginning, but until now I can see how LVGL turn pixels into black dots on the display (by clearing the 'bit' on the `buf` array).

huming2207, 2020-06-30:
Hi @C47D ,

Sorry I was extremely busy these days for some of my projects at work. Yes looks like the XDYDY mode didn't really work somehow.

Also, this official demo code may help: http://www.good-display.com/product/201.html

Regards,
Jackson

C47D, 2020-06-30:
Hi Jason,

I've been able to configure the display in landscape inverted. Will check the link later today. I been having limited access to my work computer.

Regards,
Carlos

huming2207, 2020-07-19:
Hello @C47D 

Sorry for the late reply. Have you got any luck for now?

Maybe we should also try to examine the initialise sequence.

Regards,
Jackson

C47D, 2020-07-26:
Hi @huming2207,

Sorry, I haven't been able to work on this, got a job non related to computers and almost had no free time to work on this, do you have any luck?

huming2207, 2020-07-30:
Hello @C47D 

Sorry I was busy working on another two projects for BLE with NimBLE on ESP32. I think the issue might be either the transform callback is still wrong or this display panel controller is buggy. 

We had another issue with another similar panel before. Their temperature register value is invalid if the read operation frequency is higher than 3MHz. Other registers can be read with 5MHz but that temperature register can only accept 3MHz. 
 But meanwhile, on their datasheet, the highest SPI read operation clock speed is 5MHz. So I'm not surprised if that rotation setting is really buggy.

Jackson

C47D, 2020-08-10:
Almost there...

[image]

huming2207, 2020-08-10:
@C47D Congrats! 

By the way, maybe it's a bit off-topic. But I think we can later port a QuickJS JavaScript binding. I've just ported a quick & dirty version of QuickJS for ESP32 last night: https://github.com/huming2207/quickjs-esp32

C47D, 2020-08-10:
I will create a new predefined display configuration for this e-paper display once it's working properly. Will let you know then.

huming2207, 2020-08-11:
@C47D Yep thx. I'm happy to check it out.

C47D, 2020-08-12:
I tried several data entry configurations but couldn't make it work, so I'm "mirroring" the X pixels in software, it's ugly but it seems to work, at least it displays `Hello World` properly.

[image]

Right now it only works on portrait and landscape orientations, none of the inverted options work. You need to set the display width to 128 and height to 296. It's not production ready code, if you have some free time test it and let me know if it fails or improvements you have in mind.

I pushed the changes to the `fix/il3820_portrait` branch.

huming2207, 2020-08-12:
Hi @C47D 

Great thanks for your work!

I just did a quick test and I think it looks all good for now. 

I've also tested a few times for the data entry mode. I think it is just buggy. It won't work as I expected but just generating some garbage results.

Regards,
Jackson

huming2207, 2020-08-12:
[image]

Here's my panel with your original code. It looks perfect!

Jackson

C47D, 2020-08-12:
Nice, I will create a new predefined display configuration so it's easier to choose the right options. Also this last photos are in landscape mode, not portrait, will fix that as well.

huming2207, 2020-08-13:
Hello @C47D 

Yep, thanks for your efforts too! 

Meanwhile, I think LVGL should provide its native support for rotation and better monochrome/greyscale support, preferably in the config, without any conversion callbacks in the display driver. This is for in case some weird panels that:

- have no or buggy hardware rotation in its controller chips;
- and/or, two or more primary/secondary-style controller chips in one panel (quite common in some larger e-paper panels)

This is actually quite common for some e-paper panels. It would be very helpful for us to port more panels for this library.

Regards,
Jackson

C47D, 2020-08-13:
Hi @huming2207,

I didn't knew the IL3820 controller supported cascaded controllers, but I think it should be up to the user to handle this.

What improvements on monochrome displays o you have in mind?

Carlos

huming2207, 2020-08-13:
Hi @C47D 

Sorry for the late reply. 

I was thinking about the ghost marks and those panels with 2+ controllers. But I think maybe it's enough for using the rounder_cb and set_px_cb to achieve the monochrome/greyscale conversions anyway. But it would be better if LVGL can provide a rotation function. I know this can be also done by some hacks in set_px_cb, but it may be a bit repetitive if we later encounter another a few panels that can't do hardware rotation properly. 

Jackson

C47D, 2020-08-14:
Hi Jason,

I'm going to merge #179, this is the first part of fixing the IL3820 driver issues, I removed the portrait inverted and landscape inverted options when selecting this controller.
Next week I will add the predefined display configuration for it.

If you want you can open an issue on LVGL main repository with the request you did above and see what is possible to do.

huming2207, 2020-08-15:
@C47D Yep I will leave a feature request on main LVGL repo later. Again thanks for your efforts in regards to this issue!

C47D, 2020-08-15:
No problem @huming2207, always happy to help, sorry for taking so long to get it working. I will close the issue.

closed 2020-08-15 by C47D