lvgl/lv_esp_idf #154: lv_label_set_text_fmt not working

Cikkcakk, 2020-06-10:
lv_label_set_text_fmt cause crash

every time when I update submodule I have to change
lv_conf.h > #define LV_SPRINTF_CUSTOM   0 to 1
to work

how to avoid this ?

We use GitHub issues for development related discussions.
Please use the [forum](https://forum.littlevgl.com/) to ask questions.

**Describe the issue**

**Code to reproduce the issue**

**Expected Results**

**Actual Results**

**ESP32 Chip version**

**ESP-IDF version**

**Development kit used**

**Development machine OS**

**Compilation warnings/errors (if available)**

**If possible, copy the compilation log into a file and attach it here**

Cikkcakk, 2020-06-10:
I should be a config property

C47D, 2020-06-10:
Lvgl configuration via menuconfig isn't complete, I will update it once I get some free time.
Are there any other options you change regularly?

closed 2020-07-05 by C47D