lvgl/lv_esp_idf #163: Error at linking with cmake: undefined reference to `disp_spi_send_xxx`

tobiwan42, 2020-07-03:
I followed the installation instructions from the readme and installed your projekt as submodule for my ESP-IDF Projekt.
When i compile with cmake a got following error at linking:

```
/home/tobiaskuch/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/lvgl_tft/liblvgl_tft.a(ili9341.c.obj):(.literal.ili9341_send_data+0x4): undefined reference to `disp_spi_send_data'
/home/tobiaskuch/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/lvgl_tft/liblvgl_tft.a(ili9341.c.obj):(.literal.ili9341_flush+0x0): undefined reference to `disp_spi_send_colors'
/home/tobiaskuch/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/lvgl_tft/liblvgl_tft.a(ili9341.c.obj): in function `ili9341_send_data':
/home/tobiaskuch/esp/projects/shop_device/components/lv_port_esp32/components/lvgl_esp32_drivers/lvgl_tft/ili9341.c:206: undefined reference to `disp_spi_send_data'
/home/tobiaskuch/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/lvgl_tft/liblvgl_tft.a(ili9341.c.obj): in function `ili9341_send_cmd':
/home/tobiaskuch/esp/projects/shop_device/components/lv_port_esp32/components/lvgl_esp32_drivers/lvgl_tft/ili9341.c:199: undefined reference to `disp_spi_send_data'
/home/tobiaskuch/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/lvgl_tft/liblvgl_tft.a(ili9341.c.obj): in function `ili9341_flush':
/home/tobiaskuch/esp/projects/shop_device/components/lv_port_esp32/components/lvgl_esp32_drivers/lvgl_tft/ili9341.c:157: undefined reference to `disp_spi_send_colors'
collect2: error: ld returned 1 exit status
CMakeFiles/shop_device.elf.dir/build.make:368: recipe for target 'shop_device.elf' failed
make[2]: *** [shop_device.elf] Error 1
CMakeFiles/Makefile2:604: recipe for target 'CMakeFiles/shop_device.elf.dir/all' failed
make[1]: *** [CMakeFiles/shop_device.elf.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2
make failed with exit code 2
```

I also changed the lv_port_esp32 cmake file like in the readme told.

```
cmake_minimum_required(VERSION 3.5)

#include($ENV{IDF_PATH}/tools/cmake/project.cmake)
list(APPEND EXTRA_COMPONENT_DIRS components/lvgl_esp32_drivers components/lvgl_esp32_drivers/lvgl_touch components/lvgl_esp32_drivers/lvgl_tft)

if (NOT DEFINED PROJECT_NAME)
	project(lvgl-demo)
endif (NOT DEFINED PROJECT_NAME)
```

This is my project root cmake.txt file

```
cmake_minimum_required(VERSION 3.5)

include($ENV{IDF_PATH}/tools/cmake/project.cmake)
set(EXTRA_COMPONENT_DIRS components/lv_port_esp32/components/lv_examples components/lv_port_esp32/components/lvgl components/lv_port_esp32/components/lvgl_esp32_drivers/lvgl_tft components/lv_port_esp32/components/lvgl_esp32_drivers/lvgl_touch components/lv_port_esp32/components/lvgl_esp32_drivers)
project(shop_device)
```

**ESP-IDF version**
release/v4.0

**Development kit used**
ESP_WROVER-KIT v4.1

**Development machine OS**
ubuntu18.04

**Complete Build log**
[build-log.log](https://github.com/lvgl/lv_port_esp32/files/4868418/build-log.log)

C47D, 2020-07-05:
Hi,

Thanks for the information, I was not able to replicate it, did you also did the `git submodule update --init --recursive` step?

Also, we are working on create a repo with all the lvgl_esp32_drivers so it will no longer necessary to add this whole repo as component on user projects, see #157

C47D, 2020-07-12:
@tobiwan42 Are you still having this issue?

tobiwan42, 2020-07-16:
Thanks for the quick answer.

I tried `git submodule update --init --recursiv`  but it didn't helped. I'm still having the same error at linking.

C47D, 2020-07-18:
I will try to replicate the issue again later today.

nikola-winmaker, 2020-07-24:
@C47D Just set compiler optimisation to -Os and you'll see the issue. I understand those functions were put in order to inline it in acutal driver later on... but they get optimised away, as a work around @tobiwan42 just remove inline from those functions in disp_spi.h

C47D, 2020-07-26:
Thanks for pointing out the issue @nikola-winmaker, maybe we shouldn't inline those functions to avoid the issue, let's wait on @tobiwan42 feedback so we can set up a pull request for it :)

tobiwan42, 2020-07-27:
Thanks for replicating the issue. I removed inline and moved the function body to `disp_spi.c`. Now it's compiling also in -Os optimization. Thanks for your help.

C47D, 2020-07-27:
I will update the repo with that change later today.

nikola-winmaker, 2020-07-27:
@C47D We should think about this.. removing inline will solve the issue for sure, but it will add function calls to execution which will slow down things, as I suppose this functions are called often. You need to be sure that compiler will inline those functions on -0s. You can check that in asm file produced.

C47D, 2020-07-28:
Hi @nikola-winmaker and @tobiwan42,

I was able to reproduce the issue when setting the optimization to -Os, but the error I got is, in the ST7789.c file:

```
/home/user/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/lvgl_tft/liblvgl_tft.a(st7789.c.obj):(.literal.st7789_send_cmd+0x0): undefined reference to `disp_spi_send_data'
/home/user/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/lvgl_tft/liblvgl_tft.a(st7789.c.obj):(.literal.st7789_flush+0x0): undefined reference to `disp_spi_send_colors'
/home/user/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/lvgl_tft/liblvgl_tft.a(st7789.c.obj): in function `st7789_send_cmd':
/home/user/lv_port_esp32/build/../components/lvgl_esp32_drivers/lvgl_tft/st7789.c:192: undefined reference to `disp_spi_send_data'
/home/user/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/lvgl_tft/liblvgl_tft.a(st7789.c.obj): in function `st7789_send_data':
/home/user/lv_port_esp32/build/../components/lvgl_esp32_drivers/lvgl_tft/st7789.c:199: undefined reference to `disp_spi_send_data'
/home/user/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/lvgl_tft/liblvgl_tft.a(st7789.c.obj): in function `st7789_send_color':
/home/user/lv_port_esp32/build/../components/lvgl_esp32_drivers/lvgl_tft/st7789.c:204: undefined reference to `disp_spi_send_colors'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

After some googling I found this stack overflow answer: https://stackoverflow.com/a/9399539

> The best place to put inline function is in a header file, and declare them static inline. This removes any need for an external definition, so it resolves the linker problem. However, this causes the compiler to emit the code for the function in every compilation unit that uses it, so could result in code bloat. But since the function is inline, it is probably small anyway, so this usually isn't a problem.

So in `disp_spi.h` I changed the following (added static to both inline functions):
```diff
diff --git a/components/lvgl_esp32_drivers/lvgl_tft/disp_spi.h b/components/lvgl_esp32_drivers/lvgl_tft/disp_spi.h
index 1a158a9..64da1c4 100644
--- a/components/lvgl_esp32_drivers/lvgl_tft/disp_spi.h
+++ b/components/lvgl_esp32_drivers/lvgl_tft/disp_spi.h
@@ -64,11 +64,11 @@ void disp_wait_for_pending_transactions(void);
 void disp_spi_acquire(void);
 void disp_spi_release(void);

-inline void disp_spi_send_data(uint8_t *data, size_t length) {
+static inline void disp_spi_send_data(uint8_t *data, size_t length) {
     disp_spi_transaction(data, length, DISP_SPI_SEND_POLLING, NULL, 0);
 }

-inline void disp_spi_send_colors(uint8_t *data, size_t length) {
+static inline void disp_spi_send_colors(uint8_t *data, size_t length) {
     disp_spi_transaction(data, length,
         DISP_SPI_SEND_QUEUED | DISP_SPI_SIGNAL_FLUSH,
         NULL, 0);
```

Do you think this is the most appropiate fix @nikola-winmaker ?

nikola-winmaker, 2020-07-28:
@C47D You could add static, which would declare scope of those functions to be part of every module it is included in. As you said this could bloat the code but as these functions are small enaugh I would say it is ok.
But the clean solution would be IMHO, to declare these functions in header file and add implementation in some “common” *.c file... Either way it’s fine with me 👍
Keep up the good work @C47D

embeddedt, 2020-07-28:
> But the clean solution would be IMHO, to declare these functions in header file and add implementation in some “common” *.c file...

In this case, the functions wouldn't be inlineable except within the C file, since the compiler would need access to the implementation in order to inline them. I think the `static inline` approach is the best; it's also the same practice we use in the core library.

nikola-winmaker, 2020-07-28:
@embeddedt It’s always a trade off between clean and the best (ODR - one definition rule)... static inline will work as I have stated before 👍

PR #173 (2020-07-29): Declaring inline functions as static avoid them being optimized when using -Os
Closes #163
closed 2020-07-29 by C47D