lvgl/lv_esp_idf #166: ESP-IDF v3.3 compatibility

EtienneMdv, 2020-07-16:
Hello,

I was wondering if there is any chance that I could use the lvgl library using esp-idf version 3.3.

I have encountered similar problems that those describe in this issue : [(https://github.com/lvgl/lv_port_esp32/issues/133)].

I can't switch to the latest esp-idf unfortunately...  However I was hopeful to make it work after reading the following note :

> This project tries to be compatible with both the ESP-IDF v3.x and v4.0, but using the v4.0 is recommended. Instructions here are given for the v4.x toolchain using idf.py, but it is easy to translate to make. For example instead of running idf.py menuconfig, just run make menuconfig.

Best,
Etienne

ma89vik, 2020-07-18:
The issue is probably that the project uses the CMake functions that were introduced in 4.0, e.g. `idf_component_register`. So you wont be able to compile it with CMake+3.3 without changing anything. 

You could try using make instead and see if that works.

C47D, 2020-07-18:
Hi, as far as I know using make should solve this issue. Do any of yall know if there's a way to know the idf version that we could use on the CMakeLists files?

ma89vik, 2020-07-18:
You mean a way get the current IDF version as a variable in the cmake file?

Another way to solve this would be to write the CMake file the way they were written in 3.3, which should still be compatible with newer IDFs. That way it should compile OK with all versions.

C47D, 2020-07-18:
> You mean a way get the current IDF version as a variable in the cmake file?

Yes, that. I'm taking a look at the documentation. Maybe we can use IDF_VER as described here: [Checking the Current Version](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/versions.html?highlight=idf%20version#checking-the-current-version).

EtienneMdv, 2020-07-22:
Hi guys, thanks a lot for your answers! Using make instead of CMake works fine for me.

C47D, 2020-07-26:
@EtienneMdv Thanks for letting us know, I will update the README to let users know to use make when working with the ESP-IDF v3.3 😄

closed 2020-07-26 by C47D