lvgl/lv_esp_idf #168: .platformio with espidf framework compability

DrZlo13, 2020-07-20:
Hello, i noticed that [commit](https://github.com/lvgl/lv_port_esp32/commit/a42c982c912e2f93868ebec21306e7a8c912d2f1) says this project have compability with .platformio build system, but i dont have any luck to compile that. 

For now i have bunch of "undefined reference to `lv_something'" and i don't know how to fix any of those things.

C47D, 2020-07-26:
Hi @DrZlo13, I don't personally use Platformio but hopefully @davidjade can help us out. On the meantime, could you attach a text file with the output you are getting to check what functions are not being defined?

DrZlo13, 2020-07-27:
```
Linking .pio\build\lolin32\firmware.elf
c:/users/dr_zlo/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\lolin32\esp-idf\src\main.cpp.o:(.literal._Z6lvtestPv+0x4): undefined reference to 'lv_disp_get_default'
c:/users/dr_zlo/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\lolin32\esp-idf\src\main.cpp.o:(.literal._Z6lvtestPv+0x8): undefined reference to 'lv_disp_get_scr_act'
c:/users/dr_zlo/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\lolin32\esp-idf\src\main.cpp.o:(.literal._Z6lvtestPv+0xc): undefined reference to 'lv_btn_create'
c:/users/dr_zlo/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\lolin32\esp-idf\src\main.cpp.o:(.literal._Z6lvtestPv+0x10): undefined reference to 'lv_obj_set_pos'
c:/users/dr_zlo/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\lolin32\esp-idf\src\main.cpp.o:(.literal._Z6lvtestPv+0x14): undefined reference to 'lv_obj_set_size'
c:/users/dr_zlo/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\lolin32\esp-idf\src\main.cpp.o:(.literal._Z6lvtestPv+0x18): undefined reference to 'lv_label_create'
c:/users/dr_zlo/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\lolin32\esp-idf\src\main.cpp.o:(.literal._Z6lvtestPv+0x1c): undefined reference to 'lv_label_set_text'
c:/users/dr_zlo/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\lolin32\esp-idf\src\main.cpp.o: in function 'lv_scr_act':
E:\work\projects\esp32-platformio/src\../components/lv_port_esp32/components/lvgl/lvgl/src/lv_core/lv_disp.h:99: undefined reference to 'lv_disp_get_default'
c:/users/dr_zlo/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: E:\work\projects\esp32-platformio/src\../components/lv_port_esp32/components/lvgl/lvgl/src/lv_core/lv_disp.h:99: undefined reference to 'lv_disp_get_scr_act'
c:/users/dr_zlo/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\lolin32\esp-idf\src\main.cpp.o: in function 'lvtest(void*)':
E:\work\projects\esp32-platformio/src/main.cpp:13: undefined reference to 'lv_btn_create'
c:/users/dr_zlo/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: E:\work\projects\esp32-platformio/src/main.cpp:14: undefined reference to 'lv_obj_set_pos'
c:/users/dr_zlo/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: E:\work\projects\esp32-platformio/src/main.cpp:15: undefined reference to 'lv_obj_set_size'
c:/users/dr_zlo/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: E:\work\projects\esp32-platformio/src/main.cpp:19: undefined reference to 'lv_label_create'
c:/users/dr_zlo/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: E:\work\projects\esp32-platformio/src/main.cpp:20: undefined reference to 'lv_label_set_text'
```

kisvegabor, 2020-07-27:
Hi,

I've just tried it with our [lv_platformio](https://github.com/lvgl/lv_platformio) project and I added these lines to plaformio.ini:
```ini
[env:esp32]
platform = espressif32
framework = espidf
board = esp-wrover-kit
```
Changed the efault environment to `default_envs = esp32`

And modified main.c like this:
```c
#include "lvgl.h"
// #include "driver.h"

#include "demo.h"

int app_main(void)
{
	lv_init();

	// hw_init();

	demo_create();

	// hw_loop();
  return 0;
}
```

And it compiled but I can't test it on hardware now. :slightly_frowning_face:

C47D, 2020-08-07:
Any feedback @DrZlo13?

C47D, 2020-08-29:
Will add the steps @kisvegabor took to make it work with platformio into the README. Then I will close this issue as there wasn't any feedback.

kisvegabor, 2020-08-31:
To the README of this repo or lv_platformio?

C47D, 2020-09-01:
I was thinking on this repo README, what do you think?

kisvegabor, 2020-09-01:
I'm ok with adding this to the README here.
We can say to use `lv_platformio` repos and modify its platforimio.ini file.

PR #194 (2020-09-03): Update README with instructions for platformio
I don't use platformio to test your instructions @kisvegabor 😅 

Closes #168
closed 2020-09-03 by C47D