lvgl/lv_esp_idf #169: variable "may be used uninitialized in this function"

KammutierSpule, 2020-07-24:
When selecting ILI9488

On function sh1107_set_px_cb
CONFIG_LVGL_DISPLAY_ORIENTATION_LANDSCAPE
or
CONFIG_LVGL_DISPLAY_ORIENTATION_PORTRAIT

is not defined, so 
uint16_t byte_index;
    uint8_t  bit_index;
are not initialized.

C47D, 2020-07-26:
Hi @KammutierSpule, it shouldn't raise a warning because only the source file for the chosen display is included into the build. What commit of the repo are you working on?

Anyways, I will initialize those variables to 0 if necessary.

KammutierSpule, 2020-07-26:
sorry, you're right.. it was some issues with updating to the latest repo version.

closed 2020-07-26 by KammutierSpule

C47D, 2020-07-26:
No problem, have fun and let us know if you have more issues :)