lvgl/lv_esp_idf #174: Rotate display using the lv_disp_drv_t rotated field.

C47D, 2020-07-30:
Currently we rotate the display by swapping `LV_HOR_RES_MAX` and `LV_VER_RES_MAX` in `lv_conf.h`, we should use the `rotated` field in `lv_disp_drv_t` instead.

Is this correct @kisvegabor, @embeddedt ?

embeddedt, 2020-07-30:
`rotated` is a bit of a strange flag at the moment - I don't completely understand how it works myself. I think the actual rotation still ends up being done by the display driver, not by LVGL itself, so I don't know if using the `rotated` flag really changes that much. I guess it might matter for rotating the screen on the fly.

C47D, 2020-07-30:
I used it with the ILI9341 display I have, setting it to 1 rotates the display without any change on the driver. Do you mean that?

kisvegabor, 2020-08-02:
`rotate` only swaps the horizontal and vertical resolution of the display now and the real rotation really should be done in the display driver. In the future, we can have a `set_orientation_cb` in the display driver which can be called according to the current rotation settings.

C47D, 2020-08-02:
> rotate only swaps the horizontal and vertical resolution of the display now and the real rotation really should be done in the display driver.

This is the silly workaround I did on the `lv_config.h` file, swapping the resolutions (I didn't knew the `rotate` option back then). Do you think is sane to remove the workaround and use `rotate` instead?

kisvegabor, 2020-08-04:
> Do you think is sane to remove the workaround and use rotate instead?

Yes, that's why we have rotate. :slightly_smiling_face:  (although it's very limited now).

C47D, 2020-08-26:
Not stalled

C47D, 2020-09-17:
on the backburner

closed 2020-10-17 by stale[bot]