lvgl/lv_esp_idf #177: Menuconfig issue

kisvegabor, 2020-08-06:
**Describe the issue**
I got this error on `make menuconfig`
```
Toolchain path: /home/kisvegabor/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a
Compiler version: 5.2.0
MENUCONFIG
/home/kisvegabor/esp/esp-idf/Kconfig:206: can't open file "components/lvgl_esp32_drivers/lvgl_touch/Kconfig"
make: *** No rule to make target 'menuconfig', needed by '/home/kisvegabor/projects/littlevgl/lv_port_esp32/build/include/config/auto.conf'.  Stop.
```

If I remove the use of touch's Kconfig from the Makefile the menuconfig opens but there is no TFT config menu.

**Code to reproduce the issue**
Using the latest ESP-IDF 
```
cd lv_port_esp32
make menuconfig
```

**Expected Results**
Open menuconfig

**ESP-IDF version**
Latest, just updated from GitHub

**Development kit used**

-

**Development machine OS**
Linux Mint 19

C47D, 2020-08-06:
Can you try using `idf.py menuconfig` to see if you get the same error? I haven't tested using IDF 4.x with make, will take a look at it.

C47D, 2020-08-07:
Just tried using `make menuconfig` and got it working (I'm using WSL Ubuntu). Are you doing `. export.sh` on the esp-idf directory? Do you get any more logging?

kisvegabor, 2020-08-10:
There is no `export.sh` in the es-idf directory for me.
```sh
$ ls -1
add_path.sh
CMakeLists.txt
components
CONTRIBUTING.rst
docs
examples
Kconfig
LICENSE
make
README.md
requirements.txt
tools
```

C47D, 2020-08-10:
Are you using this repo for the esp-idf? https://github.com/espressif/esp-idf

kisvegabor, 2020-08-11:
Hmm, it seems I screwed up something when pulled the esp-idf. It wasn't updated properly and I didn't notice it. 

Now I have export.sh, but a few dependencies are still missing for esp-idf.

It should work from here. Sorry, for the noise.

closed 2020-08-11 by kisvegabor

C47D, 2020-08-12:
To get those dependencies I think you can do `install.sh`, I haven't updated my toolchain in some time but that's what I remember.

Don't worry about it, always happy to try help :)

kisvegabor, 2020-08-12:
It works like a charm after `install.sh`.

Thank you very much! ^^