lvgl/lv_esp_idf #178: Some LVGL configuration options are not available on menuconfig

C47D, 2020-08-06:
@kisvegabor Some LVGL configuration options are not available on the menuconfig, such as:

```c
/* Type of coordinates. Should be `int16_t` (or `int32_t` for extreme cases) */
typedef int16_t lv_coord_t;
```

and

```c
/*=====================
 *  Compiler settings
 *====================*/
/* Define a custom attribute to `lv_tick_inc` function */
#define LV_ATTRIBUTE_TICK_INC

/* Define a custom attribute to `lv_task_handler` function */
#define LV_ATTRIBUTE_TASK_HANDLER

/* Define a custom attribute to `lv_disp_flush_ready` function */
#define LV_ATTRIBUTE_FLUSH_READY

/* With size optimization (-Os) the compiler might not align data to
 * 4 or 8 byte boundary. This alignment will be explicitly applied where needed.
 * E.g. __attribute__((aligned(4))) */
#define LV_ATTRIBUTE_MEM_ALIGN

/* Attribute to mark large constant arrays for example
 * font's bitmaps */
#define LV_ATTRIBUTE_LARGE_CONST

/* Prefix performance critical functions to place them into a faster memory (e.g RAM)
 * Uses 15-20 kB extra memory */
#define LV_ATTRIBUTE_FAST_MEM
```

This is because the option types available on Kconfig.

For the LV_ATTRIBUTE_x options we can do something like this, so the users can define them somewhere else:
```
/*=====================
 *  Compiler settings
 *====================*/
/* Define a custom attribute to `lv_tick_inc` function */
#ifndef LV_ATTRIBUTE_TICK_INC
#define LV_ATTRIBUTE_TICK_INC
#endif
```

kisvegabor, 2020-08-08:
Thank you for opening this issue!

IMO, we can leave minimal lv_conf.h (only with the non-Kconfig compatible options) to let the user set those there.  

By the way, what should happen if we have an option defined `lv_conf.h` and Kconfig too? Which one should have a higher priority?

C47D, 2020-08-08:
Hi, 

> By the way, what should happen if we have an option defined lv_conf.h and Kconfig too? Which one should have a higher priority? (It should be decided in lv_conf_internal.h)

I don't follow you there, the symbols defined in Kconfig are used in the lv_conf.h like so:
```
/* Default display refresh period.
 * Can be changed in the display driver (`lv_disp_drv_t`).*/
#define LV_DISP_DEF_REFR_PERIOD CONFIG_LVGL_DISP_DEF_REFR_PERIOD   /*[ms]*/
```

`LVGL_DISP_DEF_REFR_PERIOD` is defined in the Kconfig file (`CONFIG_ ` is appended before compiling), the value of it is given when you configure the menuconfig. You can see the value of `CONFIG_LVGL_DISP_DEF_REFR_PERIOD` in `build/config/sdkconfig.h`.

I don't know what do you mean with options defined on both `lv_conf.h` and `Kconfig` at the same time, care to expand?

kisvegabor, 2020-08-10:
In a [previous issue](https://github.com/lvgl/lv_port_esp32/pull/164#issuecomment-657418702), we talked about how `lv_conf.h`, `Kconfig` and `lv_conf_inernal.h` can live together. 

If we slightly modify that idea by always include `lv_conf.h` but give Kconfig a higher priority, user can simply define non-Kconfig compatible options in lv_conf.h.

`#warning` can be added if an `lv_conf.h` option is overwritten by `Kconfig`.

Or the opposite, give `lv_conf.h` higher priority? I don't know. :sweat_smile:

closed 2020-09-07 by stale[bot]