lvgl/lv_esp_idf #183: Can TTGO Camera Plus development board be added?

HoGC, 2020-08-16:
I tried to configure, but no success
[image]
[image]
This is a ttgo project: https://github.com/lewisxhe/esp32-camera-screen

C47D, 2020-08-16:
Hi @HoGC, what do you mean with no success? The display didn't displayed anything? It displays garbage?

Can you paste here the output of `idf.py monitor`?

HoGC, 2020-08-16:
The display didn't displayed anything,  But the backlight is on.
 I can use this drive to display: https://github.com/Xinyuan-LilyGO/TTGO-T-Display
LOG
```
I (0) cpu_start: App cpu up.
I (335) heap_init: Initializing. RAM available for dynamic allocation:
I (342) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (348) heap_init: At 3FFC4950 len 0001B6B0 (109 KiB): DRAM
I (354) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (361) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (367) heap_init: At 4008B9AC len 00014654 (81 KiB): IRAM
I (373) cpu_start: Pro cpu start user code
I (392) spi_flash: detected chip: generic
I (392) spi_flash: flash io: dio
I (393) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (20) lvgl_helpers: Display hor size: 240, ver size: 240
I (20) lvgl_helpers: Display buffer size: 9600
I (20) lvgl_helpers: Initializing SPI master for display
I (30) lvgl_helpers: Configuring SPI host HSPI_HOST (1)
I (30) lvgl_helpers: MISO pin: -1, MOSI pin: 19, SCLK pin: 21
I (40) lvgl_helpers: Max transfer size: 19200 (bytes)
I (50) lvgl_helpers: Initializing SPI bus...
I (50) disp_spi: Adding SPI device
I (50) disp_spi: Clock speed: 20000000Hz, mode: 2, CS pin: 12
ST7789 initialization.
I (470) st7789: Display orientation: LANDSCAPE
I (470) st7789: 0x36 command value: 0x60
```

C47D, 2020-08-16:
I got a ST7789 display, I will try to see if anything is wrong, and also will check the repository you use to see if there's any difference with the initialization sequence of the display controller.

HoGC, 2020-08-17:
I tried the TTGO-T-Display development board to show it successfully, but TTGO Camera Plus display didn't displayed anything，ST7789 display。

C47D, 2020-08-17:
I checked my st7789 display and it looks good.

HoGC, 2020-08-20:
I solved the problem，Add  `gpio_pad_select_gpio` when initializing GPIO
st7789.c (84)
```
    //Initialize non-SPI GPIOs
    gpio_pad_select_gpio(ST7789_DC);
    gpio_set_direction(ST7789_DC, GPIO_MODE_OUTPUT);
    gpio_pad_select_gpio(ST7789_RST);
    gpio_set_direction(ST7789_RST, GPIO_MODE_OUTPUT);
    
#if ST7789_ENABLE_BACKLIGHT_CONTROL
    gpio_pad_select_gpio(ST7789_BCKL);
    gpio_set_direction(ST7789_BCKL, GPIO_MODE_OUTPUT);
#endif
```

C47D, 2020-08-20:
Thanks for letting us know the fix, I will add it later today.

C47D, 2020-08-27:
@HoGC I've added the configuration for the TTGO Camera Plus board on the `add_ttgo_camera_plus` branch, can you test it?
[image]

HoGC, 2020-08-27:
OK, I'll test it later

HoGC, 2020-08-27:
@C47D Can be displayed normally
Thanks for added

C47D, 2020-08-27:
Thanks for testing it @HoGC, it will be merged into master.

closed 2020-08-28 by C47D