lvgl/lv_esp_idf #185: Update LVGL version?

BFFonseca, 2020-08-18:
LVGL is now on 7.3.1 already.
I see that this and other repos are still using 7.0.2 dev.

Is there any reason for this?
Can I safely update my LVGL locally to the latest version?

embeddedt, 2020-08-18:
You can safely update to newer minor releases. There shouldn't be any API breakage - the only things that may have changed are source files being added/removed.

The delay in updating is caused by the fact that the projects need to be updated manually. This generally only happens if someone sends a pull request to update the project, a core maintainer has access to the hardware to test the updated project, or a major release occurs.

closed 2020-08-18 by BFFonseca

kisvegabor, 2020-08-20:
For the next minor release, I'm planning to make updating project automatically like this:
1. A new LVGL version is released by a script
2. The same script clones the project to update
3. Each project has an update script that recreates the `conf.h` files and changes the defines important for itself and leave the others.

C47D, 2020-08-20:
I think what we need to do is use lvgl as a esp component, some people had issues when we updated from v6 to v7.

Zephyr and Nuttx people have their own kconfig system, I consider we should not add an Kconfig file directly into lvgl main repo.

@kisvegabor is possible to have a audio meeting or something?

embeddedt, 2020-08-20:
> has an update script that recreates the conf.h files and changes the defines important for itself and leave the others.

Since we have `lv_conf_internal.h`, I think you could just make a stub `lv_conf.h` for each project that only defines what needs to be overridden from the default options. It might be simpler than replacing lines from the template.

kisvegabor, 2020-08-21:
> Since we have lv_conf_internal.h, I think you could just make a stub lv_conf.h for each project that only defines what needs to be overridden from the default options. It might be simpler than replacing lines from the template.

It's also possible but might be difficult for people to change the config later.

kisvegabor, 2020-08-21:
> @kisvegabor is possible to have a audio meeting or something?

Sure, what about Tuesday?
@embeddedt Would you like to join?

kisvegabor, 2020-08-24:
Hi,

I've just sent the invitations.

> we should prepare an outline to plan the meeting.

The main topic are  "LVGL as ESP component" and Kconfig questions, right?

C47D, 2020-08-24:
> The main topic are "LVGL as ESP component" and Kconfig questions, right?

Yes, I would like to also talk a bit about the work to be done on `lv_drivers`.