lvgl/lv_esp_idf #188: Update LVGL submodule

C47D, 2020-08-25:
Currently we use LVGL 7.0.x, it would be great to update to LVGL 7.3.x to have the latest and greatest features of the project.

kisvegabor, 2020-08-27:
Let me know if you need assistance from my side!

C47D, 2020-08-27:
Will do, I've been a bit busy with my day job but I will start the update later today.

C47D, 2020-09-03:
@kisvegabor both `lvgl` and `lv_examples` are now on the latest commit of the `release/v7` branch.

Still need to update the config and Kconfig files.

kisvegabor, 2020-09-04:
Awesome!

What do you think, let's create Kconfig manually and think about auto-generating later?

C47D, 2020-09-04:
Yes, I was thinking that, I can re use the Kconfig files that we already have.

C47D, 2020-09-27:
Configuration files are still needed to update.

closed 2020-10-24 by C47D