lvgl/lv_esp_idf #197: Update component.mk file

C47D, 2020-09-08:
Maybe we can check for the `ESP_PLATFORM` variable.

kisvegabor, 2020-09-08:
It seems a simple [ifdef](https://www.gnu.org/software/make/manual/html_node/Conditional-Syntax.html) can work:
```c
ifdef $(foo)
frobozz = yes
endif
```

C47D, 2020-10-01:
Seems like `ESP_PLATFORM` is not defined when using make to bring up the menuconfig or cleaning the project, I'm getting the following warning messages:
``
/home/user/lv_port_esp32/components/lvgl_esp32_drivers/lvgl_touch/component.mk:1: warning: undefined variable 'ESP_PLATFORM'
/home/user/lv_port_esp32/components/lvgl_esp32_drivers/lvgl_tft/component.mk:1: warning: undefined variable 'ESP_PLATFORM'
/home/user/lv_port_esp32/components/lvgl_esp32_drivers/component.mk:1: warning: undefined variable 'ESP_PLATFORM'
/home/user/lv_port_esp32/components/lvgl/component.mk:1: warning: undefined variable 'ESP_PLATFORM'
/home/user/lv_port_esp32/components/lv_examples/component.mk:1: warning: undefined variable 'ESP_PLATFORM'
```

C47D, 2020-10-24:
Needs further investigation

closed 2020-11-02 by stale[bot]