lvgl/lv_esp_idf #199: Feature request: EVE3 BT81X commands

YTvW, 2020-09-18:
Would it be possible to add EVE3 specific commands to the driver. Since the EVE3 has on board RAM that is not accessible with the current commands.

C47D, 2020-09-18:
Hi,

The EVE driver was contributed by a user, as I don't have that device. I will try to take a look at it and see what can we do to add the commands you've requested.

If you have that device it would be great if you could test the changes. Is there any particular command that you want to add?

davidjade, 2020-09-23:
I ported the EVE2 driver to LVGL but I don't have an EVE3. What differences does it have to access this other RAM? Is it not like the EVE2 where everything is memory mapped?

davidjade, 2020-09-25:
@YTvW  One other question I might add: what in the EVE3 command set would benefit the LVGL driver? Access to more RAM won't actually help LVGL, as LVGL treats the FT81x as a dumb bitmap display device. 

If it's not apparent already, the existing esp32 LVGL driver for FT81x displays really does not use any part of the EVE commands set to do anything. It simply sets EVE up to display one full screen bitmap that then allows LVGL to write bits directly into it, which then show up on the screen because the underlying bitmap data has changed. As far as LVGL is concerned, the FT81x is just a chunk of memory that it writes bitmap data into. 

Yes the FT81x driver has code for all the other EVE2 commands but they are not used at all by LVGL - I left them in case someone wants to try and create a hybrid GUI, i.e. use a mix of native EVE graphics and LVGL graphics on a partitioned or split screen. Any sort of code like that would need to do all the lower-level EVE commands separate from using LVGL (and probably also make significant changes to the FT81x LVGL driver too).

C47D, 2020-10-21:
Reopen if there's some feedback.

closed 2020-10-21 by C47D