lvgl/lv_esp_idf #200: Identify ESP32 specific configurations in lv_conf

C47D, 2020-09-23:
Some configuration options require ESP32 specific values, such as
```c
#define LV_ATTRIBUTE_FAST_MEM IRAM_ATTR
```

Current `lv_conf ` file doesn't include them, so we need to identify them and update `lv_conf` file accordingly.

kisvegabor, 2020-09-23:
I think it should be added to `lv_conf_internal.h` when [LV_CONF_SKIP](https://github.com/lvgl/lvgl/blob/6b06ac4b726f5e90881607387593e402202fca37/src/lv_conf_internal.h#L1917) and `ESP_PLATFORM` are defined.

It solves to add the mandatory ESP configs if lv_conf.h is not present.

C47D, 2020-09-23:
Didn't knew there is a `LV_CONF_SKIP` (or `CONFIG_LV_CONF_SKIP`) option, we should define it in the LVGL Kconfig and set it to y. I can do that later if you want.

kisvegabor, 2020-09-23:
> Didn't knew there is a LV_CONF_SKIP (or CONFIG_LV_CONF_SKIP) option, we should define it in the LVGL Kconfig and set it to y. I can do that later if you want.

Yes, this way the user still can chose to add lv_conf.h.

> We would also need to add something like this in the top of the lv_conf_internal:

Ok, no problem.

closed 2020-10-15 by C47D