lvgl/lv_esp_idf #201: Adding support for Dual and Quad SPI modes

davidjade, 2020-09-24:
I've been working on extending the current driver to support both Dual and Quad SPI and I've got it working with the FT81x but there are some serious caveats - the main one being that reading more than 1 byte of data at a time does not appear to working in HALF_DUPLEX mode using a mix of address and dummy bits as required by the FT81x. At this point I've spent a lot of time trying to understand what is going on and from looking at logic analyzer traces and diving deep into the ESP32 hal and spi_master, I am at a loss to what it happening but I can work around it. I've raise the issue with Espressif and I'm hoping someone has some answers because the driver seems broken in this "half-duplex read with address and dummy bits" scenario that the FT81x requires. Here's all the gory details I sent to Espressif: https://esp32.com/viewtopic.php?f=14&t=17436&sid=09557265fac47f98e92f722e94c82774

The good new is that writing data seems to always work reliably and I've got the FT81x humming along at 32Mhz in Dual mode on a breadboard no problem. Quad mode also works but I'm finding that line noise on a breadboard is a problem so I'm using Dual mostly.

To make it work for reading though, I just read single bytes at time and re-constitute the uint16_t or uint32_t that are needed with the FT81x. It works and isn't a huge penalty since I rarely read more the 4 bytes at a time. I can also isolate this work around to the FT81x driver itself to keep the mess out of the disp_spi code with one exception - there is a weirdness about specifying dummy bits (which I suspect is a documentation error - they appear to be dummy clocks, not dummy bits when I look at the logic traces).

So the question is, given that there are some issues that may be found when reading data in Dual and Quad modes, should my changes make it back to the lv_port_esp32 at this point? On the one hand only the FT81x is special in that there is no separate touch driver and all the other display types with separate touch drivers probably won't be affected (as I have not touched them). But I also don't have other displays that support Dual or Quad modes to test on, nor do I have the other displays that are already supported to test that I haven't broken anything.

C47D, 2020-09-25:
Hi,

Thanks for the link, I find very useful to read others experiences.

> So the question is, given that there are some issues that may be found when reading data in Dual and Quad modes, should my changes make it back to the lv_port_esp32 at this point?

Actually we are trying to contribute the display and touch controllers back to lv_drivers. There's a post on the forum about it, [here](https://forum.lvgl.io/t/merge-display-and-input-drivers-from-lv-port-esp32-to-lv-drivers/3021).

Do you have your work available somewhere so we can take a look at it and maybe incorporate it before the "big" drivers change?

> But I also don't have other displays that support Dual or Quad modes to test on, nor do I have the other displays that are already supported to test that I haven't broken anything.

I'm trying to get a parallel driven display to test the new approach I'm working on, so the interface between the MCU and the display controller should not matter at all.

davidjade, 2020-09-25:
I saw mention of the esp32 driver migration but I have not read up on it. I will take a look.

I'm not quite sure what you mean by taking a look to incorporate before the "big" change. If you are suggesting that I can put in this Dual/Quad change now to the current lv_port_esp32 before the "big" change over to make it a separated lv_esp32_driver, then I can most likely do that as the changes to enable Dual and Quad mode really are quite modest due to the pre-work I set up the last time around. I do think it is best to get the Dual/Quad mode stuff working in the existing driver as it will likely help in understanding what is needed to re-engineer them into the next driver rewrite/decoupling.

So if you would like some smallish changes to fill out the necessary bits needed to enable Dual and Quad with the current esp32 SPI drivers, I can likely have it done in a week or less. Here's a rough outline of what will change:

1. disp_spi will need to do the right thing for the various DIO/QIO flags that are already defined. This will trickle down into the reading data code path but I think only the FT81x display even uses reading.
2. disp_spi_transaction() is going to need another flag (DISP_SPI_VARIABLE_DUMMY) and a parameter for the dummy bit count
2b. all the existing driver will take a small update to pass a zero for the dummy bit count since they don't use them
3. the FT81x driver will be updated to use DIO & QIO modes. I can put the work-arounds needed for the half-duplex mode issues I found in the FT81x driver so that it does not pollute the core disp_spi driver. This should keep the code in disp_spi clean.
4. New Kconfig settings to control SIO/DIO/QIO modes. Probably some additional FT81x specific kconfig as well.

The changes to disp_spi_transaction() won't affect existing drivers since they don't yet have code for using DIO/QIO mode. In thinking about this I really think it is best to make disp_spi just expose the basics of enabling DIO/QIO modes but leaving it up to the specialized drivers to use those capabilities as they see fit. I think it is likely that each display type will have different bus requirements, setup, timings, etc... so there isn't going to be a generic just flip the DIO/QIO mode possible to make it all work at a higher level. Some devices may be in DIO/QIO mode at start up and others, like the FT81x, start in Single SPI and then transition to DIO or QIO modes after some number of transaction (i.e. it has to be told to switch modes by sending commands). 

Doing it this way also causes the least amount of churn as each driver can individually enable DIO/QIO mode as someone later works out how to do it properly for each device type.

Let me know if you're willing to take a pull request for the current code and I'll work on getting it ready.

C47D, 2020-10-19:
Closed with #202

closed 2020-10-19 by C47D