lvgl/lv_esp_idf #203: Adding support for multiple queued DMA transactions

davidjade, 2020-09-28:
After adding the DIO/QIO SPI modes I have done some further experiments with enabling a queue of SPI in-flight transactions and the results so far are very promising. By using a queue of in-flight DMA transaction structures instead of having just one static in-flight transaction, the Esp32 DMA controller can be sent multiple SPI transactions to process instead of the current code, where it always waits for the previous one to be finished before sending the next. In the current code this is a bottleneck that slows down redraw speeds and can also lead to a jerky display especially while animating. This bottleneck also likely prevents LVGL from using both update buffers effectively as well (since it can block the update thread).

Where this really pays off though, is for drivers that have to send data line by line (because the hardware doesn't support 7an arbitrary rectangle). The FT81x is one such devices and unless the rectangle is full display width, it has to break up repaint region rectangle into individual SPI DMA requests for each line of the area in a fast loop. Enabling 20, 50 or even 200 SPI transactions to be in-flight at once has a very, very noticeable effect of the speed of the display. The difference is night and day in refresh rates especially when using animated controls, etc... It's the first time I have seen the FT81x animate while being consistently free of animation frame skipping. Here's a quick video of the before/after: https://youtu.be/HxO02oeiakw

Since I laid some of this groundwork a few months back this isn't looking to be too hard to add this feature. All the changes will be in disp_spi.c. This also means that if there are other drivers that also loop over or burst sending SPI data (which results in multiple DMA transactions), they will automatically benefit from queued DMA transactions. For instance, the FT81x driver needed no changes at all. I don't expect any existing drivers to break (or even know) either. 

At this point I have this working and I should have it ready to submit in a day or so, once I put it through some more testing, cleanup, etc... It builds on the Dual/Quad pull request I submitted earlier.

davidjade, 2020-09-30:
I've got this one pretty much ready to submit but there are maybe a few things to consider first:

1. This really, really boosts the performance of the FT81x driver but I also think it will boost the performance of the existing drivers for the il3820, jd79653a, and uc8151d. I say "think" because they all appear to send multiple SPI DMA transactions in bursts and this change reduces the latency between the bursts of individual SPI transfers. However I don't have any of those displays so I can't directly test them myself - this is just from me reviewing the other drivers.

2. The changes are all in disp_spi.c and are not very complicated but they do have the potential to send individual SPI transactions at faster rates as it removes the default send & wait-for-pending-results pattern that exists now between calls (but drivers can still wait for pending results as necessary - that has not changed). It should not really matter to the existing drivers and they likely won't even notice in a negative way. It also does not require any changes to any existing drivers. However, it potentially speeds up the data streaming to them and if they can't handle it then they might need to be adjusted. If a driver can't handle getting data at a fast burst rate then the best solution would likely be to simply sprinkle a few more disp_wait_for_pending_transactions() calls where needed to give the display controller time to process the data. But it may also be worth thinking about only enabling it once drivers are checked out to make sure they are working OK. I don't expect them to break but I also can't test them all myself as I don't have those other displays.

3. Right now, I have not done made it a per-driver opt-in -- it is enabled for all drivers. But I could switch that and only enable it for the FT81x where I know for certain it works (and works really, really well). This would be easy to do and I would likely do it through #defines and not kconfig. Each driver could be tested and then opted in once checked out. Or maybe it does make sense to opt them all in now and get some testing done by others who can test them.

4. Another thing is, while the code is fairly simple I sort of feel that maybe a few sentences somewhere about the queuing design and how it works may be warranted as there are maybe a few subtle things that future coders should think about when revising the core SPI code. The current code base doesn't have anything like that now in comments or otherwise, so I don't really know where it is preferred to put something like that (I lean towards wanting it in comments in disp_spi.c myself - a few short paragraphs). 

Lastly, I know other's time is sometimes limited and while this is separate from the current open pull request for DIA/QIO modes it also builds on that. My question is, should I leave it as separate to submit later or update my pull request to include it now? This new stuff isn't all that much but it might make it easier to look at it in isolation of the new DIO/QIO code as they are separate ideas.

Comments?

C47D, 2020-10-02:
> This really, really boosts the performance of the FT81x driver but I also think it will boost the performance of the existing drivers for the il3820, jd79653a, and uc8151d. I say "think" because [...]

I have the IL3820, so I can test the new approach on that display, @huming2207 have contributed the other two drivers, so maybe he can test it, what do you think @huming2207?

> The changes are all in disp_spi.c and are not very complicated but they do have the potential to send individual SPI transactions at faster rates as it removes the default send & [...]

It's great the new approach only changes code on the `disp_spi.c` file, this means it's really ESP32 specific and once we get the generic drivers done (hopefully one day soon) we can still using this approach.

> However, it potentially speeds up the data streaming to them and if they can't handle it then they might need to be adjusted. If a driver can't handle getting data at a fast burst rate then the best [...]

I got some displays so I can test them.

> Another thing is, while the code is fairly simple I sort of feel that maybe a few sentences somewhere about the queuing design and how it works may be warranted as there are maybe a few subtle [...]

Just saw the comments on the PR, thanks for documenting them, I don't feel particularly familiar nor expert on the ESP32 platform so having those comments is always nice to have at hand.

> Lastly, I know other's time is sometimes limited and while this is separate from the current open pull request for DIA/QIO modes it also builds on that. My question is, should I leave it as separate [...]

That PR was merged yesterday, thanks for it 😄 .

huming2207, 2020-10-02:
Hello @C47D 

I remember previously ESP32's SPI can't do DMA when it's in half-duplex, single-wire MISO+MOSI mode. But meanwhile, those e-paper panels are using those modes. It's fine if you don't read anything from the panel (i.e. assuming the panel is full-duplex, no MISO), otherwise you may have some issues when reading. The reading may be necessary for panel self-check and temperature measurement.

Jackson

davidjade, 2020-10-02:
Hi @huming2207 

By single-wire SPI do you mean the same IO pin for both MISO & MOSI? Does the ESP32 SPI even support that? I don't recall seeing a reference to that support. Does this work now? That would have to be half-duplex mode if so, so would have the half-duplex mode DMA restrictions.

The main restriction with ESP32 half-duplex and DMA is you can't mix reading and writing in one transaction as the I/O pin modes typically have to switch in DIO and QIO modes. However, when writing this isn't an issue as they don't switch so you can do DMA writes all day in half-duplex mode just fine in SIO, DIO, and QIO configs. You can also do DMA reads as well with optional dummy byte setup in SIO/DIO/QIO. What you can't do is reading and writing in the same transaction even serially as a write followed by a read phase - unless the only writing you need to do is send address and/or command bits - this is the sole exception that the ESP32 supports. The FT81x driver actually does exactly this. Because both DIO and QIO modes use both pins in the same direction at a time, this also means the both DIO & QIO modes are half-duplex by definition. For SIO (standard MISO/MOSI) I don't think it really matters but the single read or write phase only is still enforced by the ESP32 SPI driver in half-duplex mode. 

At least that is how I understand it to work. 

Note that there seems to be a ESP32 SPI bug when reading in half-duplex mode that makes it only possible to read one byte at a time, at least when address and dummy bits are involved. I reported this to Epressif and haven't heard much back yet. I had to work-around it in the FT81x driver as this is exactly how reads need to be set up for it. The bug doesn't show itself in full-duplex mode when reading but then you can't use DIO or QIO mode.

I don't see any reason why you can't start up as full-duplex, tear down and re-init in half-duplex or vice-versa as long as the particular device supports that. disp_spi.c would need tweaking thought as it currently assumes that half vs full is a compile time setting.

The FT81x display where I write and test most of this stuff supports full, half, SIO, DIO, and QIO modes so I regularly put the SPI code through all the scenarios to make sure I haven't broken things. I've tested full-duplex writing/reading in standard MISO/MOSI mode and half-duplex separate writing/reading in SIO, DIO, and QIO modes and so far it keeps working!

huming2207, 2020-10-02:
Hi @davidjade 

> By single-wire SPI do you mean the same IO pin for both MISO & MOSI? Does the ESP32 SPI even support that? I don't recall seeing a reference to that support. Does this work now? That would have to be half-duplex mode if so, so would have the half-duplex mode DMA restrictions.

Sorry I was distracted by something else. I meant the 3-wire SPI, that is CS, SCLK and DATA where DATA = MOSI and MISO together. See this:  https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/spi_master.html#c.SPI_DEVICE_3WIRE

> The main restriction with ESP32 half-duplex and DMA is you can't mix reading and writing in one transaction as the I/O pin modes typically have to switch in DIO and QIO modes. However, when writing [...]

Hmm ok, understood. Previously for our own projects, I always kept the same setting for every transaction so probably that was the actual issue that the read operations didn't work. 

I will go for a try tonight for my JD79653A panels with your improvements. Thanks.

Regards,
Jackson

davidjade, 2020-10-02:
@huming2207 

If you look at disp_spi.c you'll see that in half-duplex reading, the write transaction length has to be set to zero and the rxlength has to be valid or the ESP32 driver flips out. 

That said, this 3WIRE mode is one thing I can't test but currently it's not supported by disp_spi.c either.

PR #205 (2020-10-05): Switch to using a pool of DMA transactions for reduced latency
This update switches SPI displays to used a pool of spi_transaction_ext_t structures for DMA transactions and resolves #203. It reduces DMA transaction latency and thus increases overall transfer throughput. This is completely transparent to all existing drivers but for those that burst multiple DMA requests (such as the FT81x, il3820, jd79653a, uc8151d, etc...) it can have a really big impact on overall display responsiveness and animation speeds. It also better supports LVGL's dual buffer mechanism by making the SPI thread wait l [... cut]
closed 2020-10-05 by C47D