lvgl/lv_esp_idf #212: Typo in CONFIG_LV_FONT_MONTSERRAT12SUBPX

maxd, 2020-10-07:
**Describe the issue**

`CONFIG_LV_FONT_MONTSERRAT12SUBPX` option from `components/lvgl/lv_conf.h` has typo: `PX` instead of `PIX`. It cause compilation error if `Montserrat 12 subpixel` font is selected in menu config.

I suppose that instead of fix typo in this place the `LV_FONT_MONTSERRAT12SUBPIX` config option from `components/lvgl/Kconfig` should be renamed to `LV_FONT_MONTSERRAT12SUBPX` because `px` suffix is using in other places (see for example `lv_font_montserrat_12_subpx`, `LV_FONT_SUBPX_BGR`, `LV_USE_FONT_SUBPX`, etc.)

**Code to reproduce the issue**

Just select Montserrat 12 subpixel font from `menuconfig` to get compilation error.

**Expected Results**

No compilation errors.

**Actual Results**

Compilation error:

```
../components/lv_port_esp32/components/lvgl/lvgl/src/lv_core/../../../lv_conf.h:746:46: error: 'lv_font_montserrat_12_subpx' undeclared (first use in this function); did you mean 'lv_font_montserrat_16'?
 #define LV_THEME_DEFAULT_FONT_SMALL         &lv_font_montserrat_12_subpx
                                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~
```

**ESP32 Chip version**

**ESP-IDF version**

**Development kit used**

**Development machine OS**

**Compilation warnings/errors (if available)**

**If possible, copy the compilation log into a file and attach it here**

C47D, 2020-10-07:
Thanks for the report, I will fix it later today.

C47D, 2020-10-08:
@maxd Does #213 solves the issue?

maxd, 2020-10-08:
@C47D I have checked compilation with and without Montserrat 12 sub-pixel font in my project. Yes, everything works fine without any compilation errors :)

C47D, 2020-10-08:
Great, thanks for the quick reply @maxd, it takes a lot of time to compile in my pc (it's quite old), will merge the PR!

closed 2020-10-08 by C47D