lvgl/lv_esp_idf #215: Using a third SPI peripheral with LVGLs Display and Touch drivers

BFFonseca, 2020-10-08:
**Describe the issue**
I'm using an ESP32 with a ILI9341 driver display and a XPT2046 touch driver. They're both running on HSPI.

I'm trying to incorporate another SPI peripheral in the same bus (HSPI).

In the ESP32 documentation it says it is possible to have 3 peripherals per SPI bus.

When I try to use the same SPI bus for all of these, I get the error:
```
E (8973) spi: SPI2 already claimed by spi master.
E (8983) spi_master: spi_bus_initialize(232): host already in use
```

I am initializing the 3rd SPI peripheral like this:
```
  spi_bus_config_t bus = {
       .miso_io_num = 19,
       .mosi_io_num = 13,
       .sclk_io_num = 14,
       .quadwp_io_num = -1,
       .quadhd_io_num = -1,
       .max_transfer_sz = 0};

   ret = spi_bus_initialize(HSPI_HOST, &bus, 1);
   assert(ret == ESP_OK);

   spi_device_interface_config_t dev = {
       .clock_speed_hz = 9000000,
       .mode = 0,
       .spics_io_num = 5,
       .queue_size = 1,
       .flags = 0,
       .pre_cb = NULL};
   ret = spi_bus_add_device(HSPI_HOST, &dev, &__spi);
```
As for LVGL, I'm using the same SPI pins and 
```
DISP_SPI_CS 12
TOUCH_SPI_CS 32
```
I am initializing LVGL first and only (works normally, both the display and the touch panel) and then the other peripheral, on a separate task.

**ESP32 Chip version**
ESP32-WROOM-32

**ESP-IDF version**
ESP-IDF 4.1 

**Development kit used**
ESP 32 DEVKIT V1

**Development machine OS**
Ubuntu 14.04.5 LTS

davidjade, 2020-10-08:
I am not positive about this but I believe this issue may be trying to call spi_bus_initialize() a second time. I think it can only be called once and the esp32 LVGL drivers already do that.

BFFonseca, 2020-10-09:
> I am not positive about this but I believe this issue may be trying to call spi_bus_initialize() a second time. I think it can only be called once and the esp32 LVGL drivers already do that.

I have tried that and now I get 
`I (172413) spi_master: Allocate RX buffer for DMA`

Do you have any idea why this is?
From what I can tell, LVGL alocates the buffer size relative to the display buffer size

davidjade, 2020-10-09:
The LVGL SPI driver is set up to use DMA so if you are using buffers for transfers, then they must be in DMA capable RAM. You need to use heap_caps_malloc() and specify MALLOC_CAP_DMA for your transfer buffers.

BFFonseca, 2020-10-26:
@C47D were you able to do some testing on this?

C47D, 2020-10-28:
Hi @BFFonseca,

I did, and successfully added two devices into the SPI master, apart of the display (my touch controller isn't working). What I did in main, is:
```c
/* Initialize SPI or I2C bus used by the drivers */
lvgl_driver_init();

const spi_device_interface_config_t devcfg = {
    .clock_speed_hz = 1000*2 /* your device speed in hz */,
    .mode = 0 /* your device spi mode */,
    .spics_io_num = 13,
    .input_delay_ns = 0,
    .queue_size = 1,
    .pre_cb = NULL,
    .post_cb = NULL,
    .flags = SPI_DEVICE_NO_DUMMY | SPI_DEVICE_HALFDUPLEX,
};

esp_err_t ret = spi_bus_add_device(TFT_SPI_HOST, &devcfg, &spi);
assert(ESP_OK == ret);
```

`spi` is a global variable of type `spi_device_handle_t`, you also would need to include `driver/spi_master.h`.

I haven't tested transferring data with the new device, the new device is a SD card connected via SPI, so it will take me a bit more research.

BFFonseca, 2020-10-28:
@C47D I will do some more testing based on that code when I get the chance. 
I will let you know when I do.
Thank you!

C47D, 2020-10-28:
I will try to add a thread to do dummy transfers periodically.

C47D, 2020-11-19:
I'm currently working on adding a SD card connected via SPI (same spi of the display controller) for a personal project, will update if I'm able to make it work.