lvgl/lv_esp_idf #220: PlatformIO instructions are incomplete

cmumford, 2020-10-24:
I'm new to PlatformIO, but I'm pretty sure that the [platformio instructions](https://github.com/lvgl/lv_port_esp32#platformio-support) in README.md are incorrect. I read through issue #168 and it only recommends the following:

1. Start with [lv_platformio](https://github.com/lvgl/lv_platformio)
2. Modify `platformio.ini`
3. Modify `main.cc`

I think it is ambiguous whether it is necessary to create the git submodule in the lv_platformio project is required as described in [Build this repository standalone and run the demo](https://github.com/lvgl/lv_port_esp32#build-this-repository-standalone-and-run-the-demo). If this was done, then this would result in two copies of lvgl. One in `//components/lv_port_esp32/components/lvgl/lvgl/`, and a second in `//lib/lvgl/`. If the submodule is not created, then there is a missing step to add lv_port_esp32 as a dependency (lib_deps) in platformio.ini.

There is a [comment](https://github.com/lvgl/lv_port_esp32/issues/168#issuecomment-664386098) in issue 168 which says that following the README.md steps successfully builds - which it does, but the default app doesn't use any drivers and doesn't display anything.

Maybe somebody with more PlatformIO knowledge can see what I might be doing wrong, but from what I can see this project is not compatible (or ready to be used by) PlatformIO.

cmumford, 2020-10-25:
I managed to get my project running with lv_port_esp32. Here's what I had to do:

1. Checkout lv_port_esp32 into the //lib directory.
2. Make the following changes to lv_port_esp32:
   1. Include lvgl.h differently, PR #221 
   2. Include lv_refr.h diffently, PR #222.
   3. Added a library.json file (no PR yet - more on that later).
3. Made the following changes to my app's project:
   1. Modified my `platformio.ini` file to have "lv_port_esp32" in the "lib_deps" section.
   2. Copied `lib/lv_port_esp32/components/lvgl/lv_conf.` to `src/lv_conf.h`
   3. Copied the three Kconfig files from this project (skipping examples) to my project.
   4. Ran `platformio run -t menuconfig`, and set values as described by https://github.com/lvgl/lv_port_esp32#configuration-options.
   5. Added the LVGL code from main.c into my project's application.

Here is what I'd like to do to complete support for PlatformIO for this project - as far as I can tell lv_port_esp32 really isn't yet compatible with PlatformIO.

1. Land the two PR's above.
2. Complete and land a PR for library.json. This is necessary for platformio to see this project as a library project and to know what files to build.
3. Modify all driver files to ifdef out their code if the appropriate constant isn't defined. e.g. the contents of `ili9341.` will be surrounded by:
    ```c
    #if defined(CONFIG_LV_TFT_DISPLAY_CONTROLLER_ILI9341)
    ...
    #endif
    ```
    This is necessary because library.json doesn't have the ability (I believe) to selectively compile files, so they will need to be a no-op - except for the drivers turned on in sdkconfig.
4. Hopefully come up with an alternative to making a copy of `lv_conf.h`.
5. Hopefully come up with an alternative to copying the Kconfig files.
6. Modify the [README platformio section](https://github.com/lvgl/lv_port_esp32#platformio-support). I believe it should still say "go to lv_platformio and follow instructions there".
7. Add the actual LVGL+PlatformIO+ESP32 instructions to the lv_platformio project.

If anybody knows PlatformIO better, please let me know if I'm not seeing an easier solution, but this is the cleanest solution I can come up with at the moment.

kisvegabor, 2020-10-26:
Hi Chris,

Thanks for sharing your experiences. The things you have pointed out seems correct.

In the very near future, we will make `lvgl` and `lv_examples` directly work as an ESP component. lvgl was updated to work with Kconfig (Kconfig file will be added soon) and without `lv_conf.h`, that is you can add LVGL defines to `platfromio.ini`.  (it should solve "Hopefully come up with an alternative to making a copy of lv_conf.h."). 

Finally, the drivers part from this repo will be moved to its own repo. The changes you have suggested for the driver part seems valid to me. However, @C47D is the maintainer of this repo and he knows it much better. :slightly_smiling_face:

kisvegabor, 2020-11-02:
@cmumford Thanks for sharing this. I wonder if we can add these environment settings to [platform.ini](https://github.com/lvgl/lv_platformio/blob/master/platformio.ini#L53) file `lv_platformio`.

tlechman49, 2020-12-08:
Is more work being done on this topic? Should I just follow steps as @cmumford described? The instructions for set up with platformio are still incomplete and this post doesn't seem like it should be left closed.

C47D, 2020-12-08:
Hi @tlechman49, can you try the @cmumford steps? If they work please let us know, on the meantime I reopen the issue.

cmumford, 2020-12-08:
I'm happy to put together a PR with these changes if @tlechman49 concludes they are correct.

tlechman49, 2020-12-08:
So I actually just posted a pull request ( #234 ) for this that worked fine for me. If someone else tests that, I guess further notes should be added on how to set the .ini file in the readme and then this would be solved.

kisvegabor, 2021-01-04:
The PR is merged so I close this issue.

Thank you @tlechman49

closed 2021-01-04 by kisvegabor