lvgl/lv_esp_idf #227: Display compatibility problem (other resolution than 800x480).

neilim, 2020-11-03:
**The problem:**
480x272 display doesn't work properly.
**Why?**
I have found a bug in this file: _components/lvgl_esp32_drivers/lvgl_tft/FT81x.c_ in this function: void _TFT_bitmap_display(void)._
In this line:
_EVE_cmd_setbitmap(SCREEN_BITMAP_ADDR, EVE_RGB565, **800**, **480**);_
The resolution parameters are fixed to 800x480. If the configured display resolution is 480x272, then the library is unusable.
**Possible solution:**
If I modify this line to: _EVE_cmd_setbitmap(SCREEN_BITMAP_ADDR, EVE_RGB565, **EVE_HSIZE**, **EVE_VSIZE**);_ the display works fine with 800x480 and 480x272 also.

davidjade, 2020-11-03:
Whoops. Thanks for catching that. I will get it fixed ASAP.

C47D, 2020-11-04:
Thanks @neilim for reporting the issue, thanks @davidjade for the quick fix.

PR #228 (2020-11-04): Fix for FT81x screen size settings
This fixes #227
closed 2020-11-04 by C47D