lvgl/lv_esp_idf #229: fatal error: lvgl/lvgl.h: No such file or directory

app-z, 2020-11-07:
We use GitHub issues for development related discussions.
Please use the [forum](https://forum.littlevgl.com/) to ask questions.

**Describe the issue**

I tried  compile example for ST7789

.dir/main.c.obj.d -o esp-idf/main/CMakeFiles/__idf_main.dir/main.c.obj   -c ../main/main.c
In file included from ../components/lvgl_esp32_drivers/lvgl_helpers.h:18,
                 from ../main/main.c:25:
../components/lvgl_esp32_drivers/lvgl_tft/disp_driver.h:18:10: fatal error: lvgl/lvgl.h: No such file or directory
 #include "lvgl/lvgl.h"
          ^~~~~~~~~~~~~
compilation terminated.
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

C47D, 2020-11-07:
Hi, try editing it to:

```c
#include "lvgl.h"
```

That should make it work, if so please let me know and I will do the proper fix.

app-z, 2020-11-07:
Sorry I fogot about --recurse-submodules
At least I can build now

C47D, 2020-11-07:
Thanks for letting me know, I will leave the issue open just in case you need any help.

closed 2020-11-29 by C47D

romocitto88, 2020-12-01:
Hi I'm using this library on PlatformIO, where have I execute the command "--recurse-submodules" to solve this issue?
I've included the library directly from platformIO, in library section.
Thank you in advance.

C47D, 2020-12-01:
Hi @romocitto88, you should use the --recurse-submodules` when cloning this repo.