lvgl/lv_esp_idf #230: True double buffer with SPI DMA in ESP32

eme-bennytang, 2020-11-18:
Hi,

I am creating a lvgl project in ESP32.

Display driver: ST7735S (Max 13MHz SPI CLK)
LCD: 160*128 pixels
ESP32 CLK: 160MHz

Been trying to figure out what can be done to reduce lv_task_handler duration and to increase FPS.

My problem is:
- lv_task_handler() last for a really long time (~70ms in my case with 1/4 buffer) and sometimes exceed 100ms if UI is complicated
- Tried using screen-sized double buffer with SPI DMA, to render while flushing (as stated in LVGL documents)
-- But after some digging in the lvgl code, I found the following code in lv_refr.c
[image]

In every flush callback:
- Started SPI DMA
- 160*128*2*8/13000000=~25ms to flush the whole screen
- lv_disp_flush_ready() called in the DMA complete callback to clear "disp_drv->buffer->flushing = 0"
- "while(vdb->flushing);" is called after "lv_refr_vdb_flush();"
- Doesnt that make the lv_task_handler() just wait for the SPI DMA to finish?
- But the double buffer is supposed to render while SPI DMA is in progress

Also, I failed to put lv_disp_flush_ready() inside SPI DMA callback
Can anyone help to suggest how to implement this?

kisvegabor, 2020-11-19:
Hi,

With a screen-sized double buffer, LVGL assumes that the flushing is only the frame buffer address change. It's not the case in your configuration as you need to send the image to the display via SPI.

If you want screen-sized buffers but not "true double buffering" just use `screen_size+1` sized buffers.

See the different buffering modes here: https://docs.lvgl.io/latest/en/html/porting/display.html#display-buffer

eme-bennytang, 2020-11-20:
Hi,

Oh I see! 
So if I want to "render and flush in parallel" I should not be using "true double buffer" (2 screen sized buffer).
Instead I should configure the buffers size to screen size +1 to avoid "true double buffer" but still enjoing "rendering and flushing" in parallel?

Thanks for the help anyway, been struggling to understand the buffer configurations

closed 2020-12-24 by stale[bot]