lvgl/lv_esp_idf #231: RPi MPI3501 Compile error error: 'data' undeclared

samuelbles07, 2020-12-03:
We use GitHub issues for development related discussions.
Please use the [forum](https://forum.littlevgl.com/) to ask questions.

**Describe the issue**
Compile error using preconfigured `RPi MPI3501`.

**Code to reproduce the issue**

TFT Display controller config: 
- Predifined display: RPi MPI3501
- SPI Bus: HSPI
- SIO (MOSI/MISO)
- HALF Duplex
- Landscape
- PIN: MOSI 23, CLK 18, CS 5, DC 12, RST 4

Touch controller config:
- XPT2046
- VSPI
- Others default

**Expected Results**
compile success

**Actual Results**
```
-c ../components/lvgl_esp32_drivers/lvgl_tft/ili9486.c
In file included from ../components/lvgl_esp32_drivers/lvgl_tft/ili9486.c:12:
../components/lvgl_esp32_drivers/lvgl_tft/ili9486.c: In function 'ili9486_set_orientation':
../components/lvgl_esp32_drivers/lvgl_tft/ili9486.c:210:49: error: 'data' undeclared (first use in this function)
     ESP_LOGI(TAG, "0x36 command value: 0x%02X", data[orientation]);
                                                 ^~~~
/home/samuel/tools/esp/esp-idf/components/log/include/esp_log.h:330:137: note: in definition of macro 'ESP_LOG_LEVEL'
         if (level==ESP_LOG_ERROR )          { esp_log_write(ESP_LOG_ERROR,      tag, LOG_FORMAT(E, format), esp_log_timestamp(), tag, ##__VA_ARGS__); } \
                                                                                                                                         ^~~~~~~~~~~
/home/samuel/tools/esp/esp-idf/components/log/include/esp_log.h:297:38: note: in expansion of macro 'ESP_LOG_LEVEL_LOCAL'
 #define ESP_LOGI( tag, format, ... ) ESP_LOG_LEVEL_LOCAL(ESP_LOG_INFO,    tag, format, ##__VA_ARGS__)
                                      ^~~~~~~~~~~~~~~~~~~
../components/lvgl_esp32_drivers/lvgl_tft/ili9486.c:210:5: note: in expansion of macro 'ESP_LOGI'
     ESP_LOGI(TAG, "0x36 command value: 0x%02X", data[orientation]);
     ^~~~~~~~
../components/lvgl_esp32_drivers/lvgl_tft/ili9486.c:210:49: note: each undeclared identifier is reported only once for each function it appears in
     ESP_LOGI(TAG, "0x36 command value: 0x%02X", data[orientation]);
                                                 ^~~~
/home/samuel/tools/esp/esp-idf/components/log/include/esp_log.h:330:137: note: in definition of macro 'ESP_LOG_LEVEL'
         if (level==ESP_LOG_ERROR )          { esp_log_write(ESP_LOG_ERROR,      tag, LOG_FORMAT(E, format), esp_log_timestamp(), tag, ##__VA_ARGS__); } \
                                                                                                                                         ^~~~~~~~~~~
/home/samuel/tools/esp/esp-idf/components/log/include/esp_log.h:297:38: note: in expansion of macro 'ESP_LOG_LEVEL_LOCAL'
 #define ESP_LOGI( tag, format, ... ) ESP_LOG_LEVEL_LOCAL(ESP_LOG_INFO,    tag, format, ##__VA_ARGS__)
                                      ^~~~~~~~~~~~~~~~~~~
../components/lvgl_esp32_drivers/lvgl_tft/ili9486.c:210:5: note: in expansion of macro 'ESP_LOGI'
     ESP_LOGI(TAG, "0x36 command value: 0x%02X", data[orientation]);
     ^~~~~~~~
[923/1032] Building C object esp-idf/lv...l/src/lv_themes/lv_theme_material.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

**ESP32 Chip version**
ESP32-WROOM-32

**ESP-IDF version**
v4.3-dev-1197-g8bc19ba89
**Development kit used**
 ESP32-DevKitC
**Development machine OS**
Ubuntu 18.0

samuelbles07, 2020-12-04:
I tried to comment out all statement inside [ili9486_set_orientation](https://github.com/lvgl/lv_port_esp32/blob/master/components/lvgl_esp32_drivers/lvgl_tft/ili9486.c#L196). Compile success and works but the orientation is vertical (i think). 

[image]

The problem is the define variable `CONFIG_LV_PREDEFINED_DISPLAY_NONE` that make `data` not declared. Is there any command register for changing the orientation?

C47D, 2020-12-08:
Hi @samuelbles07 I will try to take a look at this later today.

samuelbles07, 2020-12-09:
Hi @C47D , thanks for your response, looking forward!

It's not related to MPI3501 but this driver [file](https://github.com/lvgl/lv_port_esp32/blob/master/components/lvgl_esp32_drivers/lvgl_tft/ili9488.c#L226) might have same issue too

C47D, 2020-12-11:
Hi @samuelbles07 ,

This function `ili9488_send_cmd(0x36)` lets the display driver know that we want to change it's orientation, after that we send the configuration value of the register with address 0x36. The elements of the `data` array are configuration values that works on my display.