lvgl/lv_esp_idf #233: LVGL does not build for ESP32-S2

cmumford, 2020-12-08:
We use GitHub issues for development related discussions.
Please use the [forum](https://forum.littlevgl.com/) to ask questions.

**Describe the issue**

When compiling the build fails: (see below).

I made a fix at https://github.com/cmumford/lv_port_esp32/commit/8697e2cdd05ba3d49393a52f43959cdb92c178a9. However, I don't yet have an attached screen, so it is untested - it only compiles. I will create a PR for that commit when I have a chance to test the change. (maybe this weekend :crossed_fingers:)

**Code to reproduce the issue**

Follow getting started guide: https://github.com/lvgl/lv_port_esp32#get-started except specify the esp32s2 target:

```sh
idf.py set-target esp32s2
idf.py menuconfig
```

**Expected Results**

Successful build

**Actual Results**

Failure as shown below

**ESP32 Chip version**

ESP32-S2

**ESP-IDF version**

ESP-IDF v4.3-dev-1901-g178b122c1

**Development kit used**

Cucumber R ESP32-S2 (but never ran on it)

**Development machine OS**

macOS: 10.15.7

**Compilation warnings/errors (if available)**

```
../components/lv_port_esp32/components/lvgl_esp32_drivers/lvgl_helpers.c: In function 'lvgl_spi_driver_init':
../components/lv_port_esp32/components/lvgl_esp32_drivers/lvgl_helpers.c:204:35: error: 'VSPI_HOST' undeclared (first use in this function); did you mean 'FSPI_HOST'?
     assert((SPI_HOST <= host) && (VSPI_HOST >= host));
                                   ^~~~~~~~~
~/esp/esp-idf/components/esp_common/include/esp_compiler.h:29:25: note: in definition of macro 'likely'
 #define likely(x)      (x)
                         ^
../components/lv_port_esp32/components/lvgl_esp32_drivers/lvgl_helpers.c:204:5: note: in expansion of macro 'assert'
     assert((SPI_HOST <= host) && (VSPI_HOST >= host));
     ^~~~~~
../components/lv_port_esp32/components/lvgl_esp32_drivers/lvgl_helpers.c:204:35: note: each undeclared identifier is reported only once for each function it appears in
     assert((SPI_HOST <= host) && (VSPI_HOST >= host));
                                   ^~~~~~~~~
~/esp/esp-idf/components/esp_common/include/esp_compiler.h:29:25: note: in definition of macro 'likely'
 #define likely(x)      (x)
                         ^
../components/lv_port_esp32/components/lvgl_esp32_drivers/lvgl_helpers.c:204:5: note: in expansion of macro 'assert'
     assert((SPI_HOST <= host) && (VSPI_HOST >= host));
```

C47D, 2020-12-08:
Hi @cmumford, thanks for the report.
I don't have any ESP32-S2 at hand, but if I remember correctly it doesn't have the same SPI master ports as the ESP32, let me check it's documentation and we can try figure out a way to make it work.

I will try to update the LVGL submodule as the Kconfig file now is available on the main repo, but before that I will create a release with the current state of the repo.

C47D, 2020-12-09:
Can you test using this branch https://github.com/lvgl/lv_port_esp32/tree/esp32s2_aware ? It's a hotfix but I didn't had time to build the project using ESP32S2 as target.

cmumford, 2020-12-09:
The esp32s2_aware branch built successfully for me. I did see two warnings at the end of the build.

```
[1199/1202] Generating ld/esp32s2.project.ld
warning: the choice symbol LV_TFT_DISPLAY_SPI_FSPI (defined at /Users/cmumford/src/keyboard/components/lv_port_esp32/components/lvgl_esp32_drivers/lvgl_tft/Kconfig:414, /Users/cmumford/src/keyboard/components/lv_port_esp32/components/lvgl_esp32_drivers/lvgl_touch/Kconfig:89) is defined with a prompt outside the choice
warning: the choice symbol LV_TFT_DISPLAY_SPI_FSPI (defined at /Users/cmumford/src/keyboard/components/lv_port_esp32/components/lvgl_esp32_drivers/lvgl_tft/Kconfig:414, /Users/cmumford/src/keyboard/components/lv_port_esp32/components/lvgl_esp32_drivers/lvgl_touch/Kconfig:89) is defined with a prompt outside the choice
[1202/1202] Generating binary image from built executable
esptool.py v3.0
Generated /Users/cmumford/src/keyboard/build/keyboard.bin
```

C47D, 2020-12-09:
Seems like an issue in the Kconfig file.

cmumford, 2020-12-09:
Agreed. I made one comment on https://github.com/lvgl/lv_port_esp32/commit/743104ba96d2c9275d57dc4ba370e9c2d6a4f575, but I haven't done any Kconfig work, so that might be intentional.

C47D, 2020-12-10:
@cmumford I've updated the branch, selected the FSPI by default on ESP32S2 targets and fixed a typo that triggered the warning on https://github.com/lvgl/lv_port_esp32/issues/233#issuecomment-741882137

cmumford, 2020-12-10:
I tested using commit e326044. It built OK, but still a warning:

```
[1214/1217] Generating ld/esp32s2.project.ld
warning: the default selection LV_TOUCH_CONTROLLER_SPI_FSPI (undefined) of <choice> (defined at /home/cmumford/src/keyboard/components/lv_port_esp32/components/lvgl_esp32_drivers/lvgl_touch/Kconfig:76) is not contained in the choice
```

Interesting that the warning was only emitted once - not twice like last time.

C47D, 2020-12-10:
My fault, it was another typo, just pushed 3fc2e2f, that should fix it.

cmumford, 2020-12-10:
That works - no errors/warnings. 👍

C47D, 2020-12-10:
Great, feel free to close the issue when you think it's fully solved.

cmumford, 2020-12-10:
I'm not sure I'll be able to test this weekend so I'll close this now. I'll open a new issue if I encounter any ESP32-S2 issues when I get around to testing. Thx for the quick fix @C47D.

closed 2020-12-10 by cmumford

C47D, 2020-12-10:
No rush @cmumford, test it when you have some free time to do so.

Open a new issue if you have any problems.

cmumford, 2020-12-14:
Followup... **working fine** on my ESP32-S2 ([Cucumber R ESP32-S2](https://www.gravitech.us/curesdebo.html)). I only tested FSPI - and didn't test touch - just display.

```ini
CONFIG_LV_TFT_DISPLAY_SPI_FSPI=y
#
# Display Pin Assignments
#
CONFIG_LV_DISP_SPI_MOSI=11
# CONFIG_LV_DISPLAY_USE_SPI_MISO is not set
CONFIG_LV_DISP_SPI_CLK=12
CONFIG_LV_DISPLAY_USE_SPI_CS=y
CONFIG_LV_DISP_SPI_CS=38
CONFIG_LV_DISPLAY_USE_DC=y
CONFIG_LV_DISP_PIN_DC=34
CONFIG_LV_DISP_PIN_RST=37
CONFIG_LV_DISP_PIN_BUSY=35
# CONFIG_LV_ENABLE_BACKLIGHT_CONTROL is not set
CONFIG_LV_DISP_PIN_SDA=5
CONFIG_LV_DISP_PIN_SCL=4
```

C47D, 2020-12-14:
Thanks for the follow up @cmumford !, touch code is a bit dated, I might need to work on it.

advancedtigerdenempr, 2022-02-02:
Good afternoon. The code is not compiled for either ESP32-S2 or ESP32-S3. The problem is https://github.com/lvgl/lv_port_esp32/issues/233#issue-759602094

PR #235 (2020-12-10): Esp32s2 aware
Fixes #233