lvgl/lv_esp_idf #236: Moving driver to it's own repo

kisvegabor, 2020-12-16:
Hi,

I think the last step to finish the "LVGL as an ESP component" is to separate the drivers from this repo. 

Are there any blocking issues to do this?

C47D, 2020-12-16:
I can try moving the `lvgl_esp32_drivers` directory into the other repo as is, and fix errors, if any.

C47D, 2020-12-17:
See test branch, I think I didn't broke anything 😄

kisvegabor, 2020-12-17:
Cool, thanks! :slightly_smiling_face: 

It seems to me it works the same way as lvgl, and lv_examples - a "normal" ESP32 component that can be used directly in the component folder. Is it correct?

C47D, 2020-12-17:
Yes, it's similar to those components, the only difference is the location of the Kconfig files. Because of that we need to tell the esp-idf framework the location of those files.

Just let me do another test and I will let you know when to merge it, what do you think?

kisvegabor, 2020-12-17:
> Just let me do another test and I will let you know when to merge it, what do you think?

Perfect! :heart:

C47D, 2020-12-24:
Closed with bfbc1c5d5fc729f76e0d3163f920dc26b5464c59

closed 2020-12-24 by C47D