lvgl/lv_esp_idf #238: disp_drv.rotated not working with ST7735 driver

Lisa999, 2020-12-18:
The problem is that the ST7735 display is 160x128 pixels.
So if you set:

- Landscape during compile time
- disp_drv.rotated = 1 in the code
- Swap display width and height in the code
- Call the ST7735 rotation function for a hardware rotate

It still is not working in portrait mode. The upper 128 pixels are working in portrait mode, the bottom 32 pixels are not functioning at all (pixel noise). 
I suspect the LVGL driver code is still thinking that the height is 128 pixels, despite the rotated setting. 

Maybe because of this code in LV_CONF.H ?
#define LV_VER_RES_MAX          (CONFIG_LVGL_DISPLAY_HEIGHT)

Lisa999, 2020-12-18:
Ok, one step further. Setting this helps a little:
    disp_drv.hor_res = canvas_width;
    disp_drv.ver_res = canvas_height;

C47D, 2020-12-20:
Hi,

You're right, the drivers code were developed without taking in consideration the `rotated` field. We're about to move the drivers code into its own repository. You can find it [here](https://github.com/lvgl/lvgl_esp32_drivers).

After merging that repo I will try to update the drivers orientation code.

Lisa999, 2020-12-20:
void st7735s_flush(lv_disp_drv_t * drv, const lv_area_t * area, lv_color_t * color_map)
{
	uint8_t data[4];
	ESP_LOGD(TAG, "Flush, Portrait mode: %u, width: %u , height %u", drv->rotated, drv->hor_res, drv->ver_res);

	/*Column addresses*/
	st7735s_send_cmd(ST7735_CASET);
	data[0] = (area->x1 >> 8) & 0xFF;
	data[1] = (area->x1 & 0xFF) + (drv->rotated ? COLSTART : ROWSTART);
	data[2] = (area->x2 >> 8) & 0xFF;
	data[3] = (area->x2 & 0xFF) + (drv->rotated ? COLSTART : ROWSTART);
	st7735s_send_data(data, 4);

	/*Page addresses*/
	st7735s_send_cmd(ST7735_RASET);
	data[0] = (area->y1 >> 8) & 0xFF;
	data[1] = (area->y1 & 0xFF) + (drv->rotated ? ROWSTART : COLSTART);
	data[2] = (area->y2 >> 8) & 0xFF;
	data[1] = (area->y1 & 0xFF) + (drv->rotated ? ROWSTART : COLSTART);
	st7735s_send_data(data, 4);

	/*Memory write*/
	st7735s_send_cmd(ST7735_RAMWR);

	uint32_t size = lv_area_get_width(area) * lv_area_get_height(area);

	st7735s_send_color((void*)color_map, size * 2);
}

Lisa999, 2020-12-20:
Ditch st7735s_portrait_mode var, totally unnecessary...

Lisa999, 2020-12-20:
Wrong:
uint8_t data[] = {0xC8, 0xC8, 0xA8, 0xA8};
Right:
uint8_t data[] = {0xC0, 0x00, 0x60, 0xA0};