lvgl/lv_esp_idf #241: Update README

C47D, 2020-12-31:
@kisvegabor and @embeddedt , I'm thinking on updating the README with some points:

- Better pictures, the current ones doesn't look nice (they suck to be honest, I'm not good at taking pictures).
- Remove the [install-this-project-as-a-library-submodule-in-your-own-project](https://github.com/lvgl/lv_port_esp32/#install-this-project-as-a-library-submodule-in-your-own-project) section, now the users can add LVGL and lvgl_esp32_drivers as separated submodules, they no longer need to add this whole repo.
- Update Platformio instructions.
- Cleanup in general.

Any other updates you think would be cool?

kisvegabor, 2021-01-04:
Sounds good. 

I think the info in the README could be added to the [docs](https://docs.lvgl.io/latest/en/html/get-started/espressif.html) too. So we can keep them in sync at least to some extent. 

What do you think?

C47D, 2021-01-04:
For reference: To update the README I'm working in [this](https://github.com/lvgl/lv_port_esp32/tree/update_readme) branch.

> I think the info in the README could be added to the docs too. So we can keep them in sync at least to some extent.
What do you think?

That's great, I think.

C47D, 2021-01-09:
Closed with #245 and #248

closed 2021-01-09 by C47D