lvgl/lv_esp_idf #247: RA8875 7" LCD not functioning with this library

jhancharler, 2021-01-07:
We use GitHub issues for development related discussions.
Please use the [forum](https://forum.littlevgl.com/) to ask questions.

** Issue **

After setting up the device pins on menuconfig and flashing the basic example (with some modifications that I will explain below), I see a white screen with some coloured lines. Like this:

[image]

**Code to reproduce the issue**

Basic example code as found in the latest release, using latest IDF (v4.3) however I modified the ra8875.c library to turn on the internal PWM because the LCD would not turn on otherwise.

`void ra8875_configure_clocks(bool high_speed)
{
    uint8_t val;

    val = high_speed ? ((CONFIG_LV_DISP_RA8875_PLLDIVM << 7) | CONFIG_LV_DISP_RA8875_PLLDIVN) : 0x07;
    ra8875_write_cmd(RA8875_REG_PLLC1, val);           // PLL Control Register 1 (PLLC1)
    vTaskDelay(1);

    val = high_speed ? CONFIG_LV_DISP_RA8875_PLLDIVK : 0x03;
    ra8875_write_cmd(RA8875_REG_PLLC2, val);           // PLL Control Register 2 (PLLC2)
    vTaskDelay(1);

    uint8_t P1CR_config = 0x82; //pwm enable, disable level = 0, pwm1 function, clk/8

    val = high_speed ? P1CR_config : 0x82;

    ra8875_write_cmd(RA8875_REG_P1CR, val);           // P1CR register
    vTaskDelay(1);

    uint8_t P1DCR_config = 0xfe; //pwm duty cycle 255 = 256/256 duty. 0xff == full brightness
    val = high_speed ? P1DCR_config : 0x80;

    ra8875_write_cmd(RA8875_REG_P1DCR, val);           // P1DCR register
    vTaskDelay(1);

    ra8875_write_cmd(RA8875_REG_PCSR, PCSR_VAL);            // Pixel Clock Setting Register (PCSR)
    vTaskDelay(DIV_ROUND_UP(20, portTICK_RATE_MS));
}`

**Expected Results**

The basic demo to show on the display.

**Actual Results**

See image above.

**ESP32 Chip version**

ESP32-D0WD (revision 1)

**ESP-IDF version**

4.3

**Development kit used**

ESP32-DevkitC v4

**Development machine OS**

Ubuntu v20.04.1

**Compilation warnings/errors (if available)**

None

**If possible, copy the compilation log into a file and attach it here**

C47D, 2021-01-07:
Hi, can you attach the `idf.py monitor` output?

jhancharler, 2021-01-08:
Hey C47D, thanks for your great work with this library, really enjoying using it :).

Here is the monitor output:

[ra8875_idfpy_menuconfig.txt](https://github.com/lvgl/lv_port_esp32/files/5786521/ra8875_idfpy_menuconfig.txt)

Here's the display that I'm using- it's from BuyDisplay. All the solder pads behind are soldered/left open to enable 4-wire SPI. **Resolution is 800 x 480**

[image]

Here is the logic analyzer capture from Pulseview:

[RA8875_RUN.zip](https://github.com/lvgl/lv_port_esp32/files/5786912/RA8875_RUN.zip)

In case you can't open that file, here it is via images along with my menuconfig settings:

[RA8875_Troubleshoot.pdf](https://github.com/lvgl/lv_port_esp32/files/5786948/RA8875_Troubleshoot.pdf)

C47D, 2021-01-08:
@jhancharler Thanks for all the information, I will try to take a look at it later today.

C47D, 2021-01-08:
Is it possible for you to test the latest master? I've cleaned up a bit the lvgl_esp32_drivers component.

jhancharler, 2021-01-08:
I've pulled the latest master.

`
Unpacking objects: 100% (24/24), 7.39 KiB | 1.05 MiB/s, done.
From https://github.com/lvgl/lv_port_esp32
   f1427fa..0b51387  master     -> origin/master
Fetching submodule components/lvgl_esp32_drivers
From https://github.com/lvgl/lvgl_esp32_drivers
   6b1e219..1cda575  master     -> origin/master
Updating f1427fa..0b51387
Fast-forward
 CONTRIBUTE_CONTROLLER_SUPPORT.md | 149 -----------------------------------------------------------------------------------
 README.md                        | 231 ++++++++++++++++++++++-----------------------------------------------------------------------------------------------------------
 components/lvgl_esp32_drivers    |   2 +-
 main/CMakeLists.txt              |   3 +-
 4 files changed, 41 insertions(+), 344 deletions(-)
 delete mode 100644 CONTRIBUTE_CONTROLLER_SUPPORT.md`

Unfortunately it's still showing the same issue.

I have an ILI9488 screen which I try and that one works like a charm every time FYI.

C47D, 2021-01-08:
Thanks for trying on another display, the driver for the one you're using (RA) was shared by another user. I will take a look at it because I'm not familiar with it.

davidjade, 2021-01-08:
@jhancharler I don't have this display so I can't check but I would start with a much lower clock rate as it could just be SPI corruption. Also make sure the right SPI duplex mode is configured. It looks like you are trying Full Duplex. Are you sure it is full duplex SPI? I bet it is not (full duplex means it sends and receives at the same time, unusual for a display).

jhancharler, 2021-01-09:
@davidjade Thanks for your suggestion- I thought half-duplex might have been it, but unfortunately it didn't appear to solve the issue for me.

The SPI is being clocked at 1 MHz currently which is the slowest setting the menuconfig allows.

Note: When setting SPI to half-duplex, if it is making use of DMA for SPI transactions (like how this library does), we are only allowed to send or receive, but not both. So I had to comment out `ra8875_read_cmd` call within ra8875.c, otherwise it was throwing a runtime error. 

Anyway, setting to half-duplex didn't solve the issue for me. Hopefully @C47D has better luck

C47D, 2021-01-09:
Hi, there was a recent update on the lvgl_esp32_drivers and some symbol names weren't updated (see https://github.com/lvgl/lvgl_esp32_drivers/issues/5#issuecomment-757101320), can you try using this https://github.com/lvgl/lvgl_esp32_drivers/tree/fix/update_lvgl_helpers branch of lvgl_esp32_drivers?

C47D, 2021-01-09:
This commit in particular updates the symbols used in your display driver https://github.com/lvgl/lvgl_esp32_drivers/commit/6a451cacf7d217965b66ff72afc015f51aa0a7b8

jhancharler, 2021-01-10:
Thanks @C47D , I will update this thread when I try it. I can't get round to it now but will post when I do. Thanks again for the suggestion.

C47D, 2021-01-10:
That's ok, I will update this if I find something else.

C47D, 2021-01-24:
Hi @jhancharler , did you had any chance to test the changes?

9ae8sdf76, 2021-02-24:
I'm in this boat too, except I'm getting a blank screen. What can I provide to assist? I'm using the latest master from lv_esp32_drivers (https://github.com/lvgl/lvgl_esp32_drivers/commit/72e2e765eb54409f3baa2c89c85455c3ac342246) and the latest master (https://github.com/lvgl/lv_port_esp32/commit/f2e80cbf37cb2848701c3bf83eafdfc5d5b1ed1f) from this repository. Here is my output:

```
rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7400
load:0x40078000,len:15000
load:0x40080400,len:4388
0x40080400: _init at ??:?

entry 0x40080694
I (27) boot: ESP-IDF v4.4-dev-4-g73db14240-dirty 2nd stage bootloader
I (27) boot: compile time 14:45:12
I (28) boot: chip revision: 1
I (31) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (38) qio_mode: Enabling default flash chip QIO
I (44) boot.esp32: SPI Speed      : 40MHz
I (48) boot.esp32: SPI Mode       : QIO
I (53) boot.esp32: SPI Flash Size : 4MB
I (57) boot: Enabling RNG early entropy source...
I (63) boot: Partition Table:
I (66) boot: ## Label            Usage          Type ST Offset   Length
I (74) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (81) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (89) boot:  2 factory          factory app      00 00 00010000 00100000
I (96) boot: End of partition table
I (100) boot_comm: chip revision: 1, min. application chip revision: 0
I (107) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=24fa4h (151460) map
I (161) esp_image: segment 1: paddr=00034fcc vaddr=3ffb0000 size=03194h ( 12692) load
I (166) esp_image: segment 2: paddr=00038168 vaddr=40080000 size=07eb0h ( 32432) load
I (179) esp_image: segment 3: paddr=00040020 vaddr=400d0020 size=3b2e0h (242400) map
I (251) esp_image: segment 4: paddr=0007b308 vaddr=40087eb0 size=04a34h ( 18996) load
I (265) boot: Loaded app from partition at offset 0x10000
I (265) boot: Disabling RNG early entropy source...
I (277) cpu_start: Pro cpu up.
I (277) cpu_start: Starting app cpu, entry point is 0x400810dc
0x400810dc: call_start_cpu1 at /Users/kstanl27/Downloads/esp-idf-v4.2/components/esp_system/port/cpu_start.c:143

I (0) cpu_start: App cpu up.
I (293) cpu_start: Pro cpu start user code
I (293) cpu_start: cpu freq: 240000000
I (293) cpu_start: Application information:
I (298) cpu_start: Project name:     lvgl-demo
I (303) cpu_start: App version:      v2.0-55-gf2e80cb
I (309) cpu_start: Compile time:     Feb 24 2021 15:11:11
I (315) cpu_start: ELF file SHA256:  8f5d30343ae4fa73...
I (321) cpu_start: ESP-IDF:          v4.4-dev-4-g73db14240-dirty
I (328) heap_init: Initializing. RAM available for dynamic allocation:
I (335) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (341) heap_init: At 3FFBC600 len 00023A00 (142 KiB): DRAM
I (347) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (353) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (360) heap_init: At 4008C8E4 len 0001371C (77 KiB): IRAM
I (367) spi_flash: detected chip: generic
I (371) spi_flash: flash io: qio
I (376) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
Info: lv_init ready 	(lv_obj.c #231 lv_init())
I (20) lvgl_helpers: Display hor size: 800, ver size: 480
I (20) lvgl_helpers: Display buffer size: 32000
I (30) lvgl_helpers: Initializing SPI master for display
I (30) lvgl_helpers: Configuring SPI host VSPI_HOST (2)
I (30) lvgl_helpers: MISO pin: 19, MOSI pin: 23, SCLK pin: 18, IO2/WP pin: -1, IO3/HD pin: -1
I (50) lvgl_helpers: Max transfer size: 64000 (bytes)
I (50) lvgl_helpers: Initializing SPI bus...
I (50) disp_spi: Adding SPI device
I (60) disp_spi: Clock speed: 40000000Hz, mode: 0, CS pin: 5
I (70) RA8875: Initializing RA8875...
[... 1788 lines cut ...]
# CONFIG_TCP_OVERSIZE_DISABLE is not set
CONFIG_UDP_RECVMBOX_SIZE=6
CONFIG_TCPIP_TASK_STACK_SIZE=3072
CONFIG_TCPIP_TASK_AFFINITY_NO_AFFINITY=y
# CONFIG_TCPIP_TASK_AFFINITY_CPU0 is not set
# CONFIG_TCPIP_TASK_AFFINITY_CPU1 is not set
CONFIG_TCPIP_TASK_AFFINITY=0x7FFFFFFF
# CONFIG_PPP_SUPPORT is not set
CONFIG_ESP32_PTHREAD_TASK_PRIO_DEFAULT=5
CONFIG_ESP32_PTHREAD_TASK_STACK_SIZE_DEFAULT=3072
CONFIG_ESP32_PTHREAD_STACK_MIN=768
CONFIG_ESP32_DEFAULT_PTHREAD_CORE_NO_AFFINITY=y
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_0 is not set
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_1 is not set
CONFIG_ESP32_PTHREAD_TASK_CORE_DEFAULT=-1
CONFIG_ESP32_PTHREAD_TASK_NAME_DEFAULT="pthread"
CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ABORTS=y
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_FAILS is not set
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ALLOWED is not set
CONFIG_SUPPRESS_SELECT_DEBUG_OUTPUT=y
CONFIG_SUPPORT_TERMIOS=y
CONFIG_SEMIHOSTFS_MAX_MOUNT_POINTS=1
CONFIG_SEMIHOSTFS_HOST_PATH_MAX_LEN=128
# End of deprecated options
```

C47D, 2021-02-24:
Thanks for the information @9ae8sdf76 I will take a look at it and try to figure out why you're getting a white screen output.
Did you also bought the display on BuyDisplay or is it a different display?

Regards
Carlos

9ae8sdf76, 2021-02-25:
> Did you also bought the display on BuyDisplay or is it a different display?

It is the [7" Display from Adafruit](https://www.adafruit.com/product/2354), and the controller is [their version of the RA8875](https://www.adafruit.com/product/1590).

C47D, 2021-02-27:
I just saw an error on the monitor output

```
E (350) spi_master: check_trans_valid(696): SPI half duplex mode does not support using DMA with both MOSI and MISO phases.
```

We should assert so the program doesn't continue 🤕 .

9ae8sdf76, 2021-03-02:
> From the CONFIG list I can assume you're not using any touch drivers

I disabled the touch drivers to simplify things. In my main project I do plan to use them.

> can you try disabling the MISO pin from the display configuration?
> ...
> Also, can you try configuring the display for full duplex?

Sure. Doing that now. 

```
rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7400
load:0x40078000,len:15000
load:0x40080400,len:4388
0x40080400: _init at ??:?

entry 0x40080694
I (27) boot: ESP-IDF v4.4-dev-4-g73db14240-dirty 2nd stage bootloader
I (27) boot: compile time 14:45:12
I (28) boot: chip revision: 1
I (31) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (38) qio_mode: Enabling default flash chip QIO
I (44) boot.esp32: SPI Speed      : 40MHz
I (48) boot.esp32: SPI Mode       : QIO
I (53) boot.esp32: SPI Flash Size : 4MB
I (57) boot: Enabling RNG early entropy source...
I (63) boot: Partition Table:
I (66) boot: ## Label            Usage          Type ST Offset   Length
I (74) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (81) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (89) boot:  2 factory          factory app      00 00 00010000 00100000
I (96) boot: End of partition table
I (100) boot_comm: chip revision: 1, min. application chip revision: 0
I (107) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=24fbch (151484) map
I (161) esp_image: segment 1: paddr=00034fe4 vaddr=3ffb0000 size=03194h ( 12692) load
I (166) esp_image: segment 2: paddr=00038180 vaddr=40080000 size=07e98h ( 32408) load
I (178) esp_image: segment 3: paddr=00040020 vaddr=400d0020 size=3b468h (242792) map
I (251) esp_image: segment 4: paddr=0007b490 vaddr=40087e98 size=04a4ch ( 19020) load
I (265) boot: Loaded app from partition at offset 0x10000
I (265) boot: Disabling RNG early entropy source...
I (277) cpu_start: Pro cpu up.
I (277) cpu_start: Starting app cpu, entry point is 0x400810dc
0x400810dc: call_start_cpu1 at /Users/kstanl27/Downloads/esp-idf-v4.2/components/esp_system/port/cpu_start.c:143

I (0) cpu_start: App cpu up.
I (293) cpu_start: Pro cpu start user code
I (293) cpu_start: cpu freq: 240000000
I (293) cpu_start: Application information:
I (298) cpu_start: Project name:     lvgl-demo
I (303) cpu_start: App version:      v2.0-55-gf2e80cb
I (309) cpu_start: Compile time:     Mar  1 2021 20:17:14
I (315) cpu_start: ELF file SHA256:  69b5507ac6e841f5...
I (321) cpu_start: ESP-IDF:          v4.4-dev-4-g73db14240-dirty
I (328) heap_init: Initializing. RAM available for dynamic allocation:
I (335) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (341) heap_init: At 3FFBC618 len 000239E8 (142 KiB): DRAM
I (347) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (354) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (360) heap_init: At 4008C8E4 len 0001371C (77 KiB): IRAM
I (367) spi_flash: detected chip: generic
I (371) spi_flash: flash io: qio
I (376) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
Info: lv_init ready 	(lv_obj.c #231 lv_init())
I (20) lvgl_helpers: Display hor size: 800, ver size: 480
I (20) lvgl_helpers: Display buffer size: 32000
I (30) lvgl_helpers: Initializing SPI master for display
I (30) lvgl_helpers: Configuring SPI host VSPI_HOST (2)
I (30) lvgl_helpers: MISO pin: -1, MOSI pin: 23, SCLK pin: 18, IO2/WP pin: -1, IO3/HD pin: -1
I (50) lvgl_helpers: Max transfer size: 64000 (bytes)
I (50) lvgl_helpers: Initializing SPI bus...
I (50) disp_spi: Adding SPI device
I (60) disp_spi: Clock speed: 40000000Hz, mode: 0, CS pin: 5
I (70) RA8875: Initializing RA8875...
I (270) disp_spi: Changing SPI device clock speed: 1000000
I (270) gpio: GPIO[5]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (270) disp_spi: Adding SPI device
I (280) disp_spi: Clock speed: 1000000Hz, mode: 0, CS pin: 5
I (330) disp_spi: Changing SPI device clock speed: 40000000
I (330) gpio: GPIO[5]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (330) disp_spi: Adding SPI device
I (340) disp_spi: Clock speed: 40000000Hz, mode: 0, CS pin: 5
W (1350) RA8875: WARNING: Memory clear timed out; RA8875 may be unresponsive.
I (1350) RA8875: Enabling display.
I (1350) RA8875: Enabling backlight.
```

C47D, 2021-03-02:
Hi,

I no longer see the error on the monitor output, does the display showed up something?

9ae8sdf76, 2021-03-02:
No, it still does not show up for me. I've been trying every combination I can think of, and I've matched my config with the data sheet and Adafruit's implementation. Still blank screen/no power.

9ae8sdf76, 2021-03-03:
I've been at this all day, and no matter the combination of options nothing changes. The display is just not getting turned on. I have even gone so far as change code to match Adafruit's implementation, and still no display. 

There are two things that have been bothering me:

1. When using SIO and full-duplex the memory clear operation times out, which warns the controller may be unresponsive (which it is)
2. When using DIO/QIO and half-duplex, the error about DMA comes up

I feel like I'm stuck in a catch-22 situation where the SIO for my board (wroom 32) just doesn't work, and using half-duplex won't work because somewhere there is DMA blocking its operation. I can't find any way to disable DMA or the RX/TX or anything that'll get me past that error. 

I guess it really comes down to why Adafruit can do it (albeit with slow drawing) and this library can't? (I don't mean that negatively).

C47D, 2021-03-04:
Hi,

Sorry for the late reply, I'm really busy with my day job (I keep this repo on my free time). I will try to get the same display and controller as yourself but it would take a bit to get here (I'm located in México).

viniciusmiguel, 2021-03-11:
Hello There.

I am having the same issue, on my case the display does not even turn on.

I noticed if I set the frequency to 1mhz the message the controller is unresponsive does not shows up.

But it gets stuck in the message:  

I (350) RA8875: Enabling display.

`#
# Automatically generated file. DO NOT EDIT.
# Espressif IoT Development Framework (ESP-IDF) Project Configuration
#
CONFIG_IDF_CMAKE=y
CONFIG_IDF_TARGET="esp32"
CONFIG_IDF_TARGET_ESP32=y
CONFIG_IDF_FIRMWARE_CHIP_ID=0x0000

#
# SDK tool configuration
#
CONFIG_SDK_TOOLPREFIX="xtensa-esp32-elf-"
# CONFIG_SDK_TOOLCHAIN_SUPPORTS_TIME_WIDE_64_BITS is not set
# end of SDK tool configuration

#
# Build type
#
CONFIG_APP_BUILD_TYPE_APP_2NDBOOT=y
# CONFIG_APP_BUILD_TYPE_ELF_RAM is not set
CONFIG_APP_BUILD_GENERATE_BINARIES=y
CONFIG_APP_BUILD_BOOTLOADER=y
CONFIG_APP_BUILD_USE_FLASH_SECTIONS=y
# end of Build type

#
# Application manager
#
CONFIG_APP_COMPILE_TIME_DATE=y
# CONFIG_APP_EXCLUDE_PROJECT_VER_VAR is not set
# CONFIG_APP_EXCLUDE_PROJECT_NAME_VAR is not set
# CONFIG_APP_PROJECT_VER_FROM_CONFIG is not set
CONFIG_APP_RETRIEVE_LEN_ELF_SHA=16
# end of Application manager

#
# Bootloader config
#
CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_SIZE=y
# CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_DEBUG is not set
# CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_PERF is not set
# CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_NONE is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_NONE is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_ERROR is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_WARN is not set
CONFIG_BOOTLOADER_LOG_LEVEL_INFO=y
# CONFIG_BOOTLOADER_LOG_LEVEL_DEBUG is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_VERBOSE is not set
CONFIG_BOOTLOADER_LOG_LEVEL=3
# CONFIG_BOOTLOADER_VDDSDIO_BOOST_1_8V is not set
CONFIG_BOOTLOADER_VDDSDIO_BOOST_1_9V=y
# CONFIG_BOOTLOADER_FACTORY_RESET is not set
# CONFIG_BOOTLOADER_APP_TEST is not set
CONFIG_BOOTLOADER_WDT_ENABLE=y
# CONFIG_BOOTLOADER_WDT_DISABLE_IN_USER_CODE is not set
CONFIG_BOOTLOADER_WDT_TIME_MS=9000
# CONFIG_BOOTLOADER_APP_ROLLBACK_ENABLE is not set
# CONFIG_BOOTLOADER_SKIP_VALIDATE_IN_DEEP_SLEEP is not set
CONFIG_BOOTLOADER_RESERVE_RTC_SIZE=0
[... 1571 lines cut ...]
# CONFIG_TCP_OVERSIZE_DISABLE is not set
CONFIG_UDP_RECVMBOX_SIZE=6
CONFIG_TCPIP_TASK_STACK_SIZE=3072
CONFIG_TCPIP_TASK_AFFINITY_NO_AFFINITY=y
# CONFIG_TCPIP_TASK_AFFINITY_CPU0 is not set
# CONFIG_TCPIP_TASK_AFFINITY_CPU1 is not set
CONFIG_TCPIP_TASK_AFFINITY=0x7FFFFFFF
# CONFIG_PPP_SUPPORT is not set
CONFIG_ESP32_PTHREAD_TASK_PRIO_DEFAULT=5
CONFIG_ESP32_PTHREAD_TASK_STACK_SIZE_DEFAULT=3072
CONFIG_ESP32_PTHREAD_STACK_MIN=768
CONFIG_ESP32_DEFAULT_PTHREAD_CORE_NO_AFFINITY=y
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_0 is not set
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_1 is not set
CONFIG_ESP32_PTHREAD_TASK_CORE_DEFAULT=-1
CONFIG_ESP32_PTHREAD_TASK_NAME_DEFAULT="pthread"
CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ABORTS=y
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_FAILS is not set
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ALLOWED is not set
CONFIG_SUPPRESS_SELECT_DEBUG_OUTPUT=y
CONFIG_SUPPORT_TERMIOS=y
CONFIG_SEMIHOSTFS_MAX_MOUNT_POINTS=1
CONFIG_SEMIHOSTFS_HOST_PATH_MAX_LEN=128
# End of deprecated options
`

9ae8sdf76, 2021-04-13:
@C47D Any update or progress on trying to reproduce this issue?

C47D, 2021-04-14:
Hi @9ae8sdf76 , I haven't been able to reproduce it, nor getting up to speed with any other issue in the repo, got pretty busy with my day job.

vandiwa, 2021-06-02:
I'm kind of in the same boat, but I've found that if I start by initializing the display using the Arduino XGLCD library, and then without power cycling the display run the ESP-IDF code, the display works fine.  I've slowed down the setup process on both sides to compare, and I have rewritten the init to do what the XGLCD init routine does byte for byte, but I still can't get the display to startup and turn on the backlight if I start with the display in the off state.  I don't have a physical reset pin connected to the display in either situation, so it seems like the XGLCD library is doing something that my modified routine isn't, but I haven't been able to figure out what that is yet.

vandiwa, 2021-06-03:
Took a while, but what I finally figured out is that the spimode was wrong - it should be 3 for the ra8875, but the lvgl_spi_conf.h file (in my version) doesn't have a special case for the RA8875, so it gets set to mode 0 by default.  Changed it to 3 and now my init code (that very closely matches the XGLCD Arduino sequence) works.

closed 2021-07-11 by stale[bot]

amj2900, 2022-02-21:
Has anyone managed to fix this issue? I'm running into the same thing.

kyrpav, 2022-09-04:
What is the correct then setup for RA8875 and lvgl and esp32 . spi mode 3 , spi frequency 1Mhz and full duplex only MOSI to be set and reset pin?
Am i missing something cause i can not make it work

kethort, 2023-08-05:
You have to manually drive the CS line for the entirety of each command or data write via the SPI. You can do this using the pre/post callback functions in the ESP-IDF SPI driver. The 4-wire SPI configuration is full duplex; you can read status registers from the MISO line. Here is an example where I've used SPI to communicate with a LT7683 display which is very similar to the RA887x chips and the SPI communications should work the same: https://github.com/kethort/esp32-LT7683-FT5X06-SPI. The SPI clock can go up to 50MHz on those display drivers, I'm using standard VSPI pins so I can only clock it out at 40MHz max but it works and drives the display just fine. Make sure you wire the ground of the ESP32 to the display ground and keep the communication wires short. If the lcd_spi.c/h driver doesn't work for you, consider lowering the SPI speed to 26 or 20MHz.

D1sconnected, 2023-08-09:
@vandiwa Hey there, I'm stuck with RA8875 init routine too, working with Waveshare 7 Inch Display. Could you kindly provide your init code? That would help a lot!