lvgl/lv_esp_idf #249: Static text shows up on display everytime I reboot

VedantParanjape, 2021-01-11:
here's my code: https://pastebin.com/6EQgP0Dh

when I boot my board, it shows a static text "Text" on the upper left corner of the screen as follows. I tried clearing my screen, refreshing it. just doesn't work. Is this some bug ? Tested it on multiple screens too.

here's the picture and video exhibiting this
Video: https://streamable.com/8rxck1

[image]

kisvegabor, 2021-01-12:
What happens if you add `lv_refr_now(NULL)` to line 152?

BTW, the video not available :slightly_frowning_face:

VedantParanjape, 2021-01-12:
> What happens if you add `lv_refr_now(NULL)` to line 152?
> 
> BTW, the video not available slightly_frowning_face

https://streamable.com/6zpzsf Sorry, reuploaded the video, take a look

C47D, 2021-01-12:
I saw a lot of changes on the task infinite loop that should not be there, I am able to print the labels with no problem.

VedantParanjape, 2021-01-12:
> I saw a lot of changes on the task infinite loop that should not be there, I am able to print the labels with no problem.

Did you check that video, If I try to print anything before infinite loop, "Text" appears on screen. As soon it enters loop, that specific text disappears. If you need access to complete repository let me know, it's private as of now.

C47D, 2021-01-12:
I didn't, I think, @kisvegabor correct me if I'm wrong, `lv_task_handler` shouldn't be called on the same loop than `lv_tick_inc`, so the guiTask infinite loop should look like this:

```c
    while (1) {
        /* Delay 1 tick (assumes FreeRTOS tick is 10ms */
        vTaskDelay(pdMS_TO_TICKS(10));

        /* Try to take the semaphore, call lvgl related function on success */
        if (pdTRUE == xSemaphoreTake(xGuiSemaphore, portMAX_DELAY)) {
            lv_task_handler();
            xSemaphoreGive(xGuiSemaphore);
       }
    }
```

The tick callback like this:
```c
static void lv_tick_task(void *arg) {
    (void) arg;

    lv_tick_inc(LV_TICK_PERIOD_MS);
}
```

And you should be able to update the labels using an lv_task.

> Did you check that video, If I try to print anything before infinite loop, "Text" appears on screen. As soon it enters loop, that specific text disappears. If you need access to complete repository let me know, it's private as of now.
I think "Text" is being written by default to the display, I am not able to exactly pin point how.

I didn't because of the issues mentioned above, let's wait for @kisvegabor confirmation, I might be wrong.

embeddedt, 2021-01-12:
I'm not @kisvegabor but it looks correct to me.

kisvegabor, 2021-01-13:
I'm @kisvegabor and to be honest, I have never seen serious issues with adding them to the same loop. :smile: 

The only drawback is that it makes animations inaccurate, but it should work.

VedantParanjape, 2021-01-13:
@kisvegabor @embeddedt @C47D Yes, agreed. code is shit, I am new to lvgl, learning it, and docs isn't that great :( 

I still can't understand why does "Text" appear on the screen out of nowhere, is it some default routine ?

embeddedt, 2021-01-13:
It's the default text a label displays when it is created. This would normally only happen if `lv_label_set_text` wasn't being called, or failed for some reason. Perhaps try [turning on logging](https://docs.lvgl.io/latest/en/html/porting/log.html) if you haven't already. See if anything interesting comes up.

VedantParanjape, 2021-01-14:
> It's the default text a label displays when it is created. This would normally only happen if `lv_label_set_text` wasn't being called, or failed for some reason. 

I was asking exactly this, that was the whole point of the issue 😆 

```c
    lv_obj_t *label1 = lv_label_create(scr, NULL);
    lv_obj_t *label2 = lv_label_create(scr, NULL);
    lv_label_set_text(label1, " ");
    lv_label_set_text(label2, " ");
```

Did this and all my problems went away, thanks for the help @C47D @embeddedt @kisvegabor :)

C47D, 2021-01-14:
@VedantParanjape it does look as an interesting board! and no problem, sorry for taking this long to solve it.

closed 2021-01-14 by C47D