lvgl/lv_esp_idf #253: Random crashes due to small changes in event handler

VedantParanjape, 2021-02-01:
The esp32 crashes randomly when I remove call to the external function `action_menu_page()` from the event handler. I tried doing this, I can add any function which does random things and stop it from crashing.

Here is the complete code: https://pastebin.com/im4nTZ7w

If line 106 is uncommented, it works fine.

This is my main code
```c
void app_main()
{
    init_load_switch();
    init_switches();
    activate_load_switch();

    config_adc1();
    characterize_adc1();

    xTaskCreatePinnedToCore(wifi_ntp_task, "ntp", 4096 * 2, NULL, 0, NULL, 0);
    xTaskCreatePinnedToCore(lvgl_gui_task, "gui", 4096 * 2, NULL, 0, NULL, 1);
}
```

```c
static void switch_event_handler_cb(lv_obj_t *obj, lv_event_t event)
{
    uint32_t *key_id = NULL;
    if (event == LV_EVENT_PRESSED)
    {
        ESP_LOGI("button_cb", "Clicked button - enter");
        // action_menu_page();
        int stub = 0;
    }
    else if (event == LV_EVENT_KEY)
    {
        key_id = (uint32_t *)lv_event_get_data();
        ESP_LOGI("button_cb", "Clicked button - %u", *key_id);

        if (*key_id == LV_KEY_UP)
        {
            menu_id = ((menu_id - 1) % 4 + 4) % 4;
        }
        else if (*key_id == LV_KEY_DOWN)
        {
            menu_id = ((menu_id + 1) % 4 + 4) % 4;
        }
        set_menu_page();
    }
}
```
**ESP-IDF version**
V4.2-release

**Development kit used**
ESP32 custom board

**Development machine OS**
Ubuntu 20.04

**Compilation warnings/errors (if available)**
```bash
I (693) lvgl_helpers: Setting I2C master configuration...
I (703) lvgl_helpers: Installing I2C master driver...
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x400decb0  PS      : 0x00060630  A0      : 0x800edaab  A1      : 0x3ffaf600  
0x400decb0: lv_draw_label at /home/vedant/Programming/projects/open-authenticator/open-authenticator-app/build/../components/lv_port_esp32/components/lvgl/src/lv_draw/lv_draw_label.c:126

A2      : 0x3ffb1248  A3      : 0x3ffaf77c  A4      : 0x3ffaf75c  A5      : 0x00000000  
A6      : 0x00000000  A7      : 0x3ffaf75c  A8      : 0x0000ffff  A9      : 0x0000ffff  
A10     : 0x3ffb1248  A11     : 0x00000000  A12     : 0x0000808e  A13     : 0x0000808e  
A14     : 0x3ffb6a78  A15     : 0x00000000  SAR     : 0x0000001a  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace:0x400decad:0x3ffaf600 0x400edaa8:0x3ffaf700 0x400dc13e:0x3ffaf7c0 0x400dc1af:0x3ffaf810 0x400dc1ed:0x3ffaf860 0x400dc510:0x3ffaf880 0x400dc6a9:0x3ffaf900 0x400dc792:0x3ffaf930 0x400dc7f3:0x3ffaf950 0x400e7b35:0x3ffaf980 0x400e7c20:0x3ffaf9a0 0x400d4f97:0x3ffaf9c0 0x400d4614:0x3ffaf9e0 0x400d467f:0x3ffafa00 0x40088bc5:0x3ffafa20
0x400decad: lv_draw_label at /home/vedant/Programming/projects/open-authenticator/open-authenticator-app/build/../components/lv_port_esp32/components/lvgl/src/lv_draw/lv_draw_label.c:122

0x400edaa8: lv_label_design at /home/vedant/Programming/projects/open-authenticator/open-authenticator-app/build/../components/lv_port_esp32/components/lvgl/src/lv_widgets/lv_label.c:1311

0x400dc13e: lv_refr_obj at /home/vedant/Programming/projects/open-authenticator/open-authenticator-app/build/../components/lv_port_esp32/components/lvgl/src/lv_core/lv_refr.c:699 (discriminator 1)

0x400dc1af: lv_refr_obj at /home/vedant/Programming/projects/open-authenticator/open-authenticator-app/build/../components/lv_port_esp32/components/lvgl/src/lv_core/lv_refr.c:739

0x400dc1ed: lv_refr_obj_and_children at /home/vedant/Programming/projects/open-authenticator/open-authenticator-app/build/../components/lv_port_esp32/components/lvgl/src/lv_core/lv_refr.c:641

0x400dc510: lv_refr_area_part at /home/vedant/Programming/projects/open-authenticator/open-authenticator-app/build/../components/lv_port_esp32/components/lvgl/src/lv_core/lv_refr.c:572

0x400dc6a9: lv_refr_area at /home/vedant/Programming/projects/open-authenticator/open-authenticator-app/build/../components/lv_port_esp32/components/lvgl/src/lv_core/lv_refr.c:482 (discriminator 2)

0x400dc792: lv_refr_areas at /home/vedant/Programming/projects/open-authenticator/open-authenticator-app/build/../components/lv_port_esp32/components/lvgl/src/lv_core/lv_refr.c:404

0x400dc7f3: _lv_disp_refr_task at /home/vedant/Programming/projects/open-authenticator/open-authenticator-app/build/../components/lv_port_esp32/components/lvgl/src/lv_core/lv_refr.c:203

0x400e7b35: lv_task_exec at /home/vedant/Programming/projects/open-authenticator/open-authenticator-app/build/../components/lv_port_esp32/components/lvgl/src/lv_misc/lv_task.c:409 (discriminator 1)

0x400e7c20: lv_task_handler at /home/vedant/Programming/projects/open-authenticator/open-authenticator-app/build/../components/lv_port_esp32/components/lvgl/src/lv_misc/lv_task.c:142

0x400d4f97: lv_tick_task at /home/vedant/Programming/projects/open-authenticator/open-authenticator-app/build/../main/lvgl_gui.c:70

0x400d4614: timer_process_alarm at /home/vedant/esp/esp-idf/components/esp_timer/src/esp_timer.c:310

0x400d467f: timer_task at /home/vedant/esp/esp-idf/components/esp_timer/src/esp_timer.c:331

0x40088bc5: vPortTaskWrapper at /home/vedant/esp/esp-idf/components/freertos/xtensa/port.c:143

```

**If possible, copy the compilation log into a file and attach it here**

kisvegabor, 2021-02-03:
Does it make any difference if you increase the LVGL task's stack size?

kisvegabor, 2021-02-03:
And does it always stop at `lv_draw_label.c:126`?

VedantParanjape, 2021-02-05:
> And does it always stop at `lv_draw_label.c:126`?

Yes, I tried several different ways, changing the function content, changing the location. But it always seems to crash with this error.

VedantParanjape, 2021-02-05:
> Does it make any difference if you increase the LVGL task's stack size?

Aha, Let me try with this. What's the recommended size?

embeddedt, 2021-02-05:
Try 4 kilobytes if you have the RAM.

VedantParanjape, 2021-02-07:
@embeddedt Surprisingly it worked fine this time. This is very wierd. Earlier Stack depth was 8kb, changed it to 4kb and it worked fine.

closed 2021-02-13 by VedantParanjape