lvgl/lv_esp_idf #256: ESP32-S2 Projects cannot use pin numbers higher than 39 when configuring through menuconfig

jockm, 2021-02-03:
The ESP32-S2 has more available GPIOs than the ESP32, but menuconfig only allows assigning pins 0-39.

It remains clear that no testing or consideration of the -S2 chip has been given.

Again I must stress the need to publicly document that the -S2 chip is currently unsupported

C47D, 2021-02-03:
Hi,

> The ESP32-S2 has more available GPIOs than the ESP32, but menuconfig only allows assigning pins 0-39.

This can be fixed quite "easily".

> It remains clear that no testing or consideration of the -S2 chip has been given.

There's an closed issue report about somebody running this repo on the ESP32-S2, but I don't have any ESP32-S2 based hardware available to test it.

> Again I must stress the need to publicly document that the -S2 chip is currently unsupported

From the issue I mentioned above it seems like it's kind of working but we could mention the support is in beta, thanks for the suggestion.

jockm, 2021-02-03:
Fair enough but you don’t need hardware to do a `idf.py set-target esp32s2` And see if things compile, or to glance at the data sheet and contemplate if the config needs to be updated

The -S2 has been out for almost a year now and the example project can’t compile, and there isn’t so much as a note in the readme to warn people

C47D, 2021-02-03:
True that, I will update the readme to make that clear.

Hopefully will get a esp32-s2, they seem interesting.

C47D, 2021-02-04:
Hi @jockm,

I've just seen the other issues you've opened, and I think this is the easiest one to solve, according to [this](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/api-reference/peripherals/gpio.html#overview) reference, the ESP32-S2 has 43 GPIOs. Is this the information you would like to see when using that target?

jockm, 2021-02-04:
@C47D Yes that is correct, when configuring for the ESP32-S2, menuconfig should allow for all GPIO pin numbers for that architecture

C47D, 2021-02-04:
@jockm cool, thanks for the quick reply, I'm a bit busy with my day job so I can work on this at my free time, expect an update before the weekend.