lvgl/lv_esp_idf #257: Cannot Find  Components config->LVGL Configuration

BH2SEB, 2021-02-04:
After I clone this repo , I cannot find LVGL Configuration in menuconfig while LVGL TFT Display Configuration and LVGL TOUCH Configuration is on the list.

C47D, 2021-02-04:
Hi @BH2SEB,

I guess you're using `make menuconfig` to launch the menuconfig menu? I will check your proposal and include it.

C47D, 2021-02-04:
Hi,

> After I clone this repo , I cannot find LVGL Configuration in menuconfig while LVGL TFT Display Configuration and LVGL TOUCH Configuration is on the list.

This is kind of expected because the main LVGL repository doesn't includes an `component.mk` file, this file is used when using make to build the project.

@kisvegabor Do you think it's possible to add the `component.mk` file in LVGL to support make based commands? All the ESP32 support were assuming the users using idf commands instead.

kisvegabor, 2021-02-04:
> @kisvegabor Do you think it's possible to add the component.mk file in LVGL to support make based commands? All the ESP32 support were assuming the users using idf commands instead.

Yes! We can add `if`s to turn on/off the ESP related section.

C47D, 2021-02-10:
@BH2SEB Can you try with the current master? I've added the components.mk file into LVGL

tlechman49, 2021-02-15:
I'm experiencing a similar problem I think. When I bring the repo into my project using platformio or as a submodule I'm getting a lot of errors in compilation with trouble finding lv_conf.h and when I use menuconfig it doesn't find any of the LVGL related components. Seems like it can't find the kconfig files. 
I'd prefer to use platformio lib_deps which look like this inside plaformio.ini:
```
lib_deps = 
	lvgl=https://github.com/lvgl/lvgl.git#85c75c2f34ddde5ad0301b770255f41c8ed4c72a
	lvgl_esp32_drivers=https://github.com/lvgl/lvgl_esp32_drivers.git#6b08c3fcce0e12b44a171ef88059b52efc0269ad
```
But I've also pulled in as a submodule to lib: lib/lvgl and lib/lvgl_esp32_drivers

C47D, 2021-02-15:
Hi @tlechman49,
Before we added the Kconfig files into LVGL and separated the drivers into it's own repo, we suggested to use this repo as a submodule. Correct me if I'm wrong, are you now using both LVGL and lvgl_esp32_drivers as submodules and not having issues?

I don't really have experience with Platformio, let me investigate how they manage dependencies.

tlechman49, 2021-02-15:
I was previously using this repo as a submodule. I'm now hoping to bring the repos for LVGL and lvgl_esp32_drivers in as lib_deps with pio but if that can't be done I can always bring them in as submodules similar to whats in this repo's readme. 

I know pio stores the lib_deps libraries in .pio/libdeps/$ENVIRONMENT_NAME/$LIB_DEP_NAME.

tlechman49, 2021-02-15:
I guess I misread your statement.. 

I was previously not having issues but I would like to upgrade to the latest version of the repo and am now having issues. When I pulled in the two repos as described in the readme and also when I pulled them in with lib_deps on a separate attempt I received errors during compilation regarding lv_conf.h.

C47D, 2021-02-16:
Hi,

Before this repo used the `lv_conf.h` file to configure LVGL, it is now configured by using the menuconfig tool, when using menuconfig it doesn't needs the `lv_conf.h` file. So, when using LVGL as submodule (and it includes the Kconfig file) you should be able to use menuconfig.

Maybe we should update the README to note this.

BH2SEB, 2021-02-17:
> @BH2SEB Can you try with the current master? I've added the components.mk file into LVGL

Thank you, the make menuconfig tool  is running normally now.

C47D, 2021-02-17:
@tlechman49 Can you open a new issue with your problems?

closed 2021-03-05 by BH2SEB