lvgl/lv_esp_idf #260: fatal error: lvgl/lvgl.h: No such file or directory

1404360768, 2021-02-12:
We use GitHub issues for development related discussions.
Please use the [forum](https://forum.littlevgl.com/) to ask questions.

**Describe the issue**
Using git clone --recursive https://github.com/lvgl/lv_port_esp32.git clone the project without making any changes, directly Compile after the following error, and the location is not only this place, all references to the file location have reported errors, I try to add absolute path, but this is very troublesome, is there an easy way?

**Code to reproduce the issue**
git clone --recursive https://github.com/lvgl/lv_port_esp32.git 
**Expected Results**

**Actual Results**

**ESP32 Chip version**
chip revision: 1
esp32: SPI Speed      : 40MHz
esp32: SPI Mode       : DIO
esp32: SPI Flash Size : 4MB
**ESP-IDF version**
ESP-IDF v4.2-238-g8cd16b60f
**Development kit used**
ESP-IDF
**Development machine OS**
Ubunt20.04
**Compilation warnings/errors (if available)**

**If possible, copy the compilation log into a file and attach it here**
xiang@xiang:~/esp/lv_port_esp32$ make -j32
Toolchain path: /home/xiang/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: esp-2020r3
Compiler version: 8.4.0
Python requirements from /home/xiang/esp/esp-idf/requirements.txt are satisfied.

App "lvgl-demo" version: v2.0-39-gb53d406
CC build/lvgl_esp32_drivers/lvgl_helpers.o
CC build/lvgl_tft/disp_spi.o
CC build/lvgl_tft/st7796s.o
CC build/lvgl_touch/tp_i2c.o
CC build/main/main.o
CC build/lvgl_touch/touch_driver.o
CC build/mdns/mdns_console.o
CC build/lvgl_touch/xpt2046.o
CC build/lvgl_touch/ft6x36.o
CC build/mdns/mdns_networking.o
In file included from /home/xiang/esp/lv_port_esp32/components/lvgl_esp32_drivers/lvgl_helpers.h:18,
                 from /home/xiang/esp/lv_port_esp32/components/lvgl_esp32_drivers/lvgl_helpers.c:10:
/home/xiang/esp/lv_port_esp32/components/lvgl_esp32_drivers/lvgl_tft/disp_driver.h:18:10: fatal error: lvgl/lvgl.h: No such file or directory
 #include "lvgl/lvgl.h"
          ^~~~~~~~~~~~~
compilation terminated.
CXX build/nvs_flash/src/nvs_cxx_api.o
CC build/nghttp/nghttp2/lib/nghttp2_http.o
CC build/newlib/heap.o
CXX build/nvs_flash/src/nvs_handle_simple.o
In file included from /home/xiang/esp/lv_port_esp32/components/lvgl_esp32_drivers/lvgl_touch/touch_driver.c:5:
/home/xiang/esp/lv_port_esp32/components/lvgl_esp32_drivers/lvgl_touch/touch_driver.h:20:10: fatal error: lvgl/lvgl.h: No such file or directory
 #include "lvgl/lvgl.h"
          ^~~~~~~~~~~~~
compilation terminated.
CC build/lvgl_tft/ili9488.o
CC build/mdns/mdns.o
CC build/lvgl_tft/ili9486.o
CC build/lvgl_tft/ili9481.o
CC build/lvgl_tft/il3820.o
CC build/lvgl_tft/st7735s.o
CC build/nghttp/nghttp2/lib/nghttp2_version.o
make[1]: *** [/home/xiang/esp/esp-idf/make/component_wrapper.mk:292：lvgl_helpers.o] 错误 1
make: *** [/home/xiang/esp/esp-idf/make/project.mk:635：component-lvgl_esp32_drivers-build] 错误 2
make: *** 正在等待未完成的任务....
In file included from /home/xiang/esp/lv_port_esp32/components/lvgl_esp32_drivers/lvgl_tft/st7796s.c:9:
/home/xiang/esp/lv_port_esp32/components/lvgl_esp32_drivers/lvgl_tft/st7796s.h:22:10: fatal error: lvgl/lvgl.h: No such file or directory
[... 224 lines cut ...]
CC build/mbedtls/mbedtls/library/error.o
CC build/mbedtls/mbedtls/library/md5.o
CC build/mbedtls/mbedtls/library/pk.o
CC build/mbedtls/mbedtls/library/sha1.o
CC build/mbedtls/mbedtls/library/x509_csr.o
AR build/mdns/libmdns.a
CC build/mbedtls/mbedtls/library/ssl_cli.o
CC build/mbedtls/port/mbedtls_debug.o
CC build/mbedtls/port/esp_timing.o
CC build/mbedtls/port/esp_aes_xts.o
CC build/mbedtls/port/esp_sha.o
CC build/mbedtls/port/esp_bignum.o
CC build/mbedtls/port/net_sockets.o
CC build/mbedtls/port/esp_hardware.o
CC build/mbedtls/port/esp32/esp_sha1.o
CC build/mbedtls/port/esp_mem.o
CC build/mbedtls/port/esp32/esp_sha256.o
CC build/mbedtls/port/esp32/sha.o
CC build/mbedtls/port/esp32/bignum.o
CC build/mbedtls/port/esp32/aes.o
CC build/mbedtls/port/esp32/esp_sha512.o
CC build/mbedtls/esp_crt_bundle/esp_crt_bundle.o
EMBED build/mbedtls/x509_crt_bundle.bin.o
AR build/mbedtls/libmbedtls.a
xiang@xiang:~/esp/lv_port_esp32$

Mair, 2021-02-14:
duplicate of [778219950](https://github.com/lvgl/lvgl_esp32_drivers/issues/36#issuecomment-778219950)
@C47D also reported by @coztemur as still being an issue

C47D, 2021-02-14:
The problem seems to be the symbol `LV_LVGL_H_INCLUDE_SIMPLE` not being defined, in LVGL CMakeLists.txt file we're defining  `LV_CONF_INCLUDE_SIMPLE` instead. @kisvegabor do you remember why did we kept `LV_CONF_INCLUDE_SIMPLE` instead of defining `LV_LVGL_H_INCLUDE_SIMPLE`?

C47D, 2021-02-14:
@Mair, are you using make instead of idf?

C47D, 2021-02-14:
Can you try check if 3154e57 solves the issue?

Mair, 2021-02-14:
@C47D no I'm using idf.py build. I was working with @coztemur who found it. then I tried it and replicated it.

I Just did a fresh `git clone --recurse-submodules https://github.com/lvgl/lv_port_esp32.git` in a clean directory then did an `idf.py menuconfig`. and changed the display. then 'idf.py build.
```
 -c ../components/lvgl_esp32_drivers/lvgl_tft/FT81x.c
In file included from ../components/lvgl_esp32_drivers/lvgl_tft/EVE.h:39,
                 from ../components/lvgl_esp32_drivers/lvgl_tft/FT81x.c:8:
../components/lvgl_esp32_drivers/lvgl_tft/EVE_config.h:41:10: fatal error: lvgl/lvgl.h: No such file or directory
 #include "lvgl/lvgl.h"
```

Mair, 2021-02-14:
re #3154e57 I'll try it out later. but you can easily replicate it by trying a fresh pull ☝️

C47D, 2021-02-15:
@Mair I was able to reproduce the issue (I wasn't at first because I was using the current master of `lvgl_esp32_drivers`), just updated the `lvgl_esp32_drivers` submodule, you should be able to compile the project.

1404360768, 2021-02-20:
Sorry, I still can't compile successfully, how to add lvgl alone in the existing project?

C47D, 2021-02-20:
@1404360768 You should be able to include LVGL into an existing project in two ways:
1. If your project is a git repository, you can add LVGL as a submodule with `git submodule add https://github.com/lvgl/lvgl.git`
2. If your project isn't a git repo you can add LVGL with `git clone https://github.com/lvgl/lvgl.git`

Did you updated your copy of this project? I'm able to compile it with make and idf

C47D, 2021-02-20:
@tore-espressif Can you help me out with this one? The `lvgl_esp_drivers` `component.mk` looks like this:

```
COMPONENT_SRCDIRS := .

COMPONENT_ADD_INCLUDEDIRS := .
COMPONENT_ADD_INCLUDEDIRS += lvgl_tft
COMPONENT_ADD_INCLUDEDIRS += lvgl_touch

CFLAGS += -DLV_LVGL_H_INCLUDE_SIMPLE
```

Please correct me if I'm wrong but `LV_LVGL_H_INCLUDE_SIMPLE` is being defined for this component, it should look for `lvgl.h` instead of `lvgl/lvgl.h`.

tore-espressif, 2021-02-22:
Hello,

I'm sorry to hear that you are still struggling with what we considered solved.

Nevertheless, I could not reproduce the problem on any IDF4.x version with CMake (clean clone). I'm on Windows working with Wrover-kit and Kaluga-kit.

Please note, that when you are updating your local project with remote changes with `git pull`, it won't automatically update your submodules, which can be a issue as the problem was fixed recently in a `lvgl_esp32_drivers` submodule.

Last but not least, this repository is not set-up for use with plain `GNU Make` (not maintained anymore), I can see that you are using ESP-IDF version 4.x so I'd encourage you to use `idf.py build / reconfigure / fullclean ...` commands. If there are more users that must use IDF  3.x we might update the makefiles too.

Just compare the complexity of these two files:
https://github.com/lvgl/lvgl_esp32_drivers/blob/462f5eea46de880b65a548f4ec557ed7a69045cd/lvgl_tft/CMakeLists.txt
https://github.com/lvgl/lvgl_esp32_drivers/blob/462f5eea46de880b65a548f4ec557ed7a69045cd/lvgl_tft/component.mk
and analogically the same in `lvgl_touch` directory.

@1404360768 Could you confirm that you can build this project successfully? I can  help you with adding lvgl to an existing project then. 

@C47D We should decide whether to have makefiles in the repo and maintain them properly or just delete them. Currently it is not possible to compile it with Make without touch support as the [makefile](https://github.com/lvgl/lvgl_esp32_drivers/blob/462f5eea46de880b65a548f4ec557ed7a69045cd/lvgl_tft/component.mk) forces all source files to be compiled even when you select no touch support in menuconfig. Then it fails on undefined configuration as you can see in build log https://github.com/lvgl/lv_port_esp32/issues/260#issue-807042403

1404360768, 2021-02-22:
> Hello,
> 
> I'm sorry to hear that you are still struggling with what we considered solved.
> 
> Nevertheless, I could not reproduce the problem on any IDF4.x version with CMake (clean clone). I'm on Windows working with Wrover-kit and Kaluga-kit.
> 
> Please note, that when you are updating your local project with remote changes with `git pull`, it won't automatically update your submodules, which can be a issue as the problem was fixed recently in a `lvgl_esp32_drivers` submodule.
> 
> Last but not least, this repository is not set-up for use with plain `GNU Make` (not maintained anymore), I can see that you are using ESP-IDF version 4.x so I'd encourage you to use `idf.py build / reconfigure / fullclean ...` commands. If there are more users that must use IDF 3.x we might update the makefiles too.
> 
> Just compare the complexity of these two files:
> https://github.com/lvgl/lvgl_esp32_drivers/blob/462f5eea46de880b65a548f4ec557ed7a69045cd/lvgl_tft/CMakeLists.txt
> https://github.com/lvgl/lvgl_esp32_drivers/blob/462f5eea46de880b65a548f4ec557ed7a69045cd/lvgl_tft/component.mk
> and analogically the same in `lvgl_touch` directory.
> 
> @1404360768 Could you confirm that you can build this project successfully? I can help you with adding lvgl to an existing project then.
> 
> @C47D We should decide whether to have makefiles in the repo and maintain them properly or just delete them. Currently it is not possible to compile it with Make without touch support as the [makefile](https://github.com/lvgl/lvgl_esp32_drivers/blob/462f5eea46de880b65a548f4ec557ed7a69045cd/lvgl_tft/component.mk) forces all source files to be compiled even when you select no touch support in menuconfig. Then it fails on undefined configuration as you can see in build log [#260 (comment)](https://github.com/lvgl/lv_port_esp32/issues/260#issue-807042403)

Thank you so much，I'll give it a try idf.py build

C47D, 2021-02-22:
@tore-espressif I did took a look into the esp-idf components and saw that we can check for the generated `CONFIG_` symbols, so maybe we can check for them and include the appropriate source files.

This week I will try to fix both Kconfig files (tft and touch) and other issues like this one, because we want to improve the displays code base.

1404360768, 2021-02-23:
@C47D @tore-espressif At the moment I'm successfully compiled with idf.pyidf.py build, but my original project used idf.pyidf.py build to report errors, couldn't find the relevant SDK files, and didn't make any mistakes with make, which made me wonder, sorry I'm still a novle, so I still hope I can build the project with make.

C47D, 2021-02-23:
@tore-espressif I'm working on a new branch to try get make builds done properly, I'm not familiar with this build system when using the ESP-IDF, can you please tell me if this commit goes in the right direction? https://github.com/lvgl/lvgl_esp32_drivers/commit/37715bc2b71e73faa5a07809bc0a5c1052d9c5c1

tore-espressif, 2021-02-23:
@1404360768 We are currently working on enhancing the first time user's experience, so if you are not confident enough to dig into the build system, you can wait a little and we will try to make it better.

_More info:_ ATM, when you add `lvgl_esp32_drivers` to your project (project_root/components folder), ESP-IDF won't recognize it as a component automatically; you must add it to its components extra directory, see this line: https://github.com/lvgl/lv_port_esp32/blob/master/CMakeLists.txt#L6

@C47D can you open a draft PR for this? We can discuss it in a separate thread, thanks!

C47D, 2021-02-23:
@tore-espressif Done, I've opened PRs on both this and lvgl_esp32_drivers.

tore-espressif, 2021-02-23:
Can we close this issue then?

1404360768, 2021-02-24:
Yes, thank you very much for your help

closed 2021-02-24 by 1404360768

Bouvierly, 2023-12-19:
In Arduino, there is always an error message that goes like this: "In file included from c: \ Users \ PL \ AppData \ Local \ Temp.arduinoIDE unsaved2023118-20692-17qr84m. 5xjm \ schechec8b \ libraries \ lvgl_esp32_drivers master/lvgl_helpers. h: 18,",

From C: \ Users \ PL \ AppData \ Local \ Temp.arduinoIDE unsaved20231111-10916 brtw2s. xbx16 \ fetch_ Dec19a \ sketch_ Dec19a. ino: 17:

c: \ Users \ PL \ AppData \ Local \ Temp.arduinoIDE unsaved2023118-20692-17qr84m. 5xjm \ fetch_ Dec8b \ libraries \ lvgl_ Esp32_ Drivers master/lvgl_ Tft/disp_ Driver. h:18:10: fatal error: lvgl/lvgl. h: No such file or directory

#Include "lvgl/lvgl. h"

^~~~~~~~~~~~~

Compilation terminated

Exit status 1

"Compilation error: exit status 1"

But I have this file, can anyone help me
in arduino