lvgl/lv_esp_idf #261: 运行demo有样式，但颜色不对

DylanMingan, 2021-02-13:
**Describe the issue**
使用的硬件是ESP32-S2-Saola-1开发板，外接2.4寸tft（tft），4线spi，跑默认的demo（换了两个默认主题）有样式，但颜色不对。
在lvgl设置里面的颜色选择16位RGB565，勾选sawp 2 bytes of rgb565..
[image]
[image]

**Code to reproduce the issue**
直接github下载，配置了io口，没有改动过代码

**Actual Results**
可以跑demo，但是颜色不对

**ESP32 Chip version**
esp32s2

**ESP-IDF version**
idf 4.2

**Development kit used**
Vscode+esp插件

**Development machine OS**
WIN10

**终端消息**
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6100,len:0x4
load:0x3ffe6104,len:0x1860
load:0x4004c000,len:0x1688
load:0x40050000,len:0x20fc
SHA-256 comparison failed:
Calculated: 9f1644e4e7208e4cab15ca0ca4f2763e9f734e27df2b654b7c92ca60541546cc
Expected: 4dd45dda9d7e0363a2d9ad6d8a78b1e213e3fb203ab75f0567ec193235447641
Attempting to boot anyway...
entry 0x4004c324
I (64) boot: ESP-IDF v4.2-238-g8cd16b60f-dirty 2nd stage bootloader
I (64) boot: compile time 09:47:57
I (64) boot: chip revision: 0
I (68) boot.esp32s2: SPI Speed      : 80MHz
I (72) boot.esp32s2: SPI Mode       : DIO
I (77) boot.esp32s2: SPI Flash Size : 4MB
I (82) boot: Enabling RNG early entropy source...
I (87) boot: Partition Table:
I (91) boot: ## Label            Usage          Type ST Offset   Length
I (98) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (106) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (113) boot:  2 factory          factory app      00 00 00010000 00100000
I (121) boot: End of partition table
I (125) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f000020 size=0x0b8c8 ( 47304) map
I (144) esp_image: segment 1: paddr=0x0001b8f0 vaddr=0x3ffbe730 size=0x01f18 (  7960) load
I (146) esp_image: segment 2: paddr=0x0001d810 vaddr=0x40024000 size=0x00404 (  1028) load
0x40024000: _WindowOverflow4 at D:/Workspace/esp-idf/components/freertos/xtensa/xtensa_vectors.S:1730

I (152) esp_image: segment 3: paddr=0x0001dc1c vaddr=0x40024404 size=0x023fc (  9212) load
I (163) esp_image: segment 4: paddr=0x00020020 vaddr=0x40080020 size=0x46318 (287512) map
0x40080020: _stext at ??:?

I (230) esp_image: segment 5: paddr=0x00066340 vaddr=0x40026800 size=0x07f28 ( 32552) load
0x40026800: esp_flash_init at D:/Workspace/esp-idf/components/spi_flash/esp_flash_api.c:164

I (244) boot: Loaded app from partition at offset 0x10000
I (244) boot: Disabling RNG early entropy source...
I (245) cache: Instruction cache        : size 8KB, 4Ways, cache line size 32Byte
I (252) cpu_start: Pro cpu up.
I (256) cpu_start: Application information:
I (261) cpu_start: Project name:     lvgl-demo
I (266) cpu_start: App version:      v2.0-39-gb53d406
I (272) cpu_start: Compile time:     Feb 13 2021 12:44:22
I (278) cpu_start: ELF file SHA256:  df8457e904f693b7...
I (284) cpu_start: ESP-IDF:          v4.2-238-g8cd16b60f-dirty
I (290) cpu_start: Single core mode
I (295) heap_init: Initializing. RAM available for dynamic allocation:
I (302) heap_init: At 3FFC9638 len 000329C8 (202 KiB): DRAM
I (308) heap_init: At 3FFFC000 len 00003A10 (14 KiB): DRAM
I (314) cpu_start: Pro cpu start user code
I (373) spi_flash: detected chip: generic
I (373) spi_flash: flash io: dio
I (374) cpu_start: Starting scheduler on PRO CPU.
D:/Workspace/esp-idf/components/freertos/tasks.c:1070 (prvAddNewTaskToReadyList)- assert failed!
I (405) lvgl_helpers: Display hor size: 320, ver size: 240
I (405) lvgl_helpers: Display buffer size: 12800
I (405) lvgl_helpers: Initializing SPI master for display
I (425) lvgl_helpers: Configuring SPI host  (1)
I (425) lvgl_helpers: MISO pin: 10, MOSI pin: 9, SCLK pin: 7, IO2/WP pin: -1, IO3/HD pin: -1
I (445) lvgl_helpers: Max transfer size: 25600 (bytes)
I (445) lvgl_helpers: Initializing SPI bus...
I (445) disp_spi: Adding SPI device
I (445) disp_spi: Clock speed: 20000000Hz, mode: 2, CS pin: 6
ST7789 initialization.
Enabling backlight.
I (865) st7789: Display orientation: PORTRAIT
I (865) st7789: 0x36 command value: 0xC0

C47D, 2021-02-14:
Hi,
I can't reproduce the issue with my 240x240 ST7889 based display, can you check the primary and secondary colors? and also try not swapping the RGB656 bytes.

DylanMingan, 2021-02-15:
Set primary color as **0xFF0000** （LV_COLOR_RED） and secondary color as **0x8000**（LV_COLOR_GREEN）
The following picture doesn't swap rgb565 bytes
[image]
The following picture swaps rgb565 bytes
[image]
Clearly，it should swap rgb565 bytes when using spi，otherwise there will be color confusion, such as white fonts with noise.
But, The primary color and second color seem wrong, Red changes to cyan, and green changes to purple. Is it because the bytes are messed up during spi transmission? 
please check related `sdkconfig.h` in `sdk.zip`

DylanMingan, 2021-02-15:
[Uploading sdk.zip…]()

C47D, 2021-02-15:
Thanks for attaching your config file, I will take a look at it. The inverted colors might be a display configuration issue, I remember other display (st7789 I think) being able to invert the colors with one command.

C47D, 2021-02-15:
Can you apply the following patch in `components/lvgl_esp32_drivers/lvgl_tft/st7789.c`

```diff
diff --git a/lvgl_tft/st7789.c b/lvgl_tft/st7789.c
index 2b5bd7e..8740b9d 100644
--- a/lvgl_tft/st7789.c
+++ b/lvgl_tft/st7789.c
@@ -66,7 +66,7 @@ void st7789_init(void)
         {ST7789_MADCTL, {0x00}, 1}, // Set to 0x28 if your display is flipped
         {ST7789_COLMOD, {0x55}, 1},

-#if ST7789_INVERT_COLORS == 1
+#if defined CONFIG_LV_DISPLAY_COLORS
                {ST7789_INVON, {0}, 0}, // set inverted mode
 #else
                {ST7789_INVOFF, {0}, 0}, // set non-inverted mode
```

DylanMingan, 2021-02-15:
Thank you very much for your opinion, after setting `{ST7789_INVOFF, {0}, 0}`, and swap rgb565, I successfully got the correct image.This experience gave me a better understanding of st7789, thanks again

C47D, 2021-02-15:
So, with the patch and swapping the RGB values the issue was solved? If so I can update the code with the patch.

DylanMingan, 2021-02-15:
yes, it works. but I just remove`  {ST7789_INVON, {0}, 0}` and add `{ST7789_INVOFF, {0}, 0}` in st7789' init functuion rather than use the patch.

C47D, 2021-02-15:
Ok, I will test the patch with my ST7789 based display. Feel free to close the issue when you feel like it.

closed 2021-02-16 by DylanMingan