lvgl/lv_esp_idf #266: Can't build with make

C47D, 2021-02-24:
We use GitHub issues for development related discussions.
Please use the [forum](https://forum.littlevgl.com/) to ask questions.

**Describe the issue**
Can't build using make, I got the error below.

**Code to reproduce the issue**
Clone the current master and run make.

**Expected Results**
Being able to build the project using make

**Actual Results**

**ESP32 Chip version**
`ESP32`

**ESP-IDF version**
4.2

**Development kit used**
None

**Development machine OS**
Ubuntu (WSL)

**Compilation warnings/errors (if available)**
```
Toolchain path: /home/dev/.espressif/tools/xtensa-esp32s2-elf/esp-2020r3-8.4.0/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-gcc
Toolchain version: esp-2020r3
Compiler version: 8.4.0
Python requirements from /home/dev/esp/esp-idf/requirements.txt are satisfied.
CC build/bootloader/esp_hw_support/compare_set.o
CC build/bootloader/esp_hw_support/cpu_util.o
CC build/bootloader/esp_hw_support/port/esp32/rtc_pm.o
CC build/bootloader/esp_hw_support/port/esp32/rtc_wdt.o
CC build/bootloader/esp_hw_support/port/esp32/rtc_init.o
CC build/bootloader/esp_hw_support/port/esp32/rtc_clk_init.o
CC build/bootloader/esp_hw_support/port/esp32/rtc_sleep.o
CC build/bootloader/esp_hw_support/port/esp32/rtc_time.o
CC build/bootloader/esp_hw_support/port/esp32/rtc_clk.o
AR build/bootloader/esp_hw_support/libesp_hw_support.a
CC build/bootloader/bootloader_support/src/bootloader_random_esp32.o
CC build/bootloader/bootloader_support/src/bootloader_efuse_esp32.o
CC build/bootloader/bootloader_support/src/bootloader_flash.o
CC build/bootloader/bootloader_support/src/flash_qio_mode.o
CC build/bootloader/bootloader_support/src/bootloader_init.o
CC build/bootloader/bootloader_support/src/bootloader_mem.o
/home/dev/esp/esp-idf/components/hal/esp32/include/hal/cpu_ll.h: Assembler messages:
/home/dev/esp/esp-idf/components/hal/esp32/include/hal/cpu_ll.h:60: Error: invalid register number (97) for 'wsr' instruction
make[2]: *** [/home/dev/esp/esp-idf/make/component_wrapper.mk:292: src/bootloader_mem.o] Error 1
make[1]: *** [/home/dev/esp/esp-idf/make/project.mk:635: component-bootloader_support-build] Error 2
make: *** [/home/dev/esp/esp-idf/components/bootloader/Makefile.projbuild:41: /home/dev/lv_port_esp32/build/bootloader/bootloader.bin] Error 2
```

**If possible, copy the compilation log into a file and attach it here**

C47D, 2021-02-25:
@tore-espressif Should I set the target chip before make? Like this: `TARGET=esp32 make`? I've never cross this bug before.

tore-espressif, 2021-02-26:
First line of your log says that you are using toolchain for esp32s2. It is only weird if you didn't set it yourself: esp32 should be default.
Running `idf.py set-target esp32` should fix the problem.

(Since ESP32S2 support was introduced in ESP-IDF 4.2, there is no alternative command for `make`, AFAIK)

C47D, 2021-02-26:
Yep, it's weird because I've never used the esp32-s2 target. I will try your suggestion.

C47D, 2021-02-27:
I was able to compile this project using make with the following patch https://github.com/lvgl/lvgl/pull/2108

tore-espressif, 2021-02-27:
> I was able to compile this project using make with the following patch [lvgl/lvgl#2108](https://github.com/lvgl/lvgl/pull/2108)

Yep, that was the last missing thing :)

C47D, 2021-02-27:
Just saw it got merged, will update the lvgl submodule in a bit.

closed 2021-02-27 by C47D