lvgl/lv_esp_idf #267: idf.py menuconfig fails

60north, 2021-02-24:
We use GitHub issues for development related discussions.
Please use the [forum](https://forum.littlevgl.com/) to ask questions.

hi

It seems to me that  lvgl_tft/Kconfig is missing, but referred to from higher level Kconfig

**Code to reproduce the issue**
Projects\lv_port_esp32>idf.py menuconfig

**Development kit used**

**Development machine OS**
Windows 10
**Compilation warnings/errors (if available)**
-- Building ESP-IDF components for target esp32
Traceback (most recent call last):
  File "C:\Users\fedor\Documents\Projects\esp-idf\tools\kconfig_new\confgen.py", line 584, in <module>
    main()
  File "C:\Users\fedor\Documents\Projects\esp-idf\tools\kconfig_new\confgen.py", line 230, in main
    config = kconfiglib.Kconfig(args.kconfig)
  File "C:\Users\fedor\Documents\Projects\esp-idf\tools\kconfig_new\kconfiglib.py", line 653, in __init__
    self._parse_block(None,           # end_token
  File "C:\Users\fedor\Documents\Projects\esp-idf\tools\kconfig_new\kconfiglib.py", line 1632, in _parse_block
    self._parse_block(_T_ENDMENU,
  File "C:\Users\fedor\Documents\Projects\esp-idf\tools\kconfig_new\kconfiglib.py", line 1584, in _parse_block
    prev_node = self._parse_block(None,            # end_token
  File "C:\Users\fedor\Documents\Projects\esp-idf\tools\kconfig_new\kconfiglib.py", line 1551, in _parse_block
    while self._has_tokens or self._next_line():
  File "C:\Users\fedor\Documents\Projects\esp-idf\tools\kconfig_new\kconfiglib.py", line 1199, in _next_line
    self._tokenize()
  File "C:\Users\fedor\Documents\Projects\esp-idf\tools\kconfig_new\kconfiglib.py", line 1287, in _tokenize
    self._parse_error("expected keyword as first token")
  File "C:\Users\fedor\Documents\Projects\esp-idf\tools\kconfig_new\kconfiglib.py", line 2165, in _parse_error
    raise KconfigSyntaxError(
kconfiglib.KconfigSyntaxError: C:/Users/fedor/Documents/Projects/lv_port_esp32/components/lvgl_esp32_drivers/Kconfig:1: Couldn't parse 'rsource "lvgl_tft/Kconfig"': expected keyword as first token
CMake Error at C:/Users/fedor/Documents/Projects/esp-idf/tools/cmake/kconfig.cmake:208 (message):
  Failed to run confgen.py

C47D, 2021-02-25:
Thanks for the report, I will try to reproduce the issue and get back to you.
@tore-espressif

tore-espressif, 2021-02-25:
Hello,
what version of ESP-IDF do you have?

60north, 2021-02-25:
Yeah this is the Kconfiglib.py difference, apparently - no rsource in my
version of it.

I am on 4.0.2, which means that it will be breaking for all who install
older versions of IDF for some reasons.

tore-espressif, 2021-02-25:
OK, so obvious quickfix is updating the IDF :)
We will improve the backward compatibility and add some CI steps to prevent this and similar regressions in the future.
Thanks!

tore-espressif, 2021-02-25:
@C47D I'm sorry for the breaking changes. I believe that the refactoring is going to be worth it when we tackle all the remaining details. Thanks!

C47D, 2021-02-25:
@C47D Don't worry about it, let's hope the user can update its toolchain, we should add a note on the README about the minimum ESP-IDF version required.

closed 2021-06-09 by stale[bot]