lvgl/lv_esp_idf #269: No support for ESP32C3

dastarling, 2021-03-25:
We use GitHub issues for development related discussions.
Please use the [forum](https://forum.littlevgl.com/) to ask questions.

**Describe the issue**
Compilation Issue on ESP32C3

**Code to reproduce the issue**
building with chip set to ESP32C3 

**Expected Results**
compiles properly

**Actual Results**
Does not compile

**ESP32 Chip version**
ESP32C3

**ESP-IDF version**
latest

**Development kit used**
ESP32-C3-DevKitM-1

**Development machine OS**
Ubuntu VM

**Compilation warnings/errors (if available)**
2021-03-25T04:24:48.9266177Z In file included from ../components/lvgl_esp32_drivers/lvgl_helpers.c:11:
2021-03-25T04:24:48.9266944Z ../components/lvgl_esp32_drivers/lvgl_helpers.c: In function 'lvgl_spi_driver_init':
2021-03-25T04:24:48.9267821Z ../components/lvgl_esp32_drivers/lvgl_helpers.c:219:51: error: 'spi_names' undeclared (first use in this function); did you mean 'gpio_num_t'?
2021-03-25T04:24:48.9268348Z      ESP_LOGI(TAG, "Configuring SPI host %s (%d)", spi_names[host], host);
2021-03-25T04:24:48.9268745Z                                                    ^~~~~~~~~
2021-03-25T04:24:48.9269382Z /opt/esp/idf/components/log/include/esp_log.h:334:137: note: in definition of macro 'ESP_LOG_LEVEL'
2021-03-25T04:24:48.9269963Z          if (level==ESP_LOG_ERROR )          { esp_log_write(ESP_LOG_ERROR,      tag, LOG_FORMAT(E, format), esp_log_timestamp(), tag, ##__VA_ARGS__); } \
2021-03-25T04:24:48.9270587Z                                                                                                                                          ^~~~~~~~~~~
2021-03-25T04:24:48.9271331Z /opt/esp/idf/components/log/include/esp_log.h:301:38: note: in expansion of macro 'ESP_LOG_LEVEL_LOCAL'
2021-03-25T04:24:48.9271844Z  #define ESP_LOGI( tag, format, ... ) ESP_LOG_LEVEL_LOCAL(ESP_LOG_INFO,    tag, format, ##__VA_ARGS__)
2021-03-25T04:24:48.9272273Z                                       ^~~~~~~~~~~~~~~~~~~
2021-03-25T04:24:48.9272887Z ../components/lvgl_esp32_drivers/lvgl_helpers.c:219:5: note: in expansion of macro 'ESP_LOGI'
2021-03-25T04:24:48.9273335Z      ESP_LOGI(TAG, "Configuring SPI host %s (%d)", spi_names[host], host);
2021-03-25T04:24:48.9273659Z      ^~~~~~~~
2021-03-25T04:24:48.9274095Z ../components/lvgl_esp32_drivers/lvgl_helpers.c:219:51: note: each undeclared identifier is reported only once for each function it appears in
2021-03-25T04:24:48.9274604Z      ESP_LOGI(TAG, "Configuring SPI host %s (%d)", spi_names[host], host);
2021-03-25T04:24:48.9274995Z                                                    ^~~~~~~~~
2021-03-25T04:24:48.9275617Z /opt/esp/idf/components/log/include/esp_log.h:334:137: note: in definition of macro 'ESP_LOG_LEVEL'
2021-03-25T04:24:48.9276195Z          if (level==ESP_LOG_ERROR )          { esp_log_write(ESP_LOG_ERROR,      tag, LOG_FORMAT(E, format), esp_log_timestamp(), tag, ##__VA_ARGS__); } \
2021-03-25T04:24:48.9276820Z                                                                                                                                          ^~~~~~~~~~~
2021-03-25T04:24:48.9277555Z /opt/esp/idf/components/log/include/esp_log.h:301:38: note: in expansion of macro 'ESP_LOG_LEVEL_LOCAL'
2021-03-25T04:24:48.9278061Z  #define ESP_LOGI( tag, format, ... ) ESP_LOG_LEVEL_LOCAL(ESP_LOG_INFO,    tag, format, ##__VA_ARGS__)
2021-03-25T04:24:48.9278484Z                                       ^~~~~~~~~~~~~~~~~~~
2021-03-25T04:24:48.9279096Z ../components/lvgl_esp32_drivers/lvgl_helpers.c:219:5: note: in expansion of macro 'ESP_LOGI'
2021-03-25T04:24:48.9279540Z      ESP_LOGI(TAG, "Configuring SPI host %s (%d)", spi_names[host], host);
2021-03-25T04:24:48.9279868Z      ^~~~~~~~

**If possible, copy the compilation log into a file and attach it here**

I have a fix for this already in a local repo that I can share if needed.

C47D, 2021-06-08:
Hi @dastarling thanks for your patience, is your fix available somewhere in your github profile?

dastarling, 2021-06-08:
Changes I am currently using are located here:
https://github.com/sturnusdesign/lvgl_esp32_drivers/tree/feat/esp32c3_support

Let me know if you have any questions, but the changes are not that large.
It adds a GT911 Touch controller and has to do the DMA to the larger display line by line if sufficient DMA memory is not available since the C3 variant has less RAM than its predecessors.

C47D, 2021-06-08:
Thanks for the quick reply, I don't have any ESP32C3 at the moment, I did the quickest commit in order to begin with C3 support [here](https://github.com/lvgl/lvgl_esp32_drivers/tree/feature/esp32c3_support), but it seems a bit similar to your first commit.

Is it possible for you to send a pull request to the `lvgl_esp32_drivers` repo?

dastarling, 2021-06-09:
I am a bit busy currently, but I could probably help testing or review if you need it.

C47D, 2021-06-11:
Thanks @dastarling , I will let you know when I upload both PR. 😸

closed 2021-06-21 by C47D