lvgl/lv_esp_idf #273: 2.8Zoll TFT Bugs Iili9341 Grafik Glitches

Trackhe, 2021-04-25:
**Describe the issue**

> 
> [image]
> [image]
> [image]
> [image]
> 
> 

**Code to reproduce the issue**

**Expected Results**

**Actual Results**

**ESP32 Chip version**

**ESP-IDF version**

**Development kit used**

**Development machine OS**

**Compilation warnings/errors (if available)**

**If possible, copy the compilation log into a file and attach it here**

tore-espressif, 2021-04-27:
Hello @Trackhe 

I had a brief look at your `sdkconfig` and it seems incorrect:

Please try setting these in menuconfig:
1. LVGL-> resolution: 240x320
2. LVGL-> 16 bit color depth 
3. LVGL-> Swap the 2 bytes of RGB565 color
4. LVGL TFT-> SPI half duplex (this shouldn't be a problem, but just in case)
5. LVGL TFT-> Try every orientation

Trackhe, 2021-04-27:
Hello @tore-espressif 

That was a nice tip. Thank you.

I changed the Resolution and after some testing it works. Very useful. Thank you.

closed 2021-04-27 by Trackhe