lvgl/lv_esp_idf #274: esp-idf can't link liblvgl.a

mfp20, 2021-04-26:
### Introduce the problem
I used the https://github.com/lvgl/lv_port_esp32 repo at first and it was working properly. Then I moved the benchmark demo to my folder, together with lvgl and lvgl_esp32_drivers, adapted to my needs, and it was working good.
So I replaced the benchmark demo with my display.c and, again, it was working properly. Finally I removed lvgl_esp32_drivers and included the ili9341 driver in my display component: it doesn't link.

### Examples and cases

My simple CMakeLists.txt in "display"  esp-idf component: 

```
set(COMPONENT_REQUIRES gpio spi lvgl)
set(COMPONENT_PRIV_REQUIRES log)
set(COMPONENT_ADD_INCLUDEDIRS ".")
set(COMPONENT_SRCS "spi_dev.c lvgl_ili9341.c display.c")
register_component()
```

The link command includes liblvgl.a in the long list of static libs.

```
/home/user/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ -mlongcalls -Wno-frame-address  CMakeFiles/m5stack-fire.elf.dir/project_elf_src_esp32.c.obj -o m5stack-fire.elf esp-idf/esp_pm/libesp_pm.a
[long list of static libs] esp-idf/main/libmain.a esp-idf/lvgl/liblvgl.a esp-idf/display/libdisplay.a  [more static libs] esp-idf/app_trace/libapp_trace.a  -lgcov  -lc && :
```

Liblvgl.a is built and pretty fat:

```
build/esp-idf/lvgl $ ls -alh
...
-rw-r--r--  1 user user 8,1M 24 apr 08.56 liblvgl.a
```

It has lv_init in the syms table:

```
build/esp-idf/lvgl $ objdump -t liblvgl.a | grep lv_init
00000000 l    d  .text.lv_init	00000000 .text.lv_init
00000000 l     O .bss.lv_initialized	00000001 lv_initialized
00000000 l    d  .bss.lv_initialized	00000000 .bss.lv_initialized
00000000 l    d  .literal.lv_init	00000000 .literal.lv_init
00000000 g     F .text.lv_init	00000096 lv_init
```

but the result is an undefined reference to any lv_* method:

```
/home/user/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/display/libdisplay.a(display.c.obj):(.literal.lvGuiTask+0x4): undefined reference to `lv_init'
/home/user/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/display/libdisplay.a(display.c.obj): in function `lvGuiTask':
/home/user/projects/repo-esp32/m5stack-fire/build/../components/display/display.c:33: undefined reference to `lv_init'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

Previous similar issues:
- https://github.com/lvgl/lvgl/issues/1254

### Suggested solution
No idea how to fix this.

kisvegabor, 2021-04-26:
Hi,

I moved this issue to https://github.com/lvgl/lv_port_esp32 

Unfortunately, I'm that experienced with ESP. @C47D (our ESP32 expert :slightly_smiling_face: ) is quite busy now but hopefully, he (or someone how knows ESP in more detail) can a look at it.

tore-espressif, 2021-04-27:
Hi @mfp20 

could you please build with `idf.py build -v` (verbose) and share the output? (link command should be sufficient).

And also output of `tree` command in your project root, just to be sure, thanks!

C47D, 2021-04-27:
Hi,

I don't consider myself an ESP32 expert, sorry @kisvegabor 😄 , but I will try take a look, got a couple of free days finally! coming at the end of this week.

sidwarkd, 2021-07-23:
@C47D I'm having the same issue. I can get things to build just fine using the sample and keeping everything within the main folder. As soon as I try to move the LVGL logic into my own component (including the lvgl component as required) I start getting linker errors saying it can't find lv_ functions. The exact same code compiles and links fine in my main.cpp. Copy and paste into a component folder and the build fails. Any suggestions or direction would be most appreicated.

sidwarkd, 2021-07-24:
I found the issue with my problem. The lvgl CMakeLists `REQUIRES` the `main` component so it creates a circular dependency of sorts. All I had to do was remove the `REQUIRES main` from lvgl's CMakeLists and everything builds just find now.

gfurtadoalmeida, 2021-07-29:
Thank you @sidwarkd!
Before removing `main` from `REQUIRES`, adding `lvgl_esp32_drivers` as a requirement gave tons of `undefined` errors. Now it compiles!

sidwarkd, 2021-07-30:
I've submitted https://github.com/lvgl/lvgl/pull/2420 which should resolve this issue.

tore-espressif, 2021-08-02:
closed via lvgl/lvgl#2420

closed 2021-08-02 by tore-espressif