lvgl/lv_esp_idf #278: ESP32-S2. When you touch the touchscreen (XPT2046) anywhere, the returned coordinates are always zero.

SinglWolf, 2021-05-22:
**Describe the issue**
When you touch the touchscreen anywhere, the returned coordinates are always zero.
The ILI9488 display controller being used initializes correctly. All used widgets are displayed correctly.
The XPT2046 touchscreen controller used does not display any errors during initialization.
**Code to reproduce the issue**
The display and touchscreen initialization function in the `lvgl_esp32_drivers/lvgl_helpers.c` file has been fixed for one SPI host due to the fact that the second SPI host is busy to control the `VS1053B` chip:
```c
/* Interface and driver initialization */
void lvgl_driver_init(void)
{
    gpio_num_t miso;
    gpio_num_t mosi;
    gpio_num_t sclk;
    gpio_get_spi_bus_lcd(&miso, &mosi, &sclk);
    ESP_LOGI(TAG, "Display hor size: %d, ver size: %d", LV_HOR_RES_MAX, LV_VER_RES_MAX);
        lvgl_spi_driver_init(LCD_HOST,
                         miso, mosi, sclk,
                         SPI_BUS_MAX_TRANSFER_SZ, 1,
                         -1, -1);

    disp_spi_add_device(LCD_HOST);
    tp_spi_add_device(LCD_HOST);

    disp_driver_init();
    touch_driver_init();
}
```
**Expected Results**
The XPT2046 touchscreen controller gives the correct touch coordinates.
**Actual Results**
Display and touchscreen initialization log:
```
I (4308) lvgl_helpers: Display hor size: 480, ver size: 320
I (4308) lvgl_helpers: Initializing SPI master for display
I (4308) lvgl_helpers: Configuring SPI host  (1)
I (4308) lvgl_helpers: MISO pin: 3, MOSI pin: 4, SCLK pin: 6, IO2/WP pin: -1, IO3/HD pin: -1
I (4318) lvgl_helpers: Max transfer size: 14400 (bytes)
I (4328) lvgl_helpers: Initializing SPI bus...
I (4328) disp_spi: Adding SPI device
I (4338) disp_spi: Clock speed: 40000000Hz, mode: 0, CS pin: 10
I (4538) ILI9488: ILI9488 initialization.
I (4838) ILI9488: Display orientation: LANDSCAPE_INVERTED
I (4838) ILI9488: 0x36 command value: 0xE8
I (4838) tp_spi: Initializing SPI master for touch
I (4838) XPT2046: XPT2046 Initialization
I (4838) gpio: GPIO[5]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
```
Log displaying coordinates when showing a touchscreen:
```
I (151157) XPT2046: P(0,0)
I (151157) XPT2046: P_norm(0,0)
I (151157) XPT2046: x = 0, y = 0
I (151207) XPT2046: P(0,0)
I (151207) XPT2046: P_norm(0,0)
I (151207) XPT2046: x = 0, y = 0
I (156687) XPT2046: P(0,0)
I (156687) XPT2046: P_norm(0,0)
I (156687) XPT2046: x = 0, y = 0
I (156737) XPT2046: P(0,0)
I (156737) XPT2046: P_norm(0,0)
I (156737) XPT2046: x = 0, y = 0
```
Returns coordinates are always zero.
**ESP32 Chip version**
ESP32-S2
**ESP-IDF version**
4.2.1
**Development kit used**
VSCode+PlatformIO
**Development machine OS**
Windows 7 64 bit.

Hope I described the problem correctly.
Please help sort out the problem. Thank you.

SinglWolf, 2021-06-20:
I am still waiting for the developers' response.

manuferro, 2021-07-06:
SinglWolf, I suggest to check the electrical portion of your project at first.
I had a similar problem & I start with https://github.com/Bodmer/TFT_Touch
with TFT_Touch_Raw.ino example printing the raw result (modifying a litle the same to have the raw value for the registry 0x90 and 0xD0) untill I fix the hardware problem.
After that lvgl work perfectly.

SinglWolf, 2021-07-06:
> SinglWolf, I suggest to check the electrical portion of your project at first.
> https://github.com/Bodmer/TFT_Touch

Thanks for the answer. The touchscreen works on the arduino-esp32 platform. And doesn't work on esp-idf platform. Such is the state of affairs.

manuferro, 2021-07-07:
ok,
  are you using interupt signal? If yes to you check if there is any signal (low status) when you press on the touch?

SinglWolf, 2021-07-07:
> ok,
> are you using interupt signal? If yes do you check if there is any signal (low status) when you press on the touch?

Naturally, I use the interrupt signal. When I touch anywhere on the screen, the console displays (when a low level appears)
```
I (151157) XPT2046: P(0,0)
I (151157) XPT2046: P_norm(0,0)
I (151157) XPT2046: x = 0, y = 0
```

manuferro, 2021-07-13:
how is configured the XPT2046? (sdkconfig) which SPI port?  speed, pin? what kind of test did you do?
It's a little hard to give an answer without understand your configuration, hardware and so on...

SinglWolf, 2021-07-15:
> how is configured the XPT2046? (sdkconfig) which SPI port? speed, pin? what kind of test did you do?
> It's a little hard to give an answer without understand your configuration, hardware and so on...

This is a pointless conversation.
Do you have a working example of touch screen operation on XPT2046 controller in ESP-IDF environment for ESP32-S2?

manuferro, 2021-07-15:
https://github.com/lvgl/lv_port_esp32
working for me out of the box with several display (I've tested 8 different displays)

Thanks

SinglWolf, 2021-07-15:
> working for me out of the box with several display

OK. What version of ESP-IDF do You have?

manuferro, 2021-07-15:
4.2.1

SinglWolf, 2021-07-15:
> 4.2.1

It's a good news. I had problems on an older version. I hope that everything will work and for me.
Now there is no free time to check. :-(

closed 2022-04-28 by stale[bot]

paulftw, 2022-08-22:
Came here while googling a solution for the same problem. 
My XPT2046 returns constant wrong coordinates at 40MHz, but works better when SPI is at **8-10MHz**.

royzahor, 2023-01-11:
> Came here while googling a solution for the same problem. My XPT2046 returns constant wrong coordinates at 40MHz, but works better when SPI is at **8-10MHz**.

You saved my day. I had similar problem with SPI settings

FangBoCHN, 2023-09-16:
> **描述问题** 当我触摸触摸屏时，返回的坐标始终为零。正在使用的显示控制器正确初始化。所有使用的小部件都正确显示。使用的触摸屏控制器在初始化期间不显示任何错误。 **重现问题的**代码 由于第二台主机忙于控制芯片，文件中的显示和触摸屏初始化功能已针对一台主机修复：`ILI9488``XPT2046``lvgl_esp32_drivers/lvgl_helpers.c``SPI``SPI``VS1053B`
> 
> ```c
> /* Interface and driver initialization */
> void lvgl_driver_init(void)
> {
>     gpio_num_t miso;
>     gpio_num_t mosi;
>     gpio_num_t sclk;
>     gpio_get_spi_bus_lcd(&miso, &mosi, &sclk);
>     ESP_LOGI(TAG, "Display hor size: %d, ver size: %d", LV_HOR_RES_MAX, LV_VER_RES_MAX);
>         lvgl_spi_driver_init(LCD_HOST,
>                          miso, mosi, sclk,
>                          SPI_BUS_MAX_TRANSFER_SZ, 1,
>                          -1, -1);
> 
>     disp_spi_add_device(LCD_HOST);
>     tp_spi_add_device(LCD_HOST);
> 
>     disp_driver_init();
>     touch_driver_init();
> }
> ```
> 
> **预期结果** 触摸屏控制器提供正确的触摸坐标。 **实际结果** 显示和触摸屏初始化日志：`XPT2046`
> 
> ```
> I (4308) lvgl_helpers: Display hor size: 480, ver size: 320
> I (4308) lvgl_helpers: Initializing SPI master for display
> I (4308) lvgl_helpers: Configuring SPI host  (1)
> I (4308) lvgl_helpers: MISO pin: 3, MOSI pin: 4, SCLK pin: 6, IO2/WP pin: -1, IO3/HD pin: -1
> I (4318) lvgl_helpers: Max transfer size: 14400 (bytes)
> I (4328) lvgl_helpers: Initializing SPI bus...
> I (4328) disp_spi: Adding SPI device
> I (4338) disp_spi: Clock speed: 40000000Hz, mode: 0, CS pin: 10
> I (4538) ILI9488: ILI9488 initialization.
> I (4838) ILI9488: Display orientation: LANDSCAPE_INVERTED
> I (4838) ILI9488: 0x36 command value: 0xE8
> I (4838) tp_spi: Initializing SPI master for touch
> I (4838) XPT2046: XPT2046 Initialization
> I (4838) gpio: GPIO[5]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
> ```
> 
> 触摸触摸屏时显示坐标的日志：
> 
> ```
> I (151157) XPT2046: P(0,0)
> I (151157) XPT2046: P_norm(0,0)
> I (151157) XPT2046: x = 0, y = 0
> I (151207) XPT2046: P(0,0)
> I (151207) XPT2046: P_norm(0,0)
> I (151207) XPT2046: x = 0, y = 0
> I (156687) XPT2046: P(0,0)
> I (156687) XPT2046: P_norm(0,0)
> I (156687) XPT2046: x = 0, y = 0
> I (156737) XPT2046: P(0,0)
> I (156737) XPT2046: P_norm(0,0)
> I (156737) XPT2046: x = 0, y = 0
> ```
> 
> 返回坐标始终为零。 **ESP32 芯片**版本 **ESP-IDF 版本** 开发**套件采用**++**开发机操作系统** Windows 7 64 位。`ESP32-S2``4.2.1``[ESP-12-K] MCU-32-S2``VSCode``PlatformIO`
> 
> 希望我正确描述了问题。请帮助解决问题。谢谢。

我遇到了和你一样的问题，困扰我很久了，请问你是怎么解决的

FangBoCHN, 2023-09-16:
> 我遇到了和你一样的问题，困扰我很久了，请问你是怎么解决的

我使用的是ESPIDF V4.4.4,显示一切正常，显示与触摸共用一组spi,
同样的我是移植的lvgl官方为ESPIDF创建的工程：https://github.com/lvgl/lv_port_esp32

SinglWolf, 2023-09-16:
> 我遇到了和你一样的问题，困扰我很久了，请问你是怎么解决的

If the display and touch screen are connected to the same `SPI` bus, you need to physically disconnect the `MISO` wire from the display and check the operation of the touch screen.

FangBoCHN, 2023-09-19:
> If the display and touch screen are connected to the same `SPI` bus, you need to physically disconnect the `MISO` wire from the display and check the operation of the touch screen.

首先，我非常感谢您的回复，可否详细说一下您解决的思路和过程呢

SinglWolf, 2023-09-19:
> 首先，我非常感谢您的回复，可否详细说一下您解决的思路和过程呢

I had the same problem. I just disconnected the MISO wire from the display and the touchscreen started working.

FangBoCHN, 2023-09-19:
> > 首先，我非常感谢您的回复，可否详细说一下您解决的思路和过程呢
> 
> I had the same problem. I just disconnected the `MISO` wire from the display and the touchscreen started working.

好的，感谢您的回复，顺便提一下你所在的地方是白天吗？为什么您回复这么即时？

FangBoCHN, 2023-09-19:
> > 首先，我非常感谢您的回复，可否详细说一下您解决的思路和过程呢
> 
> I had the same problem. I just disconnected the `MISO` wire from the display and the touchscreen started working.

所以，您现在还是只能将MISO线断开来使XPT2046工作吗？做不到让显示屏的MISO也与MCU连接吗

SinglWolf, 2023-09-19:
> 顺便提一下你所在的地方是白天吗？为什么您回复这么即时？

It's 20:19 on the clock. I'm just working on a computer right now. I've given you some time.

> 所以，您现在还是只能将MISO线断开来使XPT2046工作吗？做不到让显示屏的MISO也与MCU连接吗

The `XPT2046` needs all the wires. And the MISO wire is not needed for the display, because it is a reading line that is only needed for `XPT2046`. If the `MISO` line is connected to the display, then there is a conflict between the display and the `XPT2046`. I hope I explained it clearly.

FangBoCHN, 2023-09-19:
> It's 20:19 on the clock. I'm just working on a computer right now. I've given you some time.
>
>
> The `XPT2046` needs all the wires. And the MISO wire is not needed for the display, because it is a [...]

是的我懂了，就是说显示屏只需要一个MOSI来接收MCU给它的显示数据，并不需要使用MISO来向MCU发送数据

SinglWolf, 2023-09-19:
> 是的我懂了，就是说显示屏只需要一个MOSI来接收MCU给它的显示数据，并不需要使用MISO来向MCU发送数据

Yes. Everything is right. You can check it out. The display will not fail due to the absence of a `MISO` wire :-D.

FangBoCHN, 2023-09-19:
> > 是的我懂了，就是说显示屏只需要一个MOSI来接收MCU给它的显示数据，并不需要使用MISO来向MCU发送数据
> 
> Yes. Everything is right. You can check it out. The display will not fail due to the absence of a `MISO` wire :-D.
好的，真的十分感谢您的耐心回复，我有空会按您的方法尝试的，最后祝您一切顺利

SinglWolf, 2023-09-19:
> 好的，真的十分感谢您的耐心回复，我有空会按您的方法尝试的，最后祝您一切顺利

I know that You will succeed! Good luck!