lvgl/lv_esp_idf #279: I get an '#error "Unsupported orientation"' when I build a project for the SSD1306 display.

PopherTheG, 2021-05-29:
We use GitHub issues for development related discussions.
Please use the [forum](https://forum.littlevgl.com/) to ask questions.

**Describe the issue**
I can't build a project if I use the SSD1306 display. It throws and error which says '#error "Unsupported Orientation"'

**Code to reproduce the issue**
I just cloned the project 'git clone --recurse-submodules https://github.com/lvgl/lv_port_esp32.git' and in the menuconfig I choose the TFT display name SSD1306, set its orientation to landscape.

**Expected Results**
I expected the code to be built successfully with no errors since it was just from the port. I just used an SSD1306 display.

**Actual Results**
In the terminal, when I build the project using the SSD1306 I get the '#error "Unsupported Orientation"'.  It also says the the error is rooted form the ssd1306_init function.

**ESP32 Chip version**
Chip is ESP32-D0WDQ6 (revision 1)

**ESP-IDF version**
ESP-IDF v4.2

**Development kit used**
ESP32-DevKitC

**Development machine OS**
Windows 10 Pro

**Compilation warnings/errors (if available)**
`FAILED: esp-idf/lvgl_esp32_drivers/CMakeFiles/__idf_lvgl_esp32_drivers.dir/lvgl_tft/ssd1306.c.obj
ccache C:\esp\tools\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe -DHAVE_CONFIG_H -DLV_CONF_INCLUDE_SIMPLE -DLV_LVGL_H_INCLUDE_SIMPLE -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DUNITY_INCLUDE_CONFIG_H -DWITH_POSIX -Iconfig -I../components/lvgl_esp32_drivers -I../components/lvgl_esp32_drivers/lvgl_tft -IC:/esp/esp-idf/components/newlib/platform_include -IC:/esp/esp-idf/components/freertos/include -IC:/esp/esp-idf/components/freertos/xtensa/include -IC:/esp/esp-idf/components/heap/include -IC:/esp/esp-idf/components/log/include -IC:/esp/esp-idf/components/lwip/include/apps -IC:/esp/esp-idf/components/lwip/include/apps/sntp -IC:/esp/esp-idf/components/lwip/lwip/src/include -IC:/esp/esp-idf/components/lwip/port/esp32/include -IC:/esp/esp-idf/components/lwip/port/esp32/include/arch -IC:/esp/esp-idf/components/soc/src/esp32/. -IC:/esp/esp-idf/components/soc/src/esp32/include -IC:/esp/esp-idf/components/soc/include -IC:/esp/esp-idf/components/esp_rom/include -IC:/esp/esp-idf/components/esp_common/include -IC:/esp/esp-idf/components/esp_system/include -IC:/esp/esp-idf/components/xtensa/include -IC:/esp/esp-idf/components/xtensa/esp32/include -IC:/esp/esp-idf/components/esp32/include -IC:/esp/esp-idf/components/driver/include -IC:/esp/esp-idf/components/driver/esp32/include -IC:/esp/esp-idf/components/esp_ringbuf/include -IC:/esp/esp-idf/components/efuse/include -IC:/esp/esp-idf/components/efuse/esp32/include -IC:/esp/esp-idf/components/espcoredump/include -IC:/esp/esp-idf/components/esp_timer/include -IC:/esp/esp-idf/components/esp_ipc/include -IC:/esp/esp-idf/components/soc/soc/esp32/include -IC:/esp/esp-idf/components/soc/soc/esp32/../include -IC:/esp/esp-idf/components/soc/soc/esp32/private_include -IC:/esp/esp-idf/components/vfs/include -IC:/esp/esp-idf/components/esp_wifi/include -IC:/esp/esp-idf/components/esp_wifi/esp32/include -IC:/esp/esp-idf/components/esp_event/include -IC:/esp/esp-idf/components/esp_netif/include -IC:/esp/esp-idf/components/esp_eth/include -IC:/esp/esp-idf/components/tcpip_adapter/include -IC:/esp/esp-idf/components/app_trace/include -I../components/lvgl -I../components/lvgl/src -IC:/esp/esp-idf/components/mbedtls/port/include -IC:/esp/esp-idf/components/mbedtls/mbedtls/include -IC:/esp/esp-idf/components/mbedtls/esp_crt_bundle/include -IC:/esp/esp-idf/components/bootloader_support/include -IC:/esp/esp-idf/components/app_update/include -IC:/esp/esp-idf/components/spi_flash/include -IC:/esp/esp-idf/components/wpa_supplicant/include -IC:/esp/esp-idf/components/wpa_supplicant/port/include -IC:/esp/esp-idf/components/wpa_supplicant/include/esp_supplicant -IC:/esp/esp-idf/components/nvs_flash/include -IC:/esp/esp-idf/components/pthread/include -IC:/esp/esp-idf/components/perfmon/include -IC:/esp/esp-idf/components/asio/asio/asio/include -IC:/esp/esp-idf/components/asio/port/include -IC:/esp/esp-idf/components/cbor/port/include -IC:/esp/esp-idf/components/coap/port/include -IC:/esp/esp-idf/components/coap/port/include/coap -IC:/esp/esp-idf/components/coap/libcoap/include -IC:/esp/esp-idf/components/coap/libcoap/include/coap2 -IC:/esp/esp-idf/components/console -IC:/esp/esp-idf/components/nghttp/port/include -IC:/esp/esp-idf/components/nghttp/nghttp2/lib/includes -IC:/esp/esp-idf/components/esp-tls -IC:/esp/esp-idf/components/esp_adc_cal/include -IC:/esp/esp-idf/components/esp_gdbstub/include -IC:/esp/esp-idf/components/esp_hid/include -IC:/esp/esp-idf/components/tcp_transport/include -IC:/esp/esp-idf/components/esp_http_client/include -IC:/esp/esp-idf/components/esp_http_server/include -IC:/esp/esp-idf/components/esp_https_ota/include -IC:/esp/esp-idf/components/protobuf-c/protobuf-c -IC:/esp/esp-idf/components/protocomm/include/common -IC:/esp/esp-idf/components/protocomm/include/security -IC:/esp/esp-idf/components/protocomm/include/transports -IC:/esp/esp-idf/components/mdns/include -IC:/esp/esp-idf/components/esp_local_ctrl/include -IC:/esp/esp-idf/components/sdmmc/include -IC:/esp/esp-idf/components/esp_serial_slave_link/include -IC:/esp/esp-idf/components/esp_websocket_client/include -IC:/esp/esp-idf/components/expat/expat/expat/lib -IC:/esp/esp-idf/components/expat/port/include -IC:/esp/esp-idf/components/wear_levelling/include -IC:/esp/esp-idf/components/fatfs/diskio -IC:/esp/esp-idf/components/fatfs/vfs -IC:/esp/esp-idf/components/fatfs/src -IC:/esp/esp-idf/components/freemodbus/common/include -IC:/esp/esp-idf/components/idf_test/include -IC:/esp/esp-idf/components/idf_test/include/esp32 -IC:/esp/esp-idf/components/jsmn/include -IC:/esp/esp-idf/components/json/cJSON -IC:/esp/esp-idf/components/libsodium/libsodium/src/libsodium/include -IC:/esp/esp-idf/components/libsodium/port_include -IC:/esp/esp-idf/components/mqtt/esp-mqtt/include -IC:/esp/esp-idf/components/openssl/include -IC:/esp/esp-idf/components/spiffs/include -IC:/esp/esp-idf/components/ulp/include -IC:/esp/esp-idf/components/unity/include -IC:/esp/esp-idf/components/unity/unity/src -IC:/esp/esp-idf/components/wifi_provisioning/include -I../components/lv_examples -mlongcalls -Wno-frame-address   -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -Og -std=gnu99 -Wno-old-style-declaration -D_GNU_SOURCE -DIDF_VER=\"v4.2\" -DESP_PLATFORM -MD -MT esp-idf/lvgl_esp32_drivers/CMakeFiles/__idf_lvgl_esp32_drivers.dir/lvgl_tft/ssd1306.c.obj -MF esp-idf\lvgl_esp32_drivers\CMakeFiles\__idf_lvgl_esp32_drivers.dir\lvgl_tft\ssd1306.c.obj.d -o esp-idf/lvgl_esp32_drivers/CMakeFiles/__idf_lvgl_esp32_drivers.dir/lvgl_tft/ssd1306.c.obj   -c ../components/lvgl_esp32_drivers/lvgl_tft/ssd1306.c
../components/lvgl_esp32_drivers/lvgl_tft/ssd1306.c: In function 'ssd1306_init':
../components/lvgl_esp32_drivers/lvgl_tft/ssd1306.c:111:6: error: #error "Unsupported orientation"
     #error "Unsupported orientation"
      ^~~~~
[1023/1031] Building C object esp-idf/lvgl_esp32_drivers/CMakeFiles/__idf_lvgl_esp32_drivers.dir/lvgl_tft/disp_driver.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1`

**If possible, copy the compilation log into a file and attach it here**

tore-espressif, 2021-05-31:
Hello @PopherTheG ,
could you please try reverting this commit and try the build again?
https://github.com/lvgl/lvgl_esp32_drivers/commit/9fed1cc47b5a45fec6bae08b55d2147d3b50260c

C47D, 2021-06-16:
Seems like it was solved a few commits after the current `lvgl_esp32_drivers` commit, in here https://github.com/lvgl/lvgl_esp32_drivers/commit/3e1c6f3b67558aae45958cd500e1e1685c189282

@PopherTheG Can you update the `lvgl_esp32_drivers` component to it's current master?

closed 2021-07-21 by stale[bot]