lvgl/lv_esp_idf #280: 1.8inch ST7735 driver display Cannot display normally

juxiangwu, 2021-05-29:
**Development Board**:NodeMCU ESP32-S V3.0
**Display Module**:1.8inch(128x160),display controller driver:ST7735,

here is the LV config:

```
#define CONFIG_LV_USE_DEMO_WIDGETS 1
#define CONFIG_LV_DEMO_WIDGETS_SLIDESHOW 1
#define CONFIG_LV_ATTRIBUTE_FAST_MEM_USE_IRAM 1
#define CONFIG_LV_CONF_SKIP 1
#define CONFIG_LV_HOR_RES_MAX 160
#define CONFIG_LV_VER_RES_MAX 128
#define CONFIG_LV_COLOR_DEPTH_16 1
#define CONFIG_LV_COLOR_DEPTH 16
#define CONFIG_LV_COLOR_16_SWAP 1
#define CONFIG_LV_ANTIALIAS 1
#define CONFIG_LV_DISP_DEF_REFR_PERIOD 30
#define CONFIG_LV_DPI 130
#define CONFIG_LV_DISP_SMALL_LIMIT 30
#define CONFIG_LV_DISP_MEDIUM_LIMIT 50
#define CONFIG_LV_DISP_LARGE_LIMIT 70
#define CONFIG_LV_MEM_SIZE_KILOBYTES 32
#define CONFIG_LV_INDEV_DEF_READ_PERIOD 30
#define CONFIG_LV_INDEV_DEF_DRAG_LIMIT 10
#define CONFIG_LV_INDEV_DEF_DRAG_THROW 10
#define CONFIG_LV_INDEV_DEF_LONG_PRESS_TIME 400
#define CONFIG_LV_INDEV_DEF_LONG_PRESS_REP_TIME 100
#define CONFIG_LV_INDEV_DEF_GESTURE_LIMIT 50
#define CONFIG_LV_INDEV_DEF_GESTURE_MIN_VELOCITY 3
#define CONFIG_LV_USE_ANIMATION 1
#define CONFIG_LV_USE_SHADOW 1
#define CONFIG_LV_SHADOW_CACHE_SIZE 0
#define CONFIG_LV_USE_OUTLINE 1
#define CONFIG_LV_USE_PATTERN 1
#define CONFIG_LV_USE_VALUE_STR 1
#define CONFIG_LV_USE_BLEND_MODES 1
#define CONFIG_LV_USE_OPA_SCALE 1
#define CONFIG_LV_USE_IMG_TRANSFORM 1
#define CONFIG_LV_USE_GROUP 1
#define CONFIG_LV_USE_GPU 1
#define CONFIG_LV_USE_FILESYSTEM 1
#define CONFIG_LV_USE_API_EXTENSION_V6 1
#define CONFIG_LV_USE_API_EXTENSION_V7 1
#define CONFIG_LV_IMG_CF_INDEXED 1
#define CONFIG_LV_IMG_CF_ALPHA 1
#define CONFIG_LV_IMG_CACHE_DEF_SIZE 1
#define CONFIG_LV_USE_LOG 1
#define CONFIG_LV_LOG_LEVEL_TRACE 1
#define CONFIG_LV_LOG_LEVEL 0
#define CONFIG_LV_LOG_PRINTF 1
#define CONFIG_LV_USE_ASSERT_NULL 1
#define CONFIG_LV_USE_ASSERT_MEM 1
#define CONFIG_LV_FONT_FMT_TXT_LARGE 1
#define CONFIG_LV_USE_FONT_SUBPX 1
#define CONFIG_LV_FONT_MONTSERRAT_14 1
#define CONFIG_LV_FONT_MONTSERRAT_16 1
#define CONFIG_LV_FONT_DEFAULT_SMALL_MONTSERRAT_16 1
#define CONFIG_LV_FONT_DEFAULT_NORMAL_MONTSERRAT_16 1
#define CONFIG_LV_FONT_DEFAULT_SUBTITLE_MONTSERRAT_16 1
#define CONFIG_LV_FONT_DEFAULT_TITLE_MONTSERRAT_16 1
#define CONFIG_LV_THEME_MATERIAL 1
#define CONFIG_LV_THEME_DEFAULT_INIT_MATERIAL 1
#define CONFIG_LV_THEME_DEFAULT_COLOR_PRIMARY 0xFF0000
#define CONFIG_LV_THEME_DEFAULT_COLOR_SECONDARY 0x0000FF
#define CONFIG_LV_THEME_DEFAULT_FLAG_LIGHT 1
#define CONFIG_LV_TXT_ENC_UTF8 1
#define CONFIG_LV_TXT_BREAK_CHARS " ,.;:-_"
#define CONFIG_LV_TXT_LINE_BREAK_LONG_LEN 0
#define CONFIG_LV_TXT_COLOR_CMD "#"
#define CONFIG_LV_USE_OBJ_REALIGN 1
#define CONFIG_LV_USE_EXT_CLICK_AREA_TINY 1
#define CONFIG_LV_USE_ARC 1
#define CONFIG_LV_USE_BAR 1
#define CONFIG_LV_USE_BTN 1
#define CONFIG_LV_USE_BTNMATRIX 1
#define CONFIG_LV_USE_CALENDAR 1
#define CONFIG_LV_USE_CANVAS 1
#define CONFIG_LV_USE_CHECKBOX 1
#define CONFIG_LV_USE_CHART 1
#define CONFIG_LV_CHART_AXIS_TICK_MAX_LEN 256
#define CONFIG_LV_USE_CONT 1
#define CONFIG_LV_USE_CPICKER 1
#define CONFIG_LV_USE_DROPDOWN 1
#define CONFIG_LV_DROPDOWN_DEF_ANIM_TIME 200
#define CONFIG_LV_USE_GAUGE 1
#define CONFIG_LV_USE_IMG 1
#define CONFIG_LV_USE_IMGBTN 1
#define CONFIG_LV_USE_KEYBOARD 1
#define CONFIG_LV_USE_LABEL 1
#define CONFIG_LV_LABEL_DEF_SCROLL_SPEED 25
#define CONFIG_LV_LABEL_WAIT_CHAR_COUNT 3
#define CONFIG_LV_USE_LED 1
#define CONFIG_LV_LED_BRIGHT_MIN 120
#define CONFIG_LV_LED_BRIGHT_MAX 255
#define CONFIG_LV_USE_LINE 1
#define CONFIG_LV_USE_LIST 1
#define CONFIG_LV_LIST_DEF_ANIM_TIME 100
#define CONFIG_LV_USE_LINEMETER 1
#define CONFIG_LV_LINEMETER_PRECISE_NO_EXTRA_PRECISION 1
#define CONFIG_LV_USE_OBJMASK 1
#define CONFIG_LV_USE_MSGBOX 1
#define CONFIG_LV_USE_PAGE 1
#define CONFIG_LV_PAGE_DEF_ANIM_TIME 100
#define CONFIG_LV_USE_SPINNER 1
#define CONFIG_LV_SPINNER_DEF_ARC_LENGTH 60
#define CONFIG_LV_SPINNER_DEF_SPIN_TIME 1000
#define CONFIG_LV_SPINNER_TYPE_SPINNING_ARC 1
#define CONFIG_LV_USE_ROLLER 1
#define CONFIG_LV_ROLLER_DEF_ANIM_TIME 200
#define CONFIG_LV_ROLLER_INF_PAGES 7
#define CONFIG_LV_USE_SLIDER 1
#define CONFIG_LV_USE_SPINBOX 1
#define CONFIG_LV_USE_SWITCH 1
#define CONFIG_LV_USE_TEXTAREA 1
#define CONFIG_LV_TEXTAREA_DEF_CURSOR_BLINK_TIME 400
#define CONFIG_LV_TEXTAREA_DEF_PWN_SHOW_TIME 1500
#define CONFIG_LV_USE_TABLE 1
#define CONFIG_LV_TABLE_COL_MAX 12
#define CONFIG_LV_TABLE_CELL_STYLE_CNT 4
#define CONFIG_LV_USE_TABVIEW 1
#define CONFIG_LV_TABVIEW_DEF_ANIM_TIME 300
#define CONFIG_LV_USE_TILEVIEW 1
#define CONFIG_LV_TILEVIEW_DEF_ANIM_TIME 300
#define CONFIG_LV_PREDEFINED_DISPLAY_NONE 1
#define CONFIG_LV_TFT_DISPLAY_CONTROLLER_ST7735S 1
#define CONFIG_LV_TFT_DISPLAY_PROTOCOL_SPI 1
#define CONFIG_DISPLAY_ORIENTATION_LANDSCAPE 1
#define CONFIG_LV_DISPLAY_ORIENTATION 2
#define CONFIG_LV_PREDEFINED_PINS_38V1 1
#define CONFIG_LV_TFT_DISPLAY_USER_CONTROLLER_ST7735S 1
#define CONFIG_LV_TFT_DISPLAY_SPI_HSPI 1
#define CONFIG_LV_TFT_DISPLAY_SPI_TRANS_MODE_SIO 1
#define CONFIG_LV_TFT_DISPLAY_SPI_HALF_DUPLEX 1
#define CONFIG_LV_TFT_CUSTOM_SPI_CLK_DIVIDER 2
#define CONFIG_LV_DISP_SPI_MOSI 23
#define CONFIG_LV_DISP_SPI_CLK 18
#define CONFIG_LV_DISPLAY_USE_SPI_CS 1
#define CONFIG_LV_DISP_SPI_CS 5
#define CONFIG_LV_DISPLAY_USE_DC 1
#define CONFIG_LV_DISP_PIN_DC 26
#define CONFIG_LV_DISP_PIN_RST 0
#define CONFIG_LV_DISP_PIN_BUSY 35
#define CONFIG_LV_DISP_PIN_SDA 5
#define CONFIG_LV_DISP_PIN_SCL 4
#define CONFIG_LV_TOUCH_CONTROLLER 0
#define CONFIG_LV_TOUCH_CONTROLLER_NONE 1
```

after building and flashing,I got the confuse result:
https://raw.githubusercontent.com/juxiangwu/temp_images/master/01.png
[image]

juxiangwu, 2021-05-29:
while running the lv_demo_widgets,got the result below:
https://github.com/juxiangwu/temp_images/raw/master/05-1.gif
[image]

kisvegabor, 2021-05-31:
From the gifs it seems the window in  Seemingly the window area's size is off by one in the [flush function](https://github.com/lvgl/lvgl_esp32_drivers/blob/9fed1cc47b5a45fec6bae08b55d2147d3b50260c/lvgl_tft/st7735s.c#L142).

juxiangwu, 2021-06-01:
how to solve the problem?

juxiangwu, 2021-06-01:
I have solve the problem:
just make some change for the file [lvgl_tft/st7735s.h](https://github.com/lvgl/lvgl_esp32_drivers/blob/9fed1cc47b5a45fec6bae08b55d2147d3b50260c/lvgl_tft/st7735s.h)

```
//#define ST7735_GREENTAB160x80 // For 160 x 80 display (BGR, inverted, 26 / 1 offset)
//#define COLSTART            26
//#define ROWSTART            1
#define COLSTART            0
#define ROWSTART            0
```
Thanks for help!

Rajssss, 2021-06-19:
I have got a simmilar issue with a 1.44" display with same controller, and this fiex the 

> I have solve the problem:
> just make some change for the file [lvgl_tft/st7735s.h](https://github.com/lvgl/lvgl_esp32_drivers/blob/9fed1cc47b5a45fec6bae08b55d2147d3b50260c/lvgl_tft/st7735s.h)
>
> ``` [...]

I have faced a similar issue with a 1.44" display with the same controller and this fixed the issue. Spent 2 days looking for a fix, thanks :)

closed 2021-07-21 by stale[bot]