lvgl/lv_esp_idf #282: ESP32S2 can't change display DC pin to gpio45 in vscode by using SDK configureation editor

Pupil2013, 2021-05-31:
I‘m using the esp32s2,lcd driver st7789 ,vscode 
when change the lcd driver dc pin to 45,it will automatically change back to 2.

**Code to reproduce the issue**
Using the SDK Configuration editor
**Actual Results**
when change the lcd driver dc pin to 45,it will automatically change back to 2.
**ESP32 Chip version**
ESP32S2
**ESP-IDF version**
v4.2
**Development kit used**
VSCODE  ESP-IDF tool
**Development machine OS**
windows 10

tore-espressif, 2021-06-04:
ESP32S2 has 43 GPIOs, see [datasheet](https://www.espressif.com/sites/default/files/documentation/esp32-s2_datasheet_en.pdf)

Pupil2013, 2021-06-10:
> ESP32S2 has 43 GPIOs, see [datasheet](https://www.espressif.com/sites/default/files/documentation/esp32-s2_datasheet_en.pdf)

Yes it has 43 gpios,but here the io for DC line should be it's gpio pin name and esp32-s2 has gpio pin45 and pin46

tore-espressif, 2021-06-10:
I see, sry. Will be fixed soon.

closed 2021-07-14 by C47D

cggulick, 2023-01-03:
I think GPIO45 on the ESP32-S2 is input only. Really sucks. The ESP32-S3 allows for all GPIO pins to be input or output.