lvgl/lv_esp_idf #283: Maintainer inquiry

kisvegabor, 2021-06-09:
Hi @tore-espressif,

After consulting with @C47D and @embeddedt, I'd like to ask if you have the liking to become the maintainer of this repo. 

Seeing your past activity, it seems you have the same vision as us and you know ESP very well (no surprise :slightly_smiling_face: ).

If you are interested in it we could provide write access to this and the [lvgl_esp32_drivers](https://github.com/lvgl/lvgl_esp32_drivers) repo.

tore-espressif, 2021-06-14:
Hello @kisvegabor,
Thank you very much for the offer. I'd be very happy to join your efforts.

I'd like to come up with a suggestion of refactoring, focused on (but not only) on these issues:
https://github.com/lvgl/lvgl_esp32_drivers/issues/40
https://github.com/lvgl/lvgl_esp32_drivers/issues/44
https://github.com/lvgl/lvgl_esp32_drivers/issues/68
https://github.com/lvgl/lv_port_esp32/issues/272

kisvegabor, 2021-06-14:
> Thank you very much for the offer. I'd be very happy to join your efforts.

Awesome! I've sent the invitation for both repos. 
So from now you, @C47D, @embeddedt, and I have write access to these repos.

> I'd like to come up with a suggestion of refactoring,

Could you open an issue to summarize your ideas?

tore-espressif, 2021-06-14:
I hope I'll have something worth sharing within two weeks

kisvegabor, 2021-06-14:
Okay, great! :slightly_smiling_face: 

I think this issue can be closed. See you in the issues and PRs  :slightly_smiling_face:

closed 2021-06-14 by kisvegabor