lvgl/lv_esp_idf #285: error: 'CONFIG_LV_AXP192_PIN_SDA' undeclared

Rajssss, 2021-06-15:
We use GitHub issues for development related discussions.
Please use the [forum](https://forum.littlevgl.com/) to ask questions.

**Describe the issue**
I am trying to build and flash the demo widgets example on my ESP32 DevKitC V4 (Clone). I am using a 128x128 ST7735 Display, Selected predefined board as "ESP32 DevKit V4 with 38 pins". When I tried to build it, I got the error below:

```
../components/lvgl_esp32_drivers/lvgl_tft/st7735s.h:31:22: error: 'CONFIG_LV_AXP192_PIN_SDA' undeclared (first use in this function); did you mean 'CONFIG_LV_DISP_PIN_SDA'?
 #define AXP192_SDA   CONFIG_LV_AXP192_PIN_SDA
                      ^~~~~~~~~~~~~~~~~~~~~~~~

```
**Code to reproduce the issue**
demo widgets example with no change
**Expected Results**
Build sucess
**Actual Results**
```
../components/lvgl_esp32_drivers/lvgl_tft/st7735s.h:31:22: error: 'CONFIG_LV_AXP192_PIN_SDA' undeclared (first use in this function); did you mean 'CONFIG_LV_DISP_PIN_SDA'?
 #define AXP192_SDA   CONFIG_LV_AXP192_PIN_SDA
                      ^~~~~~~~~~~~~~~~~~~~~~~~

```
**ESP32 Chip version**
ESP32-D0WD
**ESP-IDF version**
V4.4.0
**Development kit used**
ESP32 DevKitC V4 (Clone)
**Development machine OS**
Linux (Xubuntu)
**Compilation warnings/errors (if available)**
```
../components/lvgl_esp32_drivers/lvgl_tft/st7735s.h:31:22: error: 'CONFIG_LV_AXP192_PIN_SDA' undeclared (first use in this function); did you mean 'CONFIG_LV_DISP_PIN_SDA'?
 #define AXP192_SDA   CONFIG_LV_AXP192_PIN_SDA
                      ^~~~~~~~~~~~~~~~~~~~~~~~

```
**If possible, copy the compilation log into a file and attach it here**

C47D, 2021-06-15:
@Rajssss Thanks for the detailed report, I guess we should ask the user if their kit includes touch or not. Will work on the fix.

Rajssss, 2021-06-15:
> @Rajssss Thanks for the detailed report, I guess we should ask the user if their kit includes touch or not. Will work on the fix.

User? Sorry, I don't get it.

C47D, 2021-06-16:
I've just created the [fix/st7735_no_touch](https://github.com/lvgl/lvgl_esp32_drivers/tree/fix/st7735_no_touch) branch on the [lvgl_esp32_drivers](https://github.com/lvgl/lvgl_esp32_drivers) repo with the fix.

Can you go to the `components/lvgl_esp32_drivers` and run the following commands? They track the new branch of the `lvgl_esp32_drivers` repo.

```console
cd components/lvgl_esp32_drivers
git fetch origin
git switch --track origin/fix/st7735_no_touch
```

Then go back to the project root directory and try to build it?

Rajssss, 2021-06-19:
> I've just created the [fix/st7735_no_touch](https://github.com/lvgl/lvgl_esp32_drivers/tree/fix/st7735_no_touch) branch on the [lvgl_esp32_drivers](https://github.com/lvgl/lvgl_esp32_drivers) repo with the fix.
> 
> Can you go to the `components/lvgl_esp32_drivers` and run the following commands? They track the new branch of the `lvgl_esp32_drivers` repo.
> 
> ```
> cd components/lvgl_esp32_drivers
> git fetch origin
> git switch --track origin/fix/st7735_no_touch
> ```
> 
> Then go back to the project root directory and try to build it?

I have just built it without any issues, this issue should be marked closed now. Thank you.

closed 2021-06-23 by tore-espressif