lvgl/lv_esp_idf #286: Incomplete TTGO T-Display board configuration

abouillot, 2021-06-19:
**Describe the issue**
The compilation of examples code cannot succeed when configuring the environment for a TTGO T-Display board. It fail with  a TFT_SPI_HOST not defined

**Code to reproduce the issue**
-launch idf.py menuconfig
-go to components config/LVGL TFT display controller/Select predefined display configuration
-Select TTGO T-Display
-save config
-compile with make flash

**Expected Results**
Example is correctly compiled

**Actual Results**
Copmilation fail in with error

**ESP32 Chip version**
ESP32 

**ESP-IDF version**
4.2.1

**Development kit used**

**Development machine OS**
Linux Debian 10

**Compilation warnings/errors (if available)**
CC build/lvgl_esp32_drivers/lvgl_helpers.o
/home/debian/code/gui/lv_port_esp32/components/lvgl_esp32_drivers/lvgl_helpers.c: In function 'lvgl_driver_init':
/home/debian/code/gui/lv_port_esp32/components/lvgl_esp32_drivers/lvgl_helpers.c:111:26: error: 'TFT_SPI_HOST' undeclared (first use in this function); did you mean 'VSPI_HOST'?
     lvgl_spi_driver_init(TFT_SPI_HOST,
                          ^~~~~~~~~~~~
                          VSPI_HOST

[bld.log](https://github.com/lvgl/lv_port_esp32/files/6680974/bld.log)

**If possible, copy the compilation log into a file and attach it here**

jsmestad, 2021-06-26:
@abouillot I hit this as well today. Did you edit anything manually to get around the issue?

abouillot, 2021-06-26:
@jsmestad I'm not KConfig expert to dig deeply in it, but I guess there are some miss use of KConfig in some places.

But a quick and dirty fix is to change lines 449/450 to
   prompt "TFT SPI Bus." if LV_TFT_DISPLAY_PROTOCOL_SPI #&& \
   #!LV_PREDEFINED_DISPLAY_TTGO

this will enable the section to select the SPI port to use.

Bear in mind, some issue will pop up on Offset selection, and remember to properly adjust the X/Y pixels to 240/135.

Good luck!

jsmestad, 2021-06-26:
@abouillot yeah I just figured out the offset issue and put up a PR with what I think is the proper fix -- lvgl/lvgl_esp32_drivers#78

C47D, 2021-06-27:
lvgl/lvgl_esp32_drivers#78 just got merged, could you check if this fix your issue @abouillot? You would need to update the `lvgl_esp32_drivers` component of this repo to do so. Let me know if you need any help.

jsmestad, 2021-06-30:
@C47D this fixed the issue for me

C47D, 2021-06-30:
@jsmestad Thanks for the confirmation, I guess we can consider this issue solved?

jsmestad, 2021-07-03:
Yeah I will submit a new ticket or reply here if things are hitting any issues

C47D, 2021-07-06:
@jsmestad Thanks 😸

closed 2021-07-06 by C47D